ghdl/ghdl #284: A missing file causes an exception in bootstrap mode

Paebbels, 2017-02-17:
I'm fixing the PowerShell compile scripts for GHDL after @1138-4EB refactored the directory structure. A missing input file causes an exception. I see only the `--bootstrap` parameter as a different to a normal flow.

```
    file: v87\textio.v87
Get-Content : Cannot find path 'D:\git\GitHub\libraries\std\textio.vhdl' because it does not exist.
At D:\git\GitHub\ghdl\dist\windows\compile-libraries.ps1:278 char:3
+         Get-Content "$VHDLSourceLibraryDirectory\$VHDLSourcesIndex\$S ...
+         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : ObjectNotFound: (D:\git\GitHub\libraries\std\textio.vhdl:String) [Get-Content], ItemNotFoundException
    + FullyQualifiedErrorId : PathNotFound,Microsoft.PowerShell.Commands.GetContentCommand

      System.Management.Automation.RemoteException
      ******************** GHDL Bug occured ****************************
      Please report this bug on https://github.com/tgingold/ghdl/issues
      GHDL release: GHDL 0.34-dev (commit: 2017-02-17;  git branch: paebbels/master';  hash: 2f2f023) [Dunoon edition]
      Compiled with GNAT Version: GPL 2016 (20160515-49)
      Target: i686-pc-mingw32
      In directory: D:\git\GitHub\build\mcode\lib\v87\std\
      Command line:
      D:\git\GitHub\ghdl\build\mcode\ghdl.exe -a -C --bootstrap --std=87 --work=std textio.v87
      Exception CONSTRAINT_ERROR raised
      Exception information:
      raised CONSTRAINT_ERROR : files_map.adb:806 range check failed
      ******************************************************************
```

Line 1 is reported by the build script (next file to process). Lines 2...9 are reported by PowerShell due to the failing GHDL execution.

tgingold, 2017-02-17:
I cannot reproduce it. I get:

```
./ghdl_mcode -a -C --bootstrap --std=87 --work=std textio.v87
./ghdl_mcode:error: cannot open textio.v87
```

Paebbels, 2017-02-17:
Sorry I can't reproduce it any more since I fixed the compile flow in #285.

tgingold, 2017-02-17:
Let's close this issue, as it is a transient build issue.

closed 2017-02-17 by tgingold