ghdl/ghdl #286: Code coverage with LLVM backend

Paebbels, 2017-02-17:
According to some webpages, LLVM can collect gcov compatible coverage information by enabling the same or similar compiler switches as used for GCC. I'm not sure if this applies to LLVM or only to CLang.

Is it possible to get code coverage for the LLVM backend by just passing a few parameters?

tgingold, 2017-02-19:
Extra instructions should be generated.  Cf: https://wiki.dlang.org/LDC_LLVM_profiling_instrumentation

So, the answer is no: not possible to get coverage with just a few parameters.

Brian-Drummond, 2017-04-12:
If anyone is looking at this, a couple of comments: 
The GCC/gcov route works well for code coverage (I use lcov/genhtml for HTML report generation) and it would be great to see the same with LLVM.

However while it implements branch coverage, the result is not as useful. In addition to branches generated by "if", "case" etc, there are a huge number of un-architected branches. For example, any signal assignment involves several branches (such as, do we schedule processes sensitive to this signal?) which are of no interest in coverage reports.

It would be useful to be able to colour branches as "track/don't track" for code coverage purposes, and for GHDL to distinguish between explicit branches in VHDL code and what I'm calling unarchitected branches. (I believe a similar problem applies to branch coverage and C++ exception mechanisms, so this doesn't just affect the VHDL community)

I should probably submit this as an enhancement request, but mentioning it here is better than nothing. If anyone feels like running with it, please do so.

tgingold, 2017-04-15:
On 12/04/17 13:12, Brian-Drummond wrote:
> If anyone is looking at this, a couple of comments:
> The GCC/gcov route works well for code coverage (I use lcov/genhtml for
> HTML report generation) and it would be great to see the same with LLVM. [...]

This is doable with llvm but no with gcc I think.

Currently, coverage is only implemented as a side-effect; maybe we 
should have a first-class implementation.