ghdl/ghdl #289: ANSI color escape codes are shown on Windows consoles.

Paebbels, 2017-02-18:
One of the latest commits introduces colored outputs for GHDL. This feature can be controlled by `-fcolor-diagnostics` and `-fno-color-diagnostics`.

Currently, ANSI color codes are emitted in the Windows command shell (`cmd.exe`) and in Windows PowerShell (`powershell.exe`).

Is there a way to detect the current shell/terminal to guess a better default setting?

Usefull environment indicators could be:
* `OS` which is `Windows_NT`, but never missing
* `windir` which is usually `C:\Windows`, but never empty or missing

PowerShell can be identified by:
* `PSModulePath`

The next revision of Windows' shell shall support ANSI escape codes. I'm not sure if this also applies to the old `cmd.exe` shell.

I can provide a PowerShell filter, which translates ANSI color codes into API calls to switch the text coloring. Shall I?

tgingold, 2017-02-19:
How does gcc behaves ?

We can simply disable ANSI sequences on windows (or use instead the Win32 API).

Paebbels, 2017-02-19:
Does Ada provide a simple way to access the console API? I have never used it on Windows except for .NET. But in .NET you are writing to a console class and not to STDOUT.

If it's to complex with the Windows API, then disable it by default on Window unless it's really supported in one of the shells. A user can still enable it by one of the switches.

tgingold, 2017-02-19:
On 19/02/17 08:13, Patrick Lehmann wrote:
> Does Ada provide a simple way to access the console API? I have never
> used it on Windows except for .NET. But in .NET you are writing to a
> console class and not to STDOUT.

Not directly, but you can use the Win32 API.

> If it's to complex with the Windows API, then disable it by default on
> Window unless it's really supported in one of the shells. A user can
> still enable it by one of the switches.

Make sense.

commit ca29d286 (2017-02-26): Disable color diagnotics by default on Windows.

Close #289
closed 2017-02-26 by tgingold