ghdl/ghdl #291: Compiling GHDL with GCC backend in MinGW on Windows doesn't recognize VHDL as a new language.

Paebbels, 2017-02-18:
As an extension to #174, I tried to compile GHDL with GCC backend in MinGW32/64 on Windows (AppVeyor). I get similar results in AppVeyor and on my local machine. The following description reflects my setup on my local machine.

It seems as if the configure process of GCC doesn't recognize VHDL as a new language in an MinGW environment. I could compile GCC with embedded GHDL in my virtual machine. So the Linux GCC build flow in itself is OK.

I downloaded the GCC 6.3.0 source files from GNU (https://ftp.gnu.org/gnu/gcc/gcc-6.3.0/) and applied all instructions as in `BUILD.txt`. Additionally, I exchanged `ortho-lang.c` with `ortho-lang-6.c`.

**This is my GCC configuration line:**
```Bash
../configure --prefix=/c/Tools/GHDL/0.34-dev-mingw64-gcc --enable-languages=c,vhdl --disable-bootstrap --disable-lto --disable-multilib --disable-libssp --disable-libgomp --disable-libquadmath
```

**This is the resulting output:**
```
configure: loading site script /mingw64/etc/config.site
checking build system type... x86_64-w64-mingw32
checking host system type... x86_64-w64-mingw32
checking target system type... x86_64-w64-mingw32
checking for a BSD-compatible install... /usr/bin/install -c
checking whether ln works... yes
checking whether ln -s works... no, using cp -p
checking for a sed that does not truncate output... /usr/bin/sed
checking for gawk... gawk
checking for libatomic support... yes
checking for libcilkrts support... no
checking for libitm support... no
checking for libsanitizer support... no
checking for libvtv support... yes
checking for libmpx support... no
checking for gcc... gcc
checking for C compiler default output file name... a.exe
checking whether the C compiler works... yes
checking whether we are cross compiling... no
checking for suffix of executables... .exe
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking for g++... g++
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking whether g++ accepts -static-libstdc++ -static-libgcc... yes
checking for gnatbind... gnatbind
checking for gnatmake... gnatmake
checking whether compiler driver understands Ada... yes
checking how to compare bootstrapped objects... tail +16c $$f1 > tmp-foo1; tail +16c $$f2 > tmp-foo2; cmp tmp-foo1 tmp-foo2
checking for objdir... .libs
checking for the correct version of gmp.h... yes
checking for the correct version of mpfr.h... yes
checking for the correct version of mpc.h... yes
checking for the correct version of the gmp/mpfr/mpc libraries... yes
checking for isl 0.16, 0.15, or deprecated 0.14... yes
checking for isl 0.16 or 0.15... yes
*** This configuration is not supported in the following subdirectories:
     target-libquadmath target-libcilkrts target-libitm target-libsanitizer target-libmpx target-libffi target-zlib target-libjava target-libgo gnattools gotools target-libada target-libstdc++-v3 target-libgfortran target-libbacktrace target-libobjc target-libgomp target-liboffloadmic target-libssp target-boehm-gc target-libvtv
    (Any other directories should still work fine.)
checking for default BUILD_CONFIG...
checking for --enable-vtable-verify... no
*** removing build-x86_64-w64-mingw32/libiberty/Makefile to force reconfigure
*** removing build-x86_64-w64-mingw32/libcpp/Makefile to force reconfigure
*** removing build-x86_64-w64-mingw32/fixincludes/Makefile to force reconfigure
*** removing intl/Makefile to force reconfigure
*** removing libiberty/Makefile to force reconfigure
*** removing zlib/Makefile to force reconfigure
*** removing libbacktrace/Makefile to force reconfigure
*** removing libcpp/Makefile to force reconfigure
*** removing libdecnumber/Makefile to force reconfigure
*** removing fixincludes/Makefile to force reconfigure
*** removing gcc/Makefile to force reconfigure
checking for bison... no
checking for byacc... no
checking for yacc... no
checking for bison... no
checking for gm4... no
checking for gnum4... no
checking for m4... m4
checking for flex... flex
checking for flex... flex
checking for makeinfo... no
/d/git/GitHub/ghdl/build/gcc-6.3.0/missing: line 81: makeinfo: command not found
checking for expect... no
checking for runtest... no
checking for ar... ar
checking for as... as
checking for dlltool... dlltool
checking for ld... (cached) C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/6.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe
checking for lipo... no
checking for nm... nm
checking for ranlib... ranlib
checking for strip... strip
checking for windres... windres
checking for windmc... windmc
checking for objcopy... objcopy
checking for objdump... objdump
checking for readelf... readelf
checking for cc... cc
checking for c++... c++
checking for gcc... gcc
checking for gcj... no
checking for gfortran... no
checking for gccgo... no
checking for ar... no
checking for ar... ar
checking for as... no
checking for as... as
checking for dlltool... no
checking for dlltool... dlltool
checking for ld... no
checking for ld... ld
checking for lipo... no
checking for lipo... no
checking for nm... no
checking for nm... nm
checking for objcopy... no
checking for objcopy... objcopy
checking for objdump... no
checking for objdump... objdump
checking for ranlib... no
checking for ranlib... ranlib
checking for readelf... no
checking for readelf... readelf
checking for strip... no
checking for strip... strip
checking for windres... no
checking for windres... windres
checking for windmc... no
checking for windmc... windmc
checking where to find the target ar... host tool
checking where to find the target as... host tool
checking where to find the target cc... just compiled
checking where to find the target c++... host tool
checking where to find the target c++ for libstdc++... host tool
checking where to find the target dlltool... host tool
checking where to find the target gcc... just compiled
checking where to find the target gcj... host tool
checking where to find the target gfortran... host tool
checking where to find the target gccgo... host tool
checking where to find the target ld... host tool
checking where to find the target lipo... host tool
checking where to find the target nm... host tool
checking where to find the target objcopy... host tool
checking where to find the target objdump... host tool
checking where to find the target ranlib... host tool
checking where to find the target readelf... host tool
checking where to find the target strip... host tool
checking where to find the target windres... host tool
checking where to find the target windmc... host tool
checking whether to enable maintainer-specific portions of Makefiles... no
configure: creating ./config.status
config.status: creating Makefile
```

I can't see any `vhdl` notice, but on the other hand I also see no error that a given language is missing.

tgingold, 2017-02-19:
The top configure doesn't do that much.  Try make configure-gcc after this configure.

Paebbels, 2017-02-19:
Running `make configure-gcc` didn't change anything. The GHDL files are still uncompiled.

tgingold, 2017-02-19:
On 19/02/17 07:48, Patrick Lehmann wrote:
> Running |make configure-gcc| didn't change anything. The GHDL files are
> still uncompiled.

Does it explain why ?

Paebbels, 2017-02-19:
Here are the last lines from `make`:

```make
g++ -fno-PIE -c  -DIN_GCC_FRONTEND -g -O2 -DIN_GCC     -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -I. -I../../gcc -I../../gcc/. -I../../gcc/../include -I../../gcc/../libcpp/include  -I../../gcc/../libdecnumber -I../../gcc/../libdecnumber/bid -I../libdecnumber -I../../gcc/../libbacktrace   -o cc1-checksum.o -MT cc1-checksum.o -MMD -MP -MF ./.deps/cc1-checksum.TPo cc1-checksum.c
g++ -no-pie   -g -O2 -DIN_GCC     -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -static-libstdc++ -static-libgcc -Wl,--stack,12582912 -o cc1.exe c/c-lang.o c-family/stub-objc.o attribs.o c/c-errors.o c/c-decl.o c/c-typeck.o c/c-convert.o c/c-aux-info.o c/c-objc-common.o c/c-parser.o c/c-array-notation.o c/c-fold.o c-family/c-common.o c-family/c-cppbuiltin.o c-family/c-dump.o c-family/c-format.o c-family/c-gimplify.o c-family/c-indentation.o c-family/c-lex.o c-family/c-omp.o c-family/c-opts.o c-family/c-pch.o c-family/c-ppoutput.o c-family/c-pragma.o c-family/c-pretty-print.o c-family/c-semantics.o c-family/c-ada-spec.o c-family/c-cilkplus.o c-family/array-notation-common.o c-family/cilk.o c-family/c-ubsan.o i386-c.o winnt-c.o msformat-c.o \
  cc1-checksum.o libbackend.a main.o libcommon-target.a libcommon.a ../libcpp/libcpp.a ../libdecnumber/libdecnumber.a libcommon.a ../libcpp/libcpp.a -lintl -liconv ../libbacktrace/.libs/libbacktrace.a ../libiberty/libiberty.a ../libdecnumber/libdecnumber.a  -lisl -lmpc -lmpfr -lgmp   -L./../zlib -lz
g++ -no-pie   -g -O2 -DIN_GCC     -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -static-libstdc++ -static-libgcc -Wl,--stack,12582912 gcov.o \
        hash-table.o ggc-none.o libcommon.a ../libcpp/libcpp.a -lintl -liconv ../libbacktrace/.libs/libbacktrace.a ../libiberty/libiberty.a ../libdecnumber/libdecnumber.a  -o gcov.exe
g++ -no-pie   -g -O2 -DIN_GCC     -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -static-libstdc++ -static-libgcc -Wl,--stack,12582912 gcov-dump.o \
        hash-table.o ggc-none.o\
        libcommon.a ../libcpp/libcpp.a -lintl -liconv ../libbacktrace/.libs/libbacktrace.a ../libiberty/libiberty.a ../libdecnumber/libdecnumber.a  -o gcov-dump.exe
g++ -no-pie   -g -O2 -DIN_GCC     -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -static-libstdc++ -static-libgcc -Wl,--stack,12582912 gcov-tool.o libgcov-util.o libgcov-driver-tool.o libgcov-merge-tool.o libcommon.a ../libcpp/libcpp.a -lintl -liconv ../libbacktrace/.libs/libbacktrace.a ../libiberty/libiberty.a ../libdecnumber/libdecnumber.a  -o gcov-tool.exe
cd vhdl; gnatmake -c -aI/d/git/GitHub/ghdl/build/gcc-6.3.0/gcc/vhdl ortho_gcc-main \
 -cargs -g -O2 -D__USE_MINGW_ACCESS -gnata -gnat05 -gnaty3befhkmr -gnatwae
gnatmake: "ortho_gcc-main" not found
make[2]: *** [../../gcc/vhdl/Make-lang.in:67: ghdl1.exe] Fehler 4
make[2]: Verzeichnis „/d/git/GitHub/ghdl/build/gcc-6.3.0/build-mingw64/gcc“ wird verlassen
make[1]: Verzeichnis „/d/git/GitHub/ghdl/build/gcc-6.3.0/build-mingw64“ wird verlassen
make[1]: *** [Makefile:4117: all-gcc] Fehler 2
make: *** [Makefile:866: all] Fehler 2
```

Paebbels, 2017-02-19:
Current status for the AppVeyor flow:
Patching the `ortho-lang.c` file seems to go wrong.

It didn't fail according to this line:
https://ci.appveyor.com/project/tgingold/ghdl/build/0.34-dev-145/job/iglwg577s8w4e387#L402

But the log shows typical signs for a wrong ortho-lang.c file:
https://ci.appveyor.com/project/tgingold/ghdl/build/0.34-dev-145/job/iglwg577s8w4e387#L1178

Here is the patching mechanism:
```PowerShell
cd $env:GCC_SOURCE_DIR\gcc\vhdl
Write-Host "Patch ortho-lang.c..." -Foreground Yellow
mv "ortho-lang.c" "ortho-lang-4.c"
mv "ortho-lang-$($env:BUILD_GCC_VERSION[0]).c" "ortho-lang.c"
```

https://github.com/Paebbels/ghdl/blob/paebbels/appveyor-mingw64-gcc5/dist/windows/appveyor/build.ps1#L64-L67

Paebbels, 2017-02-23:
I see compiling Ada files in the report...

So we hit the next problem (line 5109):
```
The directory that should contain system headers does not exist:
  /mingw/include
Makefile:2907: recipe for target 'stmp-fixinc' failed
```

The configuration of GCC and GNAT can be seen at lines 266 and following.

Hmmm there is a number missing, right?
My local installation has a `/mingw64/include` path (and the same for mingw32).

While searching for this crippled path I found that it is used at line 4166/4220/...
`-DNATIVE_SYSTEM_HEADER_DIR=\"/mingw/include\"`

I would say it comes from the GCC configuration process, right? I doesn't seem to be GHDL related.

tgingold, 2017-02-23:
On 23/02/17 12:59, Patrick Lehmann wrote:
> I see compiling Ada files in the report...
>
> So we hit the next problem (line 5109):
>
> |The directory that should contain system headers does not exist:
> /mingw/include Makefile:2907: recipe for target 'stmp-fixinc' failed |

That's a good news.

> The configuration of GCC and GNAT can be seen at lines 266 and following.
>
> Hmmm there is a number missing, right?
> My local installation has a |/mingw64/include| path (and the same for
> mingw32). [...]

Try to configure with:

--with-native-system-header-dir=/mingw64/include

Tristan.

tgingold, 2017-03-01:
I was able to configure and build gcc (C only) with msys2 using two different methods.

1) Use /mingw32 as a prefix:
```shell
../gcc-6.3.0/configure --prefix=/mingw32 --enable-languages=c --disable-bootstrap --disable-lto --disable-multilib --disable-libssp --disable-libgomp --build=i686-w64-mingw32 --target=i686-w64-mingw32 --host=i686-w64-mingw32 --with-native-system-header-dir=/mingw32/include
```
But you will have to install it carefully.

2) Install includes and libs in prefix:
```shell
$ ../gcc-6.3.0/configure --prefix=c:/prefix --enable-languages=c --disable-bootstrap --disable-lto --disable-multilib --disable-libssp --disable-libgomp --build=i686-w64-mingw32 --target=i686-w64-mingw32 --host=i686-w64-mingw32 --with-native-system-header-dir=/mingw32/include
$ cp -a /mingw32/i686-w64-mingw32/include /c/prefix/i686-w64-mingw32
$ cp -a  /mingw32/i686-w64-mingw32/lib /c/prefix/i686-w64-mingw32
$ make
```

closed 2017-06-14 by tgingold