ghdl/ghdl #293: Alias problem

LarsAsplund, 2017-02-20:
Running the testbench below results in a failing assert but it works if I remove the alias. Note that the alias isn't used.

``` vhdl
library vunit_lib;
context vunit_lib.vunit_context;

library ieee;
use ieee.float_pkg.all;

entity tb_test is
  generic (
    runner_cfg : string);
end tb_test;

architecture tb of tb_test is
begin
    test_runner : process
      alias fp32 is float32;
    begin
      test_runner_setup(runner, runner_cfg);

      assert zerofp = neg_zerofp;

      test_runner_cleanup(runner);
      wait;
    end process test_runner;
end;
```

tgingold, 2017-02-23:
I can reproduce the issue.  Interesting!

LarsAsplund, 2017-02-23:
Yes, that's a rather weird bug!

tgingold, 2017-02-23:
Thinking twice (and re-reading the vhdl08 LRM), it is not a bug but a feature:
6.6.3 e)
Finally, if the name denotes a type or a subtype, then implicit alias declarations for each predefined operation

So when the alias is present, the implicit "=" operator is visible instead of the overriding one.

Not user friendly feature!

tgingold, 2017-02-23:
I will close this issue, but feel free to comment or argue 😄

LarsAsplund, 2017-02-23:
Please do. I will read and try to understand.

LarsAsplund, 2017-02-23:
Ok, so I will get an implicit alias for "=" that maps to the "=" defined float32. The original failing code looked more like this

``` vhdl
library vunit_lib;
context vunit_lib.vunit_context;

library ieee;
use ieee.float_pkg.all;

entity tb_test2 is
  generic (
    runner_cfg : string);
end tb_test2;

architecture tb of tb_test2 is
begin
    test_runner : process
      alias fp32 is float32;
      constant zero0   : fp32 := zerofp;
      constant negzero : fp32 := neg_zerofp;
    begin
      test_runner_setup(runner, runner_cfg);

      assert zero0 = negzero;

      test_runner_cleanup(runner);
      wait;
    end process test_runner;
end;
```

Isn't the "=" I get the one I want? If I remove the alias and replace `fp32` with `float32` it works.

tgingold, 2017-02-24:
No, you don't get the "=" you need.  You get the predefined "=" that compare bits to bits, instead of the user defined "=" that knowns 0.0 = -0.0

closed 2017-02-24 by tgingold

LarsAsplund, 2017-02-24:
Oh, I see. The key is the word _predefined_. That's a horrible feature! It's seems like ModelSim and Aldec's simulators ignored that. They all use the user defined "="

commit 394c9b5f (2017-02-24): Add testcase for #293

Close #293