ghdl/ghdl #295: Failing on denormal/subnormal floating point numbers

LarsAsplund, 2017-02-23:
It seems like numbers that should be possible to represent as denormal/subnormal numbers with a 64-bit implementation of real becomes zeros. Here is an example

``` vhdl
library vunit_lib;
context vunit_lib.vunit_context;

entity tb_test4 is
  generic (
    runner_cfg : string);
end tb_test4;

architecture tb of tb_test4 is
begin
    test_runner : process
    begin
      test_runner_setup(runner, runner_cfg);

      assert 2.0**(-1023) > 0.0 report "This passes";
      assert 2.0**(-1024) > 0.0 report "This fails";

      test_runner_cleanup(runner);
      wait;
    end process test_runner;
end;
```

tgingold, 2017-02-24:
LRM08 defines exponentiation as (9.2.8):
Exponentiation with an integer exponent is equivalent to repeated multiplication of the left operand by itself for a number of times indicated by the absolute value of the exponent and from left to right; if the exponent is negative, then the result is the reciprocal of that obtained with the absolute value of the exponent.

So 2.0**(-1024) = 1.0 / (2.0**1024)
Using IEEE754:
= 1.0 / (+infinity)
= 0.0

Hence the result.

tgingold, 2017-02-24:
So VHDL exponentiation is defined by the mathematical meaning of exponentiation which doesn't work completely well with IEEE754.

LarsAsplund, 2017-02-24:
Thanks for clarifying. This does indeed pass

``` vhdl
library vunit_lib;
context vunit_lib.vunit_context;

entity tb_test4 is
  generic (
    runner_cfg : string);
end tb_test4;

architecture tb of tb_test4 is
begin
    test_runner : process
    begin
      test_runner_setup(runner, runner_cfg);

      assert 2.0**(-1023) > 0.0 report "This passes";
      assert (2.0**(-1023))/2 > 0.0 report "This fails";

      test_runner_cleanup(runner);
      wait;
    end process test_runner;
end;
```

closed 2017-02-24 by LarsAsplund