ghdl/ghdl #300: type converted actual in port map doesn't get actual's initial value

diogratia, 2017-03-01:
See <a href="https://stackoverflow.com/questions/42528345/unnatural-ghdl-behavior-of-type-conversion-between-std-logic-vector-and-unsigned">Unnatural GHDL behavior of type conversion between std_logic_vector and unsigned in association list</a> on Stackoverflow.

```vhdl
library ieee;
use ieee.std_logic_1164.all;

entity inner is
  port (
  clk : in std_logic;
 inner_counter : in std_logic_vector(6 downto 0));
end inner;

architecture default of inner is
begin
  --do something
end architecture;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test_bench is
end test_bench;

architecture default of test_bench is
  signal clk : std_logic := '0';
  signal counter : unsigned(7 downto 0) := (others => '0');

begin

  i0: entity work.inner port map (
  clk => clk,
  inner_counter => std_logic_vector(counter(6 downto 0)));

  process
  begin
    clk <= '1';
    wait for 1 ns;
    clk <= '0';
    wait for 1 ns;
  end process;

  process(clk)
  begin
    if rising_edge(clk) then
      counter <= counter + 1;
    end if;
  end process;

end default;
```

produces:

[image]

Where the initial value is not conveyed to inner_counter yet we see the individual updates for the elements as they occur.

This was checked with ghdl-0.34dev commit afe8072bf5faa1dca1705a08e88818d454ff8758 from approximately 28 Sep 2016 (+/-, international dateline).

There was a possibly related Issue now closed, on 15 Dec 2016. See <a href="https://github.com/tgingold/ghdl/issues/225">regression, function call in port association · Issue #225</a> and the fix 60f245f0bc7046e8050162b9ab26571a490c2ac4. 

The distinction this being a type conversion (a basic operation) and Issue #225 involving a conversion function.

tgingold, 2017-03-02:
On 02/03/17 00:46, Dio Gratia wrote:
> See Unnatural GHDL behavior of type conversion between std_logic_vector
> and unsigned in association list
> <https://stackoverflow.com/questions/42528345/unnatural-ghdl-behavior-of-type-conversion-between-std-logic-vector-and-unsigned>
> on Stackoverflow.

Simulation is correct, vcd output is correct but your output isn't.
Are you using FST ?

tgingold, 2017-03-02:
I can see that ghw output is not correct.

diogratia, 2017-03-02:
I was going to put in a report statement but didn't want to dig around for a to_string. 

Anyway:

```vhdl
architecture default of inner is
    function to_string (inp: std_logic_vector) return string is
        variable image_str: string (1 to inp'length);
        alias input_str:  std_logic_vector (1 to inp'length) is inp;
    begin
        for i in input_str'range loop
            image_str(i) := character'VALUE(std_ulogic'IMAGE(input_str(i)));
        end loop;
        return image_str;
    end function;
    begin
      --do something
      process (inner_counter)
      begin
          report "innner_counter = " & to_string(inner_counter);
      end process;
end architecture;
```

> ghdl -a dis.vhdl
> ghdl -e test_bench
> ghdl -r test_bench --wave=dis.ghw --stop-time=21ns
> dis.vhdl:30:11:@0ms:(report note): innner_counter = 0000000
> dis.vhdl:30:11:@0ms:(report note): innner_counter = 0000001
> dis.vhdl:30:11:@2ns:(report note): innner_counter = 0000010
> dis.vhdl:30:11:@4ns:(report note): innner_counter = 0000011
> dis.vhdl:30:11:@6ns:(report note): innner_counter = 0000100
> dis.vhdl:30:11:@8ns:(report note): innner_counter = 0000101
> dis.vhdl:30:11:@10ns:(report note): innner_counter = 0000110
> dis.vhdl:30:11:@12ns:(report note): innner_counter = 0000111
> dis.vhdl:30:11:@14ns:(report note): innner_counter = 0001000
> dis.vhdl:30:11:@16ns:(report note): innner_counter = 0001001
> dis.vhdl:30:11:@18ns:(report note): innner_counter = 0001010
> dis.vhdl:30:11:@20ns:(report note): innner_counter = 0001011
> ./test_bench:info: simulation stopped by --stop-time:

And the initial value shows up here.

The reason for the two values at time 0 are the way the clk is inited to '0', and also has a rising edge at time 0 in the testbench. The counter uses rising_edge, telling us the are two different delta cycles and have two events at time 0, and that sounds possibly related to the issue.

commit 101dd876 (2017-03-09): Add start_of_processes callback, used by grt-waves

Fix #300
closed 2017-03-09 by tgingold