ghdl/ghdl #301: GHDL internal error with signed generic

putsplut, 2017-03-02:
Cocotb test "dec_viterbi" has a few issues:
1.) Change conversions in generic_sp_ram.vhd to not use synopsis libs (NOT a ghdl bug!)
2.) GHDL has problems with the top level module being named the same as the library.  In my copy below, I've renamed the entity by appending an "_m0".  I think this is a "bug", or a least a feature.
3.) The decoder passes a signed number as a GENERIC to a sub module (acs).  GHDL produces the following failure.  Changing the GENERIC to a natural allows the elaboration to succeed.
The following error was produced using hosts of Linux and OS X, both on the same GHDL version.
4.) The test still fails, but the issues appear to already be addressed by other bug reports.

First post, so not sure how to tag this as Cocotb related (if it should be)

******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: GHDL 0.34-dev (2016-09-14-403-g439664c) [Dunoon edition]
Compiled with GNAT Version: GPL 2016 (20160515-49)
Target: x86_64-apple-darwin14.5.0
In directory: /Users/XXX/repos/cocotb/tests/test_cases/test_iteration_vhdl/sim_build/
Command line:
/opt/local/bin/ghdl -e --work=dec_viterbi dec_viterbi_m0
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : trans.adb:507
Load address: 0x109fdc000
Call stack traceback locations:
0x10a1d6b87 0x10a1d6e77 0x10a1d6fd7 0x10a1dd8a2 0x10a228948 0x10a228cd8 0x10a228d69 0x10a1c3498 0x10a1af11c 0x10a1af367 0x10a1af877 0x10a1afded 0x10a1bc95c 0x10a1bc846 0x10a20f988 0x10a1cc108 0x10a1cc7f6 0x10a1f38b4 0x10a239266 0x10a2393e8 0x10a239615 0x10a23b621 0x10a1dfb67 0x10a236db9 0x10a1ea0b0 0x10a23b8cf 0x10a245090 0x10a24204b 0x10a0bc16d 0x109fde7e8
******************************************************************

tgingold, 2017-06-09:
I can reproduce the crash; investigating.

tgingold, 2017-06-12:
The crash is now fixed, but I have to investigate point 2.

tgingold, 2017-06-12:
I think that GHDL is correct for the second point.  If you don't agree, please create a new issue.

commit d986b9af (2017-06-12): Fix crash of implicit conversion in instantiation.

Fix #301
closed 2017-06-12 by tgingold