ghdl/ghdl #310: uninstantiated package declaration type declaration depending on generic analysis fails

diogratia, 2017-03-06:
Instantiating the IEEE package fixed_generic_pkg works.

A simple test case fails analysis:

```vhdl
package b is
    generic ( X: natural := 4);
    type m is array (natural range <>) of bit_vector (X - 1 downto 0);
end package;
```
where:
> ghdl -a --std=08 b.vhdl

yields:

> ******************** GHDL Bug occured ****************************
> Please report this bug on https://github.com/tgingold/ghdl/issues
> GHDL release: GHDL 0.34dev (20151126) [Dunoon edition]
> Compiled with GNAT Version: GPL 2015 (20150428-49)
> In directory: /Users/david_koontz/Desktop/VHDL/stackoverflow/file_io/
> Command line:
> /usr/local/bin/ghdl1-llvm --std=08 -P/usr/local/lib/ghdl/v08/std/ -P/usr/local/lib/ghdl/v08/ieee/ -c -o b.o b.vhdl
> Exception TYPES.INTERNAL_ERROR raised
> Exception information:
> Exception name: TYPES.INTERNAL_ERROR
> Message: trans.adb:507
> Load address: 0x101fc2000
> Call stack traceback locations:
> 0x1020e647f 0x1020e67e4 0x1020e699a 0x10214460e 0x102144777 0x1020c5def 0x1020cac4a 0x102131a62 0x10214bdf0 0x1020f01fa 0x10215a77b 0x1020028a9 0x10200336b 0x101fc4fce
> ******************************************************************
> ghdl: compilation error

For ghdl-0.34dev llvm (on OS X 10.10.5) built from commit afe8072bf5faa1dca1705a08e88818d454ff8758 from 28 September 2016.

The error is found in trans.adb at Line 507 shows up (in that commit) in procedure Find_Scope in the immediate case statement with the expression Scope.Kind,  choice Var_Scope_None with a statement raise Internal_Error. 

The distinction in the above test case from generic_fixed_pkg is the use of a generic in a type declaration as a bound expression.

Looking through src/vhdl/translate for occurrences of you can find dependencies for Is_Uninstantiated_Package (Decl) in trans-chap2.adb that may hint at this issue being an uncovered corner case.

A presumably functional full test case would look something like:

```vhdl
package b is
    generic ( X: natural := 4);
    type m is array (natural range <>) of bit_vector (X - 1 downto 0);
end package;

package bb is new work.b generic map ( X => 6);

use work.bb.all;

entity bbb is
end entity;

architecture a of bbb is

begin
    process
        variable n:  m (0 to 0);
    begin
        report "X = " & integer'image(X);
        report "m'left = " & integer'image(n(0)'left);
        wait;
    end process;
end architecture;
```
(The VHDL highlighter failure for the reserved word entity is caused by the package instantiation).

Paebbels, 2017-03-06:
Hello David,

generics on packages got a lot of fixes and new features like types. You need a newer GHDL build to test it.

tgingold, 2017-03-06:
On 06/03/17 20:21, Patrick Lehmann wrote:
> Hello David,
>
> generics on packages got a lot of fixes and new features like types. You
> need a newer GHDL build to test it.

It still fails with master.

diogratia, 2017-03-06:
I searched for file changes before posting this issue.  It turns out to be easy using SourceTree and any subsequent changes from the commit didn't look like they were applicable.

commit 59aa09dd (2017-03-14): Handle generic package without a body.

Fix #310
closed 2017-03-14 by tgingold