ghdl/ghdl #315: Can't assign integer literal to a constant created from a generic type

Paebbels, 2017-03-10:
Hello,

the following code gives an error:
```VHDL
package pkg is
  generic (
    type T
  );
  constant b : T := 5;
end package;
```

The error is:
> .\GenericSubprogram.vhdl:5:21: can't match integer literal with type interface type "t"

All interface types implicitly define operations like allocation, assignment, type qualification and type conversion. So it should be possible to assign a universal integer literal to an interface type.

The (type) check needs to be deferred until elaboration, because the specific type of T is still unknown at analysis time. I think this is a bug/missing feature, because type generics are quite new to GHDL.

Kind regards
    Patrick

tgingold, 2017-03-11:
I think ghdl is correct here.
Assignments are allowed only if both sides are of the same type, which is not the case here.

closed 2017-03-11 by tgingold

Paebbels, 2017-03-11:
What about the implicit conversion of an universal integer literal to T? Type conversion is one of the allowed operations for every generic type.

I think the conversation is present at analysis time. A detailed check, which type conversion to choose and to apply must be delayed until elaboration time -- until T is fully known.

Otherwise, the complete generic types sections make no sense in VHDL, right?

tgingold, 2017-03-11:
On 11/03/17 11:43, Patrick Lehmann wrote:
> What about the implicit conversion of an universal integer literal to T?
> Type conversion is one of the allowed operations for every generic type.

This means that T (x) is allowed.
But an implicit conversion from universal integer is allowed only to 
integer type, which T is not (as it is not defined).

> I think the conversation is present at analysis time. A detailed check,
> which type conversion to choose and to apply must be delayed until
> elaboration time -- until T is fully known.

I think there is no such rule in the LRM.  Types are determined at 
analysis time.

> Otherwise, the complete generic types sections make no sense in VHDL, right?

You guess correctly!  You cannot index or slice an interface type for 
the same reason.

Tristan.

Paebbels, 2017-03-11:
So with [LCS 059](http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/TopLCS2016_059) adding 6.5.3.3 to the LRM, it would work?
Or do I need to specify that implicit type conversions from universal integer to any integer type (`range <>`) is allowed too?

---------------------
Notification list: @JimLewis

tgingold, 2017-03-11:
On 11/03/17 17:25, Patrick Lehmann wrote:
> So with LCS 059
> <http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/TopLCS2016_059> adding
> 6.5.3.3 to the LRM, it would work?

Yes.

> Or do I need to specify that implicit type conversions from universal
> integer to any integer type (|range <>|) is allowed too?

I think the rule already exists.

JimLewis, 2017-03-11:
It would work if you designated the type as range <>

> So with LCS 059 <http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/TopLCS2016_059> adding 6.5.3.3 to the LRM, it would work?
> Or do I need to specify that implicit type conversions from universal [...]

-- 
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Jim Lewis
VHDL Training Expert, SynthWorks
IEEE 1076 VHDL Working Group Chair
Open Source VHDL Verification Methodology (OSVVM), Chief Architect and Co-founder

1-503-590-4787
Jim@SynthWorks.com
http://www.SynthWorks.com

VHDL Training on leading-edge, best coding practices for hardware design and verification.
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

Paebbels, 2017-03-11:
I just want to make sure (as one of the authors of that big LRM change), that we didn't miss anything and we finally get tools that don't implement it as intended because it's not written into the LRM and the whole feature gets useless :).

I'm not only meaning GHDL being pedantic about the LRM, what's almost every time a great thing, but also other tools sparing the effort to think about a feature while implementing.