ghdl/ghdl #317: Exception caused, while running PoC.sort.sortnet.* testbenches.

Paebbels, 2017-03-10:
Hello,

as you started to add unconstrained record and type generic support, 3 PoC testbenches got broken. I didn't want to bother you with that problem as everything was under construction. In parallel, @JimLewis updated OSVVM and I improved these three testbenches with his new generic scoreboard.

So I'm not sure if it's related to OSVVM, the scoreboard or protected types, or something got broken while implementing unconstrained records.

The tests can be run in PoC with:
```Bash
./poc.sh -v ghdl PoC.sort.sortnet.*
```

You may need to update the pre-compiled OSVVM library:
```Bash
./tools/precompile/compile-osvvm.sh -ghdl
```

If PoC doesn't run the testbenches when using a star expression, then use either the full names or change the setting `RepositoryKind` in `./py/config.private.ini` to `private`.
(We disabled these test on Travis until GHDL is fixes.)

The full names are:
* PoC.sort.sortnet.BitonicSort
* PoC.sort.sortnet.OddEvenSort
* PoC.sort.sortnet.OddEvenMergeSort

All three testbenches use the new scoreboard.

------------------------------------------------
GHDL reports this exception while analyzing:
```
Testbench: PoC.sort.sortnet.BitonicSort
Running analysis for every vhdl file...
  ghdl analyze messages for 'D:\git\PoC\tb\sort\sortnet\sortnet_BitonicSort_tb.vhdl'
  ------------------------------------------------------------------------------

  ******************** GHDL Bug occured ****************************
  Please report this bug on https://github.com/tgingold/ghdl/issues
  GHDL release: GHDL 0.34-dev (commit: 2017-03-10;  git branch: paebbels/master';  hash: 2b13cfe) [Dunoon edition]
  Compiled with GNAT Version: GPL 2016 (20160515-49)
  Target: i686-pc-mingw32
  In directory: D:\git\PoC\temp\ghdl\
  Command line:
  C:\Tools\GHDL\0.34-dev-gnatgpl32-mcode\bin\ghdl.exe -a -fexplicit -frelaxed-rules --warn-binding --no-vital-checks --mb-comments --syn-binding -fpsl --ieee=synopsys --std=08 --work=te
st -PD:\git\PoC\temp\precompiled\ghdl D:\git\PoC\tb\sort\sortnet\sortnet_BitonicSort_tb.vhdl
  Exception TYPES.INTERNAL_ERROR raised
  Exception information:
  raised TYPES.INTERNAL_ERROR : sem_inst.adb:235
  ******************************************************************
  ------------------------------------------------------------------------------
```

Thanks for looking into this
     Patrick

tgingold, 2017-05-19:
Should be fixed.

closed 2017-05-19 by tgingold

Paebbels, 2017-05-22:
I can't reproduce that error. Thanks!