ghdl/ghdl #318: How to run ghdl-0.34-dev-mingw64-llvm without MinGW64?

Paebbels, 2017-03-13:
Hello,

I got a step further in the question on how to deploy GHDL without a full MinGW installation (~ 3GiB). The answer is to copy all required dll files from MinGW64 into the bin directory of GHDL.

There are no dependencies for `ghdl.exe`.

So far I identified these files for `ghdl1-llvm.exe`:
* from `/msys64/mingw64/bin/`
  * `libgcc_s_seh-1.dll`
  * `libstdc++-6.dll`
  * `libwinpthread-1.dll`
  * `zlib1.dll`

These dependencies could be solved by additional Makefile rules for target `install` if the platform was MinGW.

The generated executable e.g. `tb.exe` requires:
* from `/msys64/mingw64/bin/`
  * `zlib1.dll`

GHDL needs to copy `zlib1.dll` from it's binary directory to the simulator output directory. Is there a way to influence the DLL binding from within GHDL? I think we discussed such topics while talking about VPI dlls. Maybe GHDL could influence the search order if it was build with MinGW and add it's own binary directory to the possible DLL folders.

Anyhow, if I run `ghdl -e tb`, I get this:
```
D:\Temp\ranges> & $GHDL -a .\ranges.vhdl
D:\Temp\ranges> & $GHDL -e -v tb
C:\Tools\GHDL\0.34-dev-mingw64-llvm\bin/ghdl1-llvm.exe -PC:\Tools\GHDL\0.34-dev-mingw64-llvm\lib/ghdl\v93\std\ -PC:\Tools\GHDL\0.34-dev-mingw64-llvm\lib/ghdl\v93\ieee\ --elab tb -l e~tb.lst -c -o e~tb.o e~tb
C:\Tools\GNAT\2016\bin\gcc.exe -o tb e~tb.o C:\Tools\GHDL\0.34-dev-mingw64-llvm\lib/ghdl\v93\std\std_standard.o ranges.o C:\Tools\GHDL\0.34-dev-mingw64-llvm\lib/ghdl/libgrt.a -ldbghelp -L.\ -LC:/msys64/mingw64/lib/gcc/x86_64-w64-mingw32/6.2.0/adalib/ -lz -Wl,--stack=0x2000000
e~tb.o: file not recognized: File format not recognized
collect2.exe: error: ld returned 1 exit status
C:\Tools\GHDL\0.34-dev-mingw64-llvm\bin\ghdl.exe: compilation error
```

tgingold, 2017-03-13:
On 13/03/17 03:01, Patrick Lehmann wrote:
> Hello,
>
> I got a step further in the question on how to deploy GHDL without a
> full MinGW installation (~ 3GiB). The answer is to copy all required dll
> files from MinGW64 into the bin directory of GHDL.
>
> There are no dependencies for |ghdl.exe|.
>
> So far I identified these files for |ghdl1-llvm.exe|:
>
>   * from |/msys64/mingw64/bin/|
>       o |libgcc_s_seh-1.dll|
>       o |libstdc++-6.dll|
>       o |libwinpthread-1.dll|
>       o |zlib1.dll|
>
> These dependencies could be solved by additional Makefile rules for
> target |install| if the platform was MinGW.
>
> The generated executable e.g. |tb.exe| requires:
>
>   * from |/msys64/mingw64/bin/|
>       o |zlib1.dll|
>
> GHDL needs to copy |zlib1.dll| from it's binary directory to the
> simulator output directory. Is there a way to influence the DLL binding
> from within GHDL? I think we discussed such topics while talking about
> VPI dlls. Maybe GHDL could influence the search order if it was build
> with MinGW and add it's own binary directory to the possible DLL folders.
>
> Anyhow, if I run |ghdl -e tb|, I get this:
>
> |D:\Temp\ranges> & $GHDL -a .\ranges.vhdl D:\Temp\ranges> & $GHDL -e -v
> tb C:\Tools\GHDL\0.34-dev-mingw64-llvm\bin/ghdl1-llvm.exe
> -PC:\Tools\GHDL\0.34-dev-mingw64-llvm\lib/ghdl\v93\std\
> -PC:\Tools\GHDL\0.34-dev-mingw64-llvm\lib/ghdl\v93\ieee\ --elab tb -l
> e~tb.lst -c -o e~tb.o e~tb C:\Tools\GNAT\2016\bin\gcc.exe -o tb e~tb.o
> C:\Tools\GHDL\0.34-dev-mingw64-llvm\lib/ghdl\v93\std\std_standard.o
> ranges.o C:\Tools\GHDL\0.34-dev-mingw64-llvm\lib/ghdl/libgrt.a -ldbghelp
> -L.\ -LC:/msys64/mingw64/lib/gcc/x86_64-w64-mingw32/6.2.0/adalib/ -lz
> -Wl,--stack=0x2000000 e~tb.o: file not recognized: File format not
> recognized collect2.exe: error: ld returned 1 exit status
> C:\Tools\GHDL\0.34-dev-mingw64-llvm\bin\ghdl.exe: compilation error |

I suppose either gcc or ld for x84_64 is missing or not the first in the 
path.
Isn't C:\Tools\GNAT\2016\bin\gcc.exe a 32-bit compiler ?

Paebbels, 2017-03-13:
What does GHDL call?
* gcc
* ld
* collect2
* ...?

I could add a `gcc` directory to GHDL:
```
0.34-dev-mingw64-llvm/
  bin
    ghdl.exe
    ghdl1-llvm.exe
  gcc-6.3.0-mingw64
    ld.exe
    ...
  lib
    ...
```

And add this path to PATH.

tgingold, 2017-03-13:
On 13/03/17 20:38, Patrick Lehmann wrote:
> What does GHDL call?
>
>   * gcc
>   * ld
>   * collect2
>   * ...?

To link, ghdl calls gcc, which calls collect2 and ld.

Paebbels, 2017-03-13:
OK, I think we are reaching a good size ...
I'm down to 110 files in 15 directories (need to remove empty dirs) and 149 MiB.

i think I can bring it to 50 MiB. That would match the embedded MinGW in e.g. Vivado

tgingold, 2017-03-14:
I don't think this should be part of the Makefile.  Adding the minimal mingw environment should be done by a script.

So I think ideally we can produce two zip: one for users with mingw64 and one that is standalone.

Does it make sense ?

Paebbels, 2017-03-14:
I don't know how compatible MinGWs are among each other...

The package manager can show version numbers, but the installed version is always the latest. So if a user doesn't update, he might run a MinGW with GCC 5.4 whereas we ship a GHDL build with 6.3.

E.g. the MinGW on AppVeyor is quite old compared to mine even if I run a complete upgrade on AppVeyor. They seem to use other mirrors or a package cache that uses older packages.

We could reduce the size of the installation package by compiling the VHDL sources on the users machine.

If one has a full MinGW, he wouldn't use a package to copy files into my MinGW. He would probably either expect a MinGW package for pacman or compile GHDL on his own, because he has a MinGW :)

tgingold, 2017-03-14:
On 14/03/17 06:58, Patrick Lehmann wrote:
> I don't know how compatible MinGWs are among each other...

That shouldn't be a problem.  After all, you have the same issue with 
the Ada compiler.

> The package manager can show version numbers, but the installed version
> is always the latest. So if a user doesn't update, he might run a MinGW
> with GCC 5.4 whereas we ship a GHDL build with 6.3.

>
> E.g. the MinGW on AppVeyor is quite old compared to mine even if I run a
> complete upgrade on AppVeyor. They seem to use other mirrors or a
> package cache that uses older packages.

That's fine.

> We could reduce the size of the installation package by compiling the
> VHDL sources on the users machine.

I'd prefer not, as it would slow down the installation.

> If one has a full MinGW, he wouldn't use a package to copy files into my
> MinGW. He would probably either expect a MinGW package for pacman or
> compile GHDL on his own, because he has a MinGW :)

Make sense too.

closed 2017-06-18 by tgingold