ghdl/ghdl #319: How to run ghdl-0.34-dev-ming64-mcode without MinGW64?

Paebbels, 2017-03-13:
Hello,

I got a step further in the question on how to deploy GHDL without a full MinGW installation (~ 3GiB). The answer is to copy all required dll files from MinGW64 into the bin directory of GHDL.

So far I identified these files for `ghdl.exe`:
* from `/msys64/mingw64/bin/`
  * `zlib1.dll`

These dependencies could be solved by additional Makefile rules for target `install` if the platform was MinGW.

Anyhow, running GHDL resulted in an exception:
```
D:\Temp\ranges> & $GHDL -e tb
D:\Temp\ranges> & $GHDL -r tb

******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: GHDL 0.34-dev (2016-09-14-176-g6abb2ed-dirty) [Dunoon edition]
Compiled with GNAT Version: 6.2.0
In directory: D:\Temp\ranges\
Command line:
C:\Tools\GHDL\0.34-dev-mingw64-mcode\bin\ghdl.exe -r tb
Exception PROGRAM_ERROR raised
Exception information:
raised PROGRAM_ERROR : EXCEPTION_ACCESS_VIOLATION
Call stack traceback locations:
0x6dd3cb 0x7704812b 0x7703855d 0x7706bcb6 0x402f92 0x474caf 0x643b20 0x64a039 0x648b11 0x4da41e 0x402b15 0x4013f6 0x401519 0x76f159cb 0x7704a55f
******************************************************************
```

tgingold, 2017-03-13:
On 13/03/17 05:06, Patrick Lehmann wrote:
> Hello,
>
> I got a step further in the question on how to deploy GHDL without a
> full MinGW installation (~ 3GiB). The answer is to copy all required dll
> files from MinGW64 into the bin directory of GHDL.
>
> So far I identified these files for |ghdl.exe|:
>
>   * from |/msys64/mingw64/bin/|
>       o |zlib1.dll|
>
> These dependencies could be solved by additional Makefile rules for
> target |install| if the platform was MinGW.

mcode for win64 is not supported!

Tristan.

Paebbels, 2017-03-13:
Oh sorry, I should remove that one from my build triggers.

closed 2017-03-13 by Paebbels