ghdl/ghdl #322: GHDL Bug Occurred

gardners, 2017-03-23:
Received the following output from "make simulate" using https://github.com/gardners/gs4502b/commit/8aac928aa9d20be5ebcf2867b7c9bba58cfd6b53

It was working fine with the previous commit, https://github.com/gardners/gs4502b/commit/23efee978af7f17f6f91c54dde567b97103680df

This is on a Mac running OSX 10.12.3.

I'll try further narrowing down exactly which change in the source has triggered the bug.

I'll also try updating GHDL to the latest version as soon as I have time to rebuild it, but if there is anything I can do using the current version to further investigate this, please let me know.  

> Pauls-MBP-3:newcpu gardners$ make simulate
> /usr/local/ghdl-0.34/bin/ghdl -i cpu_test.vhdl debugtools.vhdl alu.vhdl instructions.vhdl instruction_lengths.vhdl addressing_modes.vhdl instruction_equations.vhdl extra_instruction_equations.vhdl ram0.vhdl ram1.vhdl ram2.vhdl ram3.vhdl address_translator.vhdl gs4502b.vhdl memory_controller.vhdl gs4502b_core.vhdl gs4502b_instruction_prefetch.vhdl gs4502b_stage_decode.vhdl gs4502b_stage_validate.vhdl gs4502b_stage_execute.vhdl disassemble.vhdl visualise.vhdl
> /usr/local/ghdl-0.34/bin/ghdl -m cpu_test
> ./cpu_test || /usr/local/ghdl-0.34/bin/ghdl -r cpu_test
> /bin/sh: ./cpu_test: No such file or directory
> 
> ******************** GHDL Bug occured ****************************
> Please report this bug on https://github.com/tgingold/ghdl/issues
> GHDL release: GHDL 0.34dev (20151126) [Dunoon edition]
> Compiled with GNAT Version: GPL 2015 (20150428-49)
> In directory: /Users/gardners/Projects/c64accel/git/newcpu/
> Command line:
> /usr/local/ghdl-0.34/bin/ghdl -r cpu_test
> Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
> Exception information:
> Exception name: SYSTEM.ASSERTIONS.ASSERT_FAILURE
> Message: trans-chap6.adb:916
> Load address: 0x10c209000
> Call stack traceback locations:
> 0x10c4d5c71 0x10c3d137d 0x10c3d1e4b 0x10c3d5dee 0x10c3d5ec4 0x10c3d6257 0x10c3d6504 0x10c3d6727 0x10c3fdb65 0x10c3fa4a0 0x10c3fadb4 0x10c43eaf6 0x10c3e975e 0x10c43cf6c 0x10c3f4010 0x10c44194b 0x10c448433 0x10c4473a2 0x10c2dec5c 0x10c20b222
> ******************************************************************
> make: *** [simulate] Error 2

gardners, 2017-03-23:
Okay, I have narrowed it down to the following statements in memory_controler.vhdl:

+
+  constant ram0name : string :=  entity_name & ".ram0";
+  constant ram1name : string :=  entity_name & ".ram1";
+  constant ram2name : string :=  entity_name & ".ram2";
+  constant ram3name : string :=  entity_name & ".ram3";

I am passing in an entity name, that I append a suffix to, to create the entity name for child entities, so that in my visualisation tool, I can tell which part I am referring to.  It seems that this is the problem.

I guess I will try making entity_name a generic instead of a port.

Paebbels, 2017-03-23:
(You can update your original question by using the pencil icon.)

Is it possible to use a newer version of GHDL or compile it locally with e.g. LLVM on your Mac machine? The linked GHDL version (Git submodule) is over one year old.

gardners, 2017-03-23:
I will indeed try a newer version

gardners, 2017-04-05:
Problem persists with head of master branch of ghdl:

> Pauls-MBP-3:newcpu gardners$ make simulate
> /usr/local/ghdl-0.34/bin/ghdl -i cpu_test.vhdl debugtools.vhdl alu.vhdl instructions.vhdl instruction_lengths.vhdl addressing_modes.vhdl instruction_equations.vhdl extra_instruction_equations.vhdl ram0.vhdl ram1.vhdl ram2.vhdl ram3.vhdl address_translator.vhdl gs4502b.vhdl memory_controller.vhdl gs4502b_core.vhdl gs4502b_instruction_prefetch.vhdl gs4502b_stage_decode.vhdl gs4502b_stage_validate.vhdl gs4502b_stage_execute.vhdl disassemble.vhdl visualise.vhdl
> /usr/local/ghdl-0.34/bin/ghdl -m cpu_test
> ./cpu_test || /usr/local/ghdl-0.34/bin/ghdl -r cpu_test
> /bin/sh: ./cpu_test: No such file or directory
> 
> ******************** GHDL Bug occured ****************************
> Please report this bug on https://github.com/tgingold/ghdl/issues
> GHDL release: GHDL 0.34dev (20151126) [Dunoon edition]
> Compiled with GNAT Version: GPL 2015 (20150428-49)
> In directory: /Users/gardners/Projects/c64accel/git/newcpu/
> Command line:
> /usr/local/ghdl-0.34/bin/ghdl -r cpu_test
> Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
> Exception information:
> Exception name: SYSTEM.ASSERTIONS.ASSERT_FAILURE
> Message: trans-chap6.adb:916
> Load address: 0x106e3b000
> Call stack traceback locations:
> 0x107107c71 0x10700337d 0x107003e4b 0x107007dee 0x107007ec4 0x107008257 0x107008504 0x107008727 0x10702fb65 0x10702c4a0 0x10702cdb4 0x107070af6 0x10701b75e 0x10706ef6c 0x107026010 0x10707394b 0x10707a433 0x1070793a2 0x106f10c5c 0x106e3d222
> ******************************************************************
> make: *** [simulate] Error 2
> Pauls-MBP-3:newcpu gardners$ git log | head -1
> commit 8aac928aa9d20be5ebcf2867b7c9bba58cfd6b53

gardners, 2017-04-05:
Oops.. That was still using the old GHDL binary. Here it is with the new one:

> Pauls-MBP-3:newcpu gardners$ make simulate
> /usr/local/bin/ghdl -i cpu_test.vhdl debugtools.vhdl alu.vhdl instructions.vhdl instruction_lengths.vhdl addressing_modes.vhdl instruction_equations.vhdl extra_instruction_equations.vhdl ram0.vhdl ram1.vhdl ram2.vhdl ram3.vhdl address_translator.vhdl gs4502b.vhdl memory_controller.vhdl gs4502b_core.vhdl gs4502b_instruction_prefetch.vhdl gs4502b_stage_decode.vhdl gs4502b_stage_validate.vhdl gs4502b_stage_execute.vhdl disassemble.vhdl visualise.vhdl
> /usr/local/bin/ghdl -m cpu_test
> ./cpu_test || /usr/local/bin/ghdl -r cpu_test
> /bin/sh: ./cpu_test: No such file or directory
> 
> ******************** GHDL Bug occured ****************************
> Please report this bug on https://github.com/tgingold/ghdl/issues
> GHDL release: GHDL 0.34-dev (tarball) [Dunoon edition]
> Compiled with GNAT Version: GPL 2016 (20160515-49)
> Target: x86_64-apple-darwin14.5.0
> In directory: /Users/gardners/Projects/c64accel/git/newcpu/
> Command line:
> /usr/local/bin/ghdl -r cpu_test
> Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
> Exception information:
> raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : trans-chap6.adb:970
> Load address: 0x1040ab000
> Call stack traceback locations:
> 0x10439d131 0x104298ae1 0x104299549 0x10429cfa4 0x10429d08b 0x10429d462 0x10429d789 0x10429e29e 0x1042c5348 0x1042c1a7e 0x1042c246c 0x10430ad2b 0x1042b1339 0x104309258 0x1042bbb44 0x10430dd47 0x1043157db 0x1043144c3 0x10418cd9d 0x1040adb4e
> ******************************************************************
> make: *** [simulate] Error 2

This is ghdl commit 7eb4e6f322074102bc1b038fdf8c5bea1beb88ff

For my project, this newer version of GHDL did identify some sub-type mismatches for coreid, which I have fixed in commit 4dfdfe96152d2235d4231213877bc4bf10271fba, but with the error still occurring as above.

gardners, 2017-04-05:
Work around found: I have moved the entity_name elements from ports to generics
(see commit 8d12d8d644ab958a4358d33f1ca88d83402cb534)

This then works.

However, it is not clear if it will still be synthesisable, as I have read that most synthesis tools don't like non-integer generics.  Thus, it would still be nice to have GHDL accept the concatenation operator on the right hand side of port map expressions, where the strings are constant.

Paebbels, 2017-04-05:
@gardners, please review your commit hashes. You need to prefix your hashes if the are part of another repository. [see GitHub auto linking](https://help.github.com/articles/autolinked-references-and-urls/) for details.

gardners, 2017-04-05:
Sorry about that -- commit references fixed.

tgingold, 2017-04-05:
Using the master version, I get:
gs4502b_core.vhdl:192:17:error: bounds or direction of actual don't match with port "coreid"

(which looks correct).

gardners, 2017-04-05:
As described above, I encountered and fixed that when I moved to using the head of master for GHDL in https://github.com/gardners/gs4502b/commit/4dfdfe96152d2235d4231213877bc4bf10271fba.  With that fixed, the GHDL error persists:

> Pauls-MBP-3:newcpu gardners$ git checkout 4dfdfe96152d2235d4231213877bc4bf10271fba
> make simulateM	ghdl
> Note: checking out '4dfdfe96152d2235d4231213877bc4bf10271fba'.
> 
> You are in 'detached HEAD' state. You can look around, make experimental
> changes and commit them, and you can discard any commits you make in this
> state without impacting any branches by performing another checkout.
> 
> If you want to create a new branch to retain commits you create, you may
> do so (now or later) by using -b with the checkout command again. Example:
> 
>   git checkout -b <new-branch-name>
> 
> HEAD is now at 4dfdfe9... make coreid type consistent
> Pauls-MBP-3:newcpu gardners$ make simulate
> gcc -g -Wall -o makeram makeram.c
> ./makeram mega65mem.bin
> /usr/local/bin/ghdl -i cpu_test.vhdl debugtools.vhdl alu.vhdl instructions.vhdl instruction_lengths.vhdl addressing_modes.vhdl instruction_equations.vhdl extra_instruction_equations.vhdl ram0.vhdl ram1.vhdl ram2.vhdl ram3.vhdl address_translator.vhdl gs4502b.vhdl memory_controller.vhdl gs4502b_core.vhdl gs4502b_instruction_prefetch.vhdl gs4502b_stage_decode.vhdl gs4502b_stage_validate.vhdl gs4502b_stage_execute.vhdl disassemble.vhdl visualise.vhdl
> /usr/local/bin/ghdl -m cpu_test
> ./cpu_test || /usr/local/bin/ghdl -r cpu_test
> /bin/sh: ./cpu_test: No such file or directory
> 
> ******************** GHDL Bug occured ****************************
> Please report this bug on https://github.com/tgingold/ghdl/issues
> GHDL release: GHDL 0.34-dev (tarball) [Dunoon edition]
> Compiled with GNAT Version: GPL 2016 (20160515-49)
> Target: x86_64-apple-darwin14.5.0
> In directory: /Users/gardners/Projects/c64accel/git/newcpu/
> Command line:
> /usr/local/bin/ghdl -r cpu_test
> Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
> Exception information:
> raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : trans-chap6.adb:970
> Load address: 0x10ffb5000
> Call stack traceback locations:
> 0x1102a7131 0x1101a2ae1 0x1101a3549 0x1101a6fa4 0x1101a708b 0x1101a7462 0x1101a7789 0x1101a829e 0x1101cf348 0x1101cba7e 0x1101cc46c 0x110214d2b 0x1101bb339 0x110213258 0x1101c5b44 0x110217d47 0x11021f7db 0x11021e4c3 0x110096d9d 0x10ffb7b4e
> ******************************************************************
> make: *** [simulate] Error 2
> Pauls-MBP-3:newcpu gardners$

tgingold, 2017-04-06:
On 05/04/17 23:41, gardners wrote:
> As described above, I encountered and fixed that when I moved to using
> the head of master for GHDL in gardners/gs4502b@4dfdfe9
> <https://github.com/gardners/gs4502b/commit/4dfdfe96152d2235d4231213877bc4bf10271fba>.
> With that fixed, the GHDL error persists:

With gs4502b head, I get:

disassemble.vhdl:105:18:error: no declaration for "get_addressing_modes"

Where is it declared ?

Tristan.

gardners, 2017-04-06:
In addressing_modes.vhdl, which is auto-generated by
addressingmodeequations.c.  If you type "make simulate", it will do all the
autogeneration steps.

Paul.

tgingold, 2017-04-06:
> In addressing_modes.vhdl, which is auto-generated by
> addressingmodeequations.c. If you type "make simulate", it will do all the
> autogeneration steps.

addressing_modes.vhdl declares:
   function addressing_modes(opcode : std_logic_vector(8 downto 0)) 
return addressing_mode;

but it doesn't declare get_addressing_modes.

$ make GHDL=ghdl_mcode  simulate
ghdl_mcode -i cpu_test.vhdl debugtools.vhdl alu.vhdl instructions.vhdl 
instruction_lengths.vhdl addressing_modes.vhdl 
instruction_equations.vhdl extra_instruction_equations.vhdl ram0.vhdl 
ram1.vhdl ram2.vhdl ram3.vhdl address_translator.vhdl gs4502b.vhdl 
memory_controller.vhdl gs4502b_core.vhdl 
gs4502b_instruction_prefetch.vhdl gs4502b_stage_decode.vhdl 
gs4502b_stage_validate.vhdl gs4502b_stage_execute.vhdl disassemble.vhdl 
visualise.vhdl
ghdl_mcode -m cpu_test
disassemble.vhdl:105:18:error: no declaration for "get_addressing_modes"
make: *** [cpu_test] Error 1

Tristan.

gardners, 2017-04-06:
Hello,

Ah, that's a mystery how I ever had the correctly named function generated
in my build directory.  I have fixed the naming of the function in master:

https://github.com/gardners/gs4502b/commit/
02f01eb56e090c12b8abcf704e6d5b852a99923a

Sorry for the confusion,

Paul.

tgingold, 2017-04-06:
On 06/04/17 20:45, gardners wrote:
> Hello,
>
> Ah, that's a mystery how I ever had the correctly named function generated
> in my build directory. I have fixed the naming of the function in master:
>
> https://github.com/gardners/gs4502b/commit/
> 02f01eb56e090c12b8abcf704e6d5b852a99923a

Ok, it now compiles and runs but doesn't crash.

Maybe it's a Mac OSX vs Linux issue ?

Should I try with gnat gpl 2014 ?

Tristan.

gardners, 2017-04-12:
I just tried cherry-picking the fixed function name onto the commit where it was failing, and now I can't reproduce it either using the latest GHDL.  I think its best to just close the issue, and if it recurs, we can worry about it then.  Sorry for the inconvenience.

Paul.

closed 2017-04-12 by gardners