ghdl/ghdl #323: [RFE] support LLVM 4.x

ignatenkobrain, 2017-03-24:
At this point, there is 4.0.0 and we are updating to it in Fedora. Would be cool if ghdl would support it.

1138-4EB, 2017-03-24:
Did you try it or just supposing that it is not working? I don't mean to be rude. Just to try it, in case you didn't do it yet.

ignatenkobrain, 2017-03-24:
@1138-4EB it definitely doesn't work, it shows "llvm-4.0.0 is not recognized" or something very similar.

1138-4EB, 2017-03-24:
Please, if you don't mind, post the log or, at least, the exact commands you are using. Are you running the script? Or writing configure and make commands by hand?

It may be just a problem with the script.

1138-4EB, 2017-03-25:
Well, I investigated a bit, and this is where I got stuck:

Inside a `fedora:latest` docker container (`docker run --rm -it fedora:latest bash`):

```
# dnf install -y dnf-plugins-core gcc-gnat git
# dnf copr enable -y alonid/llvm-4.0.0
# dnf install -y llvm-4.0.0-devel clang-4.0.0 lldb-4.0.0 compiler-rt-4.0.0
# export PATH=$PATH:/opt/llvm-4.0.0/bin
# git clone https://github.com/tgingold/ghdl.git
# cd ghdl
```
Edit `configure` and comment `exit 1` in line 170. Put these two lines instead:

```
llvm_be=llvm-nodebug
llvm_be_ver=39
```
Then:

```
# ./dist/linux/buildtest.sh -b llvm -f dummy.tgz
[| GHDL] Prepare /ghdl 
[| GHDL] Environment 
[| GHDL - build] Configure 
[| GHDL - build] Make 
# cat log.log
FBR=f25
HOSTNAME=09370e022750
TERM=xterm
LS_COLORS=rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=01;05;37;41:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.Z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.jpg=01;35:*.jpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=01;36:*.au=01;36:*.flac=01;36:*.m4a=01;36:*.mid=01;36:*.midi=01;36:*.mka=01;36:*.mp3=01;36:*.mpc=01;36:*.ogg=01;36:*.ra=01;36:*.wav=01;36:*.oga=01;36:*.opus=01;36:*.spx=01;36:*.xspf=01;36:
PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/opt/llvm-4.0.0/bin
PWD=/ghdl/build-llvm
FGC=f25
HOME=/root
SHLVL=2
DISTTAG=f25docker
_=/usr/bin/env
OLDPWD=/ghdl
Build machine is: x86_64-redhat-linux
Unhandled version llvm 4.0.0
create pic/ subdirectory
Creating ghdl.gpr
Creating Makefile
Generate default_pathes.ads
if [ -d ../.git ]; then \
  desc=`cd ..; git describe --dirty`; \
  sed -e "s/[(].*[)]/($desc)/" < ../src/version.in > version.tmp; \
else \
  cp ../src/version.in version.tmp; \
fi
if [ ! -r version.ads ] || ! cmp version.tmp version.ads > /dev/null; then cp version.tmp version.ads; fi
make -f ../src/ortho/llvm-nodebug/Makefile \
 ortho_srcdir=../src/ortho ortho_exec=ghdl1-llvm \
 GNAT_FLAGS="-aI../src -aI../src/vhdl -aI../src/psl -aI../src/vhdl/translate -aI../src/ghdldrv -aI../src/grt -aI../src/ortho -aI../src/ortho/llvm-nodebug -gnaty3befhkmr -gnatwae -gnatf -gnat05 -g -gnata" LDFLAGS="" \
 LLVM_CONFIG="llvm-config" CXX="clang++" \
 llvm_be_ver="39" all
make[1]: Entering directory '/ghdl/build-llvm'
clang++ -c `llvm-config --cxxflags` -o llvm-cbindings.o ../src/ortho/llvm-nodebug/llvm-cbindings.cpp
clang-4.0: warning: argument unused during compilation: '-specs=/usr/lib/rpm/redhat/redhat-hardened-cc1' [-Wunused-command-line-argument]
warning: unknown warning option '-Wno-maybe-uninitialized'; did you mean '-Wno-uninitialized'? [-Wunknown-warning-option]
1 warning generated.
echo "with Ortho_Code_Main39;" > ortho_code_main.tmp
echo "procedure Ortho_Code_Main is" >> ortho_code_main.tmp
echo "begin" >> ortho_code_main.tmp
echo "   Ortho_Code_Main39;" >> ortho_code_main.tmp
echo "end Ortho_Code_Main;" >> ortho_code_main.tmp
mv ortho_code_main.tmp ortho_code_main.adb
gnatmake -o ghdl1-llvm -aI../src/ortho/llvm-nodebug -aI../src/ortho \
-aI../src -aI../src/vhdl -aI../src/psl -aI../src/vhdl/translate -aI../src/ghdldrv -aI../src/grt -aI../src/ortho -aI../src/ortho/llvm-nodebug -gnaty3befhkmr -gnatwae -gnatf -gnat05 -g -gnata ortho_code_main -bargs -E \
[... 92 lines cut ...]
gcc -c -I./ -I../src/ortho/llvm-nodebug -I../src/ortho -I../src -I../src/vhdl -I../src/psl -I../src/vhdl/translate -I../src/ghdldrv -I../src/grt -I../src/ortho -I../src/ortho/llvm-nodebug -gnaty3befhkmr -gnatwae -gnatf -gnat05 -g -gnata -I- /ghdl/src/vhdl/translate/trans-chap9.adb
gcc -c -I./ -I../src/ortho/llvm-nodebug -I../src/ortho -I../src -I../src/vhdl -I../src/psl -I../src/vhdl/translate -I../src/ghdldrv -I../src/grt -I../src/ortho -I../src/ortho/llvm-nodebug -gnaty3befhkmr -gnatwae -gnatf -gnat05 -g -gnata -I- /ghdl/src/vhdl/translate/trans-chap8.adb
gcc -c -I./ -I../src/ortho/llvm-nodebug -I../src/ortho -I../src -I../src/vhdl -I../src/psl -I../src/vhdl/translate -I../src/ghdldrv -I../src/grt -I../src/ortho -I../src/ortho/llvm-nodebug -gnaty3befhkmr -gnatwae -gnatf -gnat05 -g -gnata -I- /ghdl/src/vhdl/translate/trans-chap14.adb
gcc -c -I./ -I../src/ortho/llvm-nodebug -I../src/ortho -I../src -I../src/vhdl -I../src/psl -I../src/vhdl/translate -I../src/ghdldrv -I../src/grt -I../src/ortho -I../src/ortho/llvm-nodebug -gnaty3befhkmr -gnatwae -gnatf -gnat05 -g -gnata -I- /ghdl/src/vhdl/translate/trans-foreach_non_composite.adb
[... cut ...]
/ghdl/src/ortho/llvm-nodebug/ortho_llvm.adb:1680: undefined reference to `LLVMAddFunctionAttr'
clang-4.0: error: linker command failed with exit code 1 (use -v to see invocation)
gnatlink: error when calling /opt/llvm-4.0.0/bin/clang++
gnatmake: *** link failed.
../src/ortho/llvm-nodebug/Makefile:12: recipe for target 'ghdl1-llvm' failed
make[1]: *** [ghdl1-llvm] Error 4
make[1]: Leaving directory '/ghdl/build-llvm'
Makefile:268: recipe for target 'ghdl1-llvm' failed
make: *** [ghdl1-llvm] Error 2
```

Paebbels, 2017-03-25:
According to the [LLVM 4.0.0 change](http://releases.llvm.org/4.0.0/docs/ReleaseNotes.html) log:
* The C API functions `LLVMAddFunctionAttr`, `LLVMGetFunctionAttr`, `LLVMRemoveFunctionAttr`, `LLVMAddAttribute`, `LLVMRemoveAttribute`, `LLVMGetAttribute`, `LLVMAddInstrAttribute` and `LLVMRemoveInstrAttribute` have been removed.
* The C API enum `LLVMAttribute` has been deleted.

tgingold, 2017-03-27:
On 25/03/17 23:37, Patrick Lehmann wrote:
> According to the LLVM 4.0.0 change
> <http://releases.llvm.org/4.0.0/docs/ReleaseNotes.html> log:
>
>   * The C API functions |LLVMAddFunctionAttr|, |LLVMGetFunctionAttr|,
>     |LLVMRemoveFunctionAttr|, [...]

Argh.  Now I need to find what it is replaced by.

Paebbels, 2017-03-27:
More information on the changes:

A new **LLVMAttributeRef API** was introduced in LLVM 3.9. The existing API was deprecated in 3.9 and removed in 4.0.  [see LLVM 3.9 Release Notes](http://releases.llvm.org/3.9.0/docs/ReleaseNotes.html)

From https://reviews.llvm.org/D23039:
> The C API enum LLVMAttribute and associated API is deprecated in favor of the new **LLVMAttributeRef** API. The deprecated functions are ....

The review page also contains a diff showing where the old API was changed.

closed 2017-04-04 by tgingold

1138-4EB, 2017-04-04:
@tgingold, shall I modify the default test contaner list to include one with LLMV 4.0? At now we are cheking LLVM-3.5 on Ubuntu 14.04 and LLVM-3.8 on Ubuntu 12.04. On top of that, mcode is being tested on Fedora.

I'd test both mcode and LLVM-4.0 on Fedora, and mcode and LLVM-3.8 on Ubuntu 16.04 (which is LTS). Therefore, Ubuntu 12.04 and 14.04 would be dropped. What do you think?

Paebbels, 2017-04-04:
The old Ubuntu builds are still needed until dependent CIs like PoC are based on such a build :).

tgingold, 2017-04-04:
On 04/04/17 19:13, 1138-4EB wrote:
> @tgingold <https://github.com/tgingold>, shall I modify the default test
> contaner list to include one with LLMV 4.0? At now we are cheking
> LLVM-3.5 on Ubuntu 14.04 and LLVM-3.8 on Ubuntu 12.04. On [...]

Yes, I think it would make sense to test llvm-4.0

BTW, is your doc rewrite work ready to be merged ?

1138-4EB, 2017-04-04:
@Paebbels , which of them? Is it a dependency on the LLVM version or the distribution itself?

@tgingold, it is quite ready, but I haven't properly reviewed yet. I'd prefer to wait a couple of weeks to have it well done. Furthermore, I'd like to do these docker-related changes before, in order to reflect possible changes.

commit e2769e81 (2017-04-04): Preliminary support of LLVM 4

Close #323