ghdl/ghdl #324: ubuntu 16.10 gnatlink-5 fail

kandido, 2017-03-26:
I can not install ghdl compiler on my computer
my platform:
x86_64, ubuntu 16.0, gcc-5 (gcc version 5.4.1 20160929 (Ubuntu 5.4.1-2ubuntu2)), gnat-5 (GNAT GPL 2016 (20160515-49)) , zlib1g-dev (zlib1g_1.2.8.dfsg-2ubuntu5_amd64.deb)
`./configure` ->ok
`make` -> gcc-5 compilation ok
            -> `gnatlink-5 ghdl_jit.ali -o ghdl_mcode -g memsegs_c.o chkstk.o jumps.o times.o grt-cbinding.o grt-cvpi.o fstapi.o lz4.o fastlz.o -ldl -lm -Wl,--version-script=./src/grt/grt.ver -Wl,--export-dynamic`
 -> ERROR: 
../git/ghdl/src/ortho/mcode/ortho_code-x86-abi.adb:775: referencia to `__muldi3' undefine
../git/ghdl/src/ortho/mcode/ortho_code-x86-abi.adb:779: referencia to `__divdi3' undefine
../git/ghdl/src/ortho/mcode/ortho_code-x86-abi.adb:783: reference to `__moddi3' undefine
collect2: error: ld returned 1 exit status
gnatlink-5: error when calling /usr/bin/gcc-5
gnatmake: *** link failed.
Makefile:95: fail 'ghdl_mcode'
make: *** [ghdl_mcode] Error 4

there are some linking problems...why?
Thanks in advance

1138-4EB, 2017-03-26:
GHDL is actually being built with `gcc (Ubuntu 5.4.0-6ubuntu1~16.04.4) 5.4.0 20160609`. I'm rebuilding [ubuntu-mcode](https://hub.docker.com/r/ghdl/ghdl-tools/tags/) to check if the latest docker image is updated to 5.4.1.

Paebbels, 2017-03-26:
Hello @kandido,

the interfaces in GCC 5.x and 6.x changed compared to GCC 4.x. You have to use another `ortho-lang.c` file to compile with a newer GCC version. The files are located in `src/ortho/gcc`.

**Rename:**
* for GCC 5.x:
  1. `ortho-lang.c` to `ortho-lang-4.c`
  2. `ortho-lang-5.c` to `ortho-lang.c`
* for GCC 6.x:
  1. `ortho-lang.c` to `ortho-lang-4.c`
  2. `ortho-lang-6.c` to `ortho-lang.c`

Kind regards
    Patrick

1138-4EB, 2017-03-26:
I just tried `ubuntu:rolling`, which is 16.10 with `gcc (Ubuntu 6.2.0-5ubuntu12) 6.2.0 20161005`, and the default build procedure is working:

```
git clone https://github.com/tgingold/ghdl.git
cd ghdl
./dist/linux/buildtest.sh -b mcode -f package.tgz
```

@kandido, which version of GHDL sources are you using?

kandido, 2017-03-27:
Thanks, i have clone it again and now `make` run without errors

`ghdl --version`
GHDL 0.34-dev (2017-03-01-29-g1814966) [Dunoon edition]
 Compiled with GNAT Version: 5.4.1
 mcode code generator
Written by Tristan Gingold.

closed 2017-03-27 by kandido