ghdl/ghdl #325: Error with ftabstop and options.c

rqou, 2017-03-27:
Compiling GHDL with the GCC backend with GCC version 6.3.0 (using Debian Sid) gives the following error:

```
options.c:1813:2: error: #error Multiple different help strings for ftabstop=:
```

Adding a period at the end of the `-ftabstop=<number>	Distance between tab stops for column reporting` line in `gcc/vhdl/lang.opt` seems to fix it.

tgingold, 2017-03-27:
On 27/03/17 10:31, Robert Ou wrote:
> Compiling GHDL with the GCC backend with GCC version 6.3.0 (using Debian
> Sid) gives the following error:
>
> |options.c:1813:2: error: #error Multiple different help strings for
> ftabstop=: |
>
> Adding a [...]

I think it would be better to remove the help line in lang.opt
Adding a dot would make it incompatible with previous versions of gcc :-(

tgingold, 2017-03-29:
Should be fixed.  Please, reopen if the issue is still present.

closed 2017-03-29 by tgingold