ghdl/ghdl #326: VHDL2008: output port in sensitivity list

pacalet, 2017-03-27:
VHDL 2008 allows reading an output port but GHDL seems to have a restricted interpretation of this:
```
cat $ foo.vhd
-- File foo.vhd
entity foo is
	port(a, b: out bit);
end entity foo;
architecture bug of foo is
begin
	a <= '0', '1' after 1 ns, '0' after 2 ns, '1' after 3 ns;
	process(a)
	begin
		b <= not a;
	end process;
end architecture bug;
$ ghdl -v
GHDL 0.34-dev (2017-03-01-29-g1814966) [Dunoon edition]
 Compiled with GNAT Version: 4.9.2
 llvm code generator
$ ghdl -a --std=08 hdl/ghdl_bugs/foo.vhd 
hdl/ghdl_bugs/foo.vhd:7:16:error: 'a' of mode out can't be in a sensivity list
ghdl:error: compilation error
```
While:
```
cat $ bar.vhd
-- File bar.vhd
entity bar is
	port(a, b: out bit);
end entity bar;
architecture gub of bar is
begin
	a <= '0', '1' after 1 ns, '0' after 2 ns, '1' after 3 ns;
	process(all)
	begin
		b <= not a;
	end process;
end architecture gub;
$ ghdl -v
GHDL 0.34-dev (2017-03-01-29-g1814966) [Dunoon edition]
 Compiled with GNAT Version: 4.9.2
 llvm code generator
$ ghdl -a --std=08 hdl/ghdl_bugs/bar.vhd
```
I tried to clear this with the LRM but I have no definite answer about whether output ports shall be allowed in sensitivity lists or not. But whatever the decision on this, isn't it a bit strange that a compilation error is raised on the first code and not on the second?

tgingold, 2017-03-27:
On 27/03/17 13:02, pacalet wrote:
> I tried to clear this with the LRM but I have no definite answer about
> whether output ports shall be allowed in sensitivity lists or not. But
> whatever the decision on this, isn't it a bit strange that a compilation
> error is raised on the first code and not on the second?

Dear R. Pacalet,

this issue is now fixed.  The check is now relaxed for vhdl 2008.
It works for process(all) because the analyzer automatically extract the 
sensitivity list without checking whether is it readable or not.

pacalet, 2017-03-28:
Thanks. Did you push?

tgingold, 2017-03-28:
On 28/03/17 07:58, pacalet wrote:
> Thanks. Did you push?

Not yet, but as soon as it passes the regression testsuite.

commit 69a6b074 (2017-03-28): vhdl2008: allow out port in sensitivity lists.

Fix #326
closed 2017-03-28 by tgingold