ghdl/ghdl #327: Exception CONSTRAINT_ERROR raised

declanca, 2017-03-30:
******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: GHDL 0.34-dev (2017-03-01-35-g92973eef) [Dunoon edition]
Compiled with GNAT Version: GPL 2016 (20160515-49)
Target: x86_64-apple-darwin14.5.0
In directory: /Users/dc/cyclops/tools/cocotb/examples/endian_swapper/tests/sim_build/
Command line:
/usr/local/bin/ghdl -r endian_swapper_vhdl --vpi=/Users/dc/cyclops/tools/cocotb/build/libs/x86_64/libvpi.so
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : grt-vpi.adb:1195 discriminant check failed
Load address: 0x1027a2000
Call stack traceback locations:
0x102828827 0x1028289be 0x1027c100a 0x1027fbcce 0x1027fc214 0x1027fc428 0x1027a487c 0x1028116ab 0x1029fea7f 0x102a043ac 0x102a02fb3 0x10287b88d 0x1027a44a8
******************************************************************

Paebbels, 2017-03-30:
You haven't provided any example source code to recreate this error for investigation.
How can we reproduce this error?

declanca, 2017-03-31:
[cocotb.zip](https://github.com/tgingold/ghdl/files/885773/cocotb.zip)

Patrick - was hoping it was a known issue possibly resolved by configuration or build versions - source is attached - it is simply the coco-tb initial build.

unzip then:

cd cocotb/examples/endian_swapper/tests
make TOPLEVEL_LANG=vhdl SIM=GHDL

Paebbels, 2017-03-31:
Thank you.
@tgingold has a Cocotb installation and can now test your setup with the mentioned make parameters.

The support for Cocotb is currently not very high ranked, because the project is not going forward.

tgingold, 2017-04-04:
I am closing this issue as it duplicates existing ones (#281)
(But noting the increasing priority).

closed 2017-04-04 by tgingold