ghdl/ghdl #329: Compiler errors include colour escape codes, even when output is not to a tty

gardners, 2017-04-05:
Trying to compile via emacs, for example, causes emacs to see the compiler output like:

> [1mgs4502b.vhdl:78:3:[1;31merror:[0;1m no actual for generic "entity_name"[0m
> [1mgs4502b.vhdl:82:31:[1;31merror:[0;1m no interface for 'entity_name' in association[0m

The [1m, [1;31m and [0m are all parts of escape codes that normally give the error line colour.
This is fine when ghdl is outputting to a shell, but not when being called by something like emacs.

Ideally GHDL should realise when this is happening, and not colour the compiler messages.  It might be as easy as testing whether stdout goes to a tty or not.

Paebbels, 2017-04-05:
You can disable colored outputs with option [`-fno-color-diagnostics`](http://ghdl.readthedocs.io/en/new-documentation-structure/using/InvokingGHDL.html#diagnostics-control).

gardners, 2017-04-05:
Thanks for info on the work-around

tgingold, 2017-04-05:
This is already implemented: if you pipe through 'cat', there are no colours.
So I suppose emacs uses a pseudo-tty, but doesn't interpret ANSI sequences.

Any idea to improve that (using TERM ?) is welcome.

closed 2017-04-05 by tgingold

jgarvin, 2020-01-04:
The no color diagnostics workaround doesn't appear to work:

```
ghdl -a add4.vhd -fno-color-diagnostics
ghdl -e add4 -fno-color-diagnostics
ghdl -r add4 -fno-color-diagnostics

```

Still results in color escape codes on the screen:

```
[1madd4.vhd:3:10:[1;31merror:[0;1m unit "std_logic_signed" not found in library "ieee"[0m
[1madd4.vhd:3:10:[1;31merror:[0;1m  (use --ieee=synopsys for non-standard synopsys packages)[0m
[1madd4.vhd:4:10:[1;31merror:[0;1m unit "std_logic_unsigned" not found in library "ieee"[0m
[1madd4.vhd:4:10:[1;31merror:[0;1m  (use --ieee=synopsys for non-standard synopsys packages)[0m
[1madd4.vhd:14:14:[1;31merror:[0;1m entity 'add4' was not analysed[0m
[1mghdl:[1;31merror:[0;1m cannot find entity or configuration add4[0m
[1mghdl:[1;31merror:[0;1m cannot find entity or configuration add4[0m

```

tgingold, 2020-01-04:
The -fno-color-diagnostics option must appear before the file name or the unit name.  So try:

> ghdl -a -fno-color-diagnostics add4.vhd 
> ghdl -e -fno-color-diagnostics add4