ghdl/ghdl #330: Compilation of Vendor Libraries Powershell Scripts Fail

grilialex, 2017-04-06:
I downloaded the latest 0.34dev zip file  for Windows (although the same happened when I compiled from source). My system is Windows 10. 
I went to an empty folder and started the Powershell script for OSVVM and UVVM compilation (analysis), but both scripts failed.

OSVVM:
```
F:\Users\Work\Perseus\HW\Test2> powershell -executionpolicy bypass -command "e:\Prog\CAD\GHDL\Lib\Vendors\compile-osvvm.ps1 -OSVVM -Output osvvm"
```
```
Vendor directory 'osvvm' already exists.
Compiling library 'osvvm' ...
Analyzing package file 'F:\Users\Work\Frameworks\HW\OSVVM\NamePkg.vhd'
Analyzing package file 'F:\Users\Work\Frameworks\HW\OSVVM\OsvvmGlobalPkg.vhd'
Analyzing package file 'F:\Users\Work\Frameworks\HW\OSVVM\VendorCovApiPkg.vhd'
Analyzing package file 'F:\Users\Work\Frameworks\HW\OSVVM\TranscriptPkg.vhd'
Analyzing package file 'F:\Users\Work\Frameworks\HW\OSVVM\TextUtilPkg.vhd'
Analyzing package file 'F:\Users\Work\Frameworks\HW\OSVVM\AlertLogPkg.vhd'
Analyzing package file 'F:\Users\Work\Frameworks\HW\OSVVM\MessagePkg.vhd'
Analyzing package file 'F:\Users\Work\Frameworks\HW\OSVVM\SortListPkg_int.vhd'
Analyzing package file 'F:\Users\Work\Frameworks\HW\OSVVM\RandomBasePkg.vhd'
Analyzing package file 'F:\Users\Work\Frameworks\HW\OSVVM\RandomPkg.vhd'
Analyzing package file 'F:\Users\Work\Frameworks\HW\OSVVM\CoveragePkg.vhd'
Analyzing package file 'F:\Users\Work\Frameworks\HW\OSVVM\MemoryPkg.vhd'
Analyzing package file 'F:\Users\Work\Frameworks\HW\OSVVM\ScoreboardGenericPkg.vhd'
Analyzing package file 'F:\Users\Work\Frameworks\HW\OSVVM\ScoreboardPkg_slv.vhd'
Analyzing package file 'F:\Users\Work\Frameworks\HW\OSVVM\ScoreboardPkg_int.vhd'
Analyzing package file 'F:\Users\Work\Frameworks\HW\OSVVM\ResolutionPkg.vhd'
Analyzing package file 'F:\Users\Work\Frameworks\HW\OSVVM\TbUtilPkg.vhd'
Analyzing package file 'F:\Users\Work\Frameworks\HW\OSVVM\OsvvmContext.vhd'
cd : Cannot find path 'F:\Users\Work\Perseus\HW\Test2\osvvm\osvvm\osvvm\v08\osvvm' because it does not exist.
At E:\Prog\CAD\GHDL\Lib\Vendors\shared.psm1:298 char:2
+     cd $DestinationDirectory
+     ~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : ObjectNotFound: (F:\Users\Work\...osvvm\v08\osvvm:String) [Set-Location], ItemNotFoundException
    + FullyQualifiedErrorId : PathNotFound,Microsoft.PowerShell.Commands.SetLocationCommand

--------------------------------------------------------------------------------
Compiling OSVVM packages [SUCCESSFUL]
```

A .cf file is generated (is it complete?) on osvvm/osvvm/osvvm/v08.

For UVVM:
```
powershell -executionpolicy bypass -command "e:\Prog\CAD\GHDL\Lib\Vendors\compile-uvvm.ps1 -All -Output uvvm"
```

I have a bunch of errors (so many I have lost the beginning on my window).
It starts like this:
```
Vendor directory 'uvvm' already exists.
Compiling library 'uvvm_util' ...
Analyzing package file 'F:\Users\Work\Frameworks\HW\UVVM\uvvm_util\src\types_pkg.vhd'
  WARNING: F:\Users\Work\Frameworks\HW\UVVM\uvvm_util\src\types_pkg.vhd:132:14:warning: package "types_pkg" does not require a body [-Wbody]
Analyzing package file 'F:\Users\Work\Frameworks\HW\UVVM\uvvm_util\src\adaptations_pkg.vhd'
  WARNING: F:\Users\Work\Frameworks\HW\UVVM\uvvm_util\src\adaptations_pkg.vhd:63:19:warning: type of a shared variable must be a protected type [-Wshared]
  WARNING: F:\Users\Work\Frameworks\HW\UVVM\uvvm_util\src\adaptations_pkg.vhd:200:19:warning: type of a shared variable must be a protected type [-Wshared]
  WARNING: F:\Users\Work\Frameworks\HW\UVVM\uvvm_util\src\adaptations_pkg.vhd:257:14:warning: package "adaptations_pkg" does not require a body [-Wbody]
Analyzing package file 'F:\Users\Work\Frameworks\HW\UVVM\uvvm_util\src\string_methods_pkg.vhd'
Analyzing package file 'F:\Users\Work\Frameworks\HW\UVVM\uvvm_util\src\protected_types_pkg.vhd'
Analyzing package file 'F:\Users\Work\Frameworks\HW\UVVM\uvvm_util\src\hierarchy_linked_list_pkg.vhd'
Analyzing package file 'F:\Users\Work\Frameworks\HW\UVVM\uvvm_util\src\alert_hierarchy_pkg.vhd'
Analyzing package file 'F:\Users\Work\Frameworks\HW\UVVM\uvvm_util\src\license_pkg.vhd'
Analyzing package file 'F:\Users\Work\Frameworks\HW\UVVM\uvvm_util\src\methods_pkg.vhd'
  WARNING: F:\Users\Work\Frameworks\HW\UVVM\uvvm_util\src\methods_pkg.vhd:33:19:warning: type of a shared variable must be a protected type [-Wshared]
  WARNING: F:\Users\Work\Frameworks\HW\UVVM\uvvm_util\src\methods_pkg.vhd:34:19:warning: type of a shared variable must be a protected type [-Wshared]
  WARNING: F:\Users\Work\Frameworks\HW\UVVM\uvvm_util\src\methods_pkg.vhd:35:19:warning: type of a shared variable must be a protected type [-Wshared]
  WARNING: F:\Users\Work\Frameworks\HW\UVVM\uvvm_util\src\methods_pkg.vhd:36:19:warning: type of a shared variable must be a protected type [-Wshared]
  WARNING: F:\Users\Work\Frameworks\HW\UVVM\uvvm_util\src\methods_pkg.vhd:37:19:warning: type of a shared variable must be a protected type [-Wshared]
  WARNING: F:\Users\Work\Frameworks\HW\UVVM\uvvm_util\src\methods_pkg.vhd:38:19:warning: type of a shared variable must be a protected type [-Wshared]
  WARNING: F:\Users\Work\Frameworks\HW\UVVM\uvvm_util\src\methods_pkg.vhd:39:19:warning: type of a shared variable must be a protected type [-Wshared]
[... 40 lines cut ...]
  ERROR: F:\Users\Work\Frameworks\HW\UVVM\uvvm_vvc_framework\src_target_dependent\td_target_support_pkg.vhd:35:52: no declaration for "t_protected_acknowledge_cmd_idx"
  WARNING: F:\Users\Work\Frameworks\HW\UVVM\uvvm_vvc_framework\src_target_dependent\td_target_support_pkg.vhd:35:19:warning: type of a shared variable must be a protected type [-Wshared]
  ERROR: F:\Users\Work\Frameworks\HW\UVVM\uvvm_vvc_framework\src_target_dependent\td_target_support_pkg.vhd:39:40: no declaration for "c_log_scope_width"
  ERROR: F:\Users\Work\Frameworks\HW\UVVM\uvvm_vvc_framework\src_target_dependent\td_target_support_pkg.vhd:41:28: no declaration for "t_channel"
  eval_discrete_type_length: cannot handle IIR_KIND_SIMPLE_NAME (*std_standard*:1:1)
  
  ******************** GHDL Bug occured ****************************
  Please report this bug on https://github.com/tgingold/ghdl/issues
  GHDL release: GHDL 0.34-dev (commit: 2017-03-02;  git branch: paebbels/master';  hash: 78dd267) [Dunoon edition]
  Compiled with GNAT Version: GPL 2016 (20160515-49)
  Target: i686-pc-mingw32
  In directory: F:\Users\Work\Perseus\HW\Test2\uvvm\uvvm\uvvm_util\v08\uvvm\uvvm\uvvm_vvc_framework\v08\uvvm\uvvm\bitvis_vip_axistream\v08\
  Command line:
  E:\Prog\CAD\GHDL\bin\ghdl.exe -a -fexplicit -frelaxed-rules --mb-comments --warn-binding --ieee=synopsys --no-vital-checks --std=08 -Puvvm --work=bitvis_vip_axistream F:\Users\Work\Frameworks\HW\UVVM\uvvm_vvc_framework\src_target_dependent\td_target_support_pkg.vhd
  Exception TYPES.INTERNAL_ERROR raised
  Exception information:
  raised TYPES.INTERNAL_ERROR : errorout.adb:246
  ******************************************************************
```

See full log, attached.

[log.txt](https://github.com/tgingold/ghdl/files/903303/log.txt)

Ilias

Paebbels, 2017-04-06:
Hello @grilialex, which OSVVM and UVVM version do you use?

grilialex, 2017-04-06:
Latest versions downloaded from git. ie. cloned repositories. As of today.

Paebbels, 2017-05-31:
**For UVVM:**
There is still a bug ([#361 - Loading AXILite BFM package from UVVM causes an exception](https://github.com/tgingold/ghdl/issues/361)) in GHDL, when used with UVVM, but the latest GHDL version (build from sources) let's you pre-compile UVVM without errors. Some VIPs from UVVM can already be used.

**For OSVVM:**
Normally, there shouldn't be any osvvm directory nested 2 or three times in another osvvm directory ...

I can't reproduce that error on my local machine. Maybe it's related to executing PowerShell scripts not directly. Please start a real PowerShell session and launch the script. I'll upload an improved version of all scripts in a few days.

grilialex, 2017-06-19:
Retried the updated scripts (with my older ghdl executable) on OSVVM.
It is better but fails on something minor in the end. 

```
PS F:\Users\Work\Tools\ghdl\install\lib\ghdl\vendors> powershell -executionpolicy bypass -command "./compile-osvvm
.ps1 -OSVVM -Output osvvm"
Creating vendor directory: 'osvvm'.
Compiling library 'osvvm' ...
Analyzing package file 'F:\Users\Work\Tools\ghdl\install\lib\ghdl\src\osvvm\NamePkg.vhd'
Analyzing package file 'F:\Users\Work\Tools\ghdl\install\lib\ghdl\src\osvvm\OsvvmGlobalPkg.vhd'
Analyzing package file 'F:\Users\Work\Tools\ghdl\install\lib\ghdl\src\osvvm\VendorCovApiPkg.vhd'
Analyzing package file 'F:\Users\Work\Tools\ghdl\install\lib\ghdl\src\osvvm\TranscriptPkg.vhd'
Analyzing package file 'F:\Users\Work\Tools\ghdl\install\lib\ghdl\src\osvvm\TextUtilPkg.vhd'
Analyzing package file 'F:\Users\Work\Tools\ghdl\install\lib\ghdl\src\osvvm\AlertLogPkg.vhd'
Analyzing package file 'F:\Users\Work\Tools\ghdl\install\lib\ghdl\src\osvvm\MessagePkg.vhd'
Analyzing package file 'F:\Users\Work\Tools\ghdl\install\lib\ghdl\src\osvvm\SortListPkg_int.vhd'
Analyzing package file 'F:\Users\Work\Tools\ghdl\install\lib\ghdl\src\osvvm\RandomBasePkg.vhd'
Analyzing package file 'F:\Users\Work\Tools\ghdl\install\lib\ghdl\src\osvvm\RandomPkg.vhd'
Analyzing package file 'F:\Users\Work\Tools\ghdl\install\lib\ghdl\src\osvvm\CoveragePkg.vhd'
Analyzing package file 'F:\Users\Work\Tools\ghdl\install\lib\ghdl\src\osvvm\MemoryPkg.vhd'
Analyzing package file 'F:\Users\Work\Tools\ghdl\install\lib\ghdl\src\osvvm\ScoreboardGenericPkg.vhd'
Analyzing package file 'F:\Users\Work\Tools\ghdl\install\lib\ghdl\src\osvvm\ScoreboardPkg_slv.vhd'
Analyzing package file 'F:\Users\Work\Tools\ghdl\install\lib\ghdl\src\osvvm\ScoreboardPkg_int.vhd'
Analyzing package file 'F:\Users\Work\Tools\ghdl\install\lib\ghdl\src\osvvm\ResolutionPkg.vhd'
Analyzing package file 'F:\Users\Work\Tools\ghdl\install\lib\ghdl\src\osvvm\TbUtilPkg.vhd'
Analyzing package file 'F:\Users\Work\Tools\ghdl\install\lib\ghdl\src\osvvm\OsvvmContext.vhd'
cd : Cannot find path 'F:\Users\Work\Tools\ghdl\install\lib\ghdl\vendors\osvvm\osvvm\osvvm\v08\osvvm' because it
does not exist.
At F:\Users\Work\Tools\ghdl\install\lib\ghdl\vendors\shared.psm1:298 char:2
+     cd $DestinationDirectory
+     ~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : ObjectNotFound: (F:\Users\...osvvm\v08\osvvm:String) [Set-Location], ItemNotFoundE
   xception
    + FullyQualifiedErrorId : PathNotFound,Microsoft.PowerShell.Commands.SetLocationCommand

--------------------------------------------------------------------------------
Compiling OSVVM packages [SUCCESSFUL]
```

It still creates a series of osvvm folders (osvvm/osvvm/osvvm/v08).
If I create the folder osvvm under v08 it is fully successful (ie. osvvm/osvvm/osvvm/v08/osvvm).
In all cases the osvvm-obj08.cf is generated at osvvm/osvvm/osvvm/v08.

I am running this command on my powershell but because of my security restrictions I need to issue  "powershell -executionpolicy bypass" when running the script.

Paebbels, 2017-06-19:
So you are not allowed to execute `Set-ExecutionPolicy RemoteSigned`?

Please execute the command including the `-Debug` switch:
```PowerShell
./compile-osvvm.ps1 -All -Output osvvm -Debug
```

Please note: the script will already create an `osvvm` directory for the library called `osvvm`. (But that doesn't explain that you see 3 times `osvvm` in your path.)

My output looks like this:
```PowerShell
C:\git\GitHub\Paebbels\ghdl\build [paebbels/master ↑7 +0 ~10 -0 !]> ..\libraries\vendors\compile-osvvm.ps1 -All -Output . -Source ..\..\PoC\lib\osvvm\ -Debug -GHDL C:\Tools\GHDL\0.34-dev-gnatgpl32-mcode\bin\
Vendor directory 'C:\git\GitHub\Paebbels\ghdl\build\.' already exists.
  mkdir C:\git\GitHub\Paebbels\ghdl\build\./osvvm/v08
  cd C:\git\GitHub\Paebbels\ghdl\build\./osvvm/v08
Compiling library 'osvvm' ...
Analyzing package file 'C:\git\GitHub\Paebbels\PoC\lib\osvvm\NamePkg.vhd'
  C:\Tools\GHDL\0.34-dev-gnatgpl32-mcode\bin\ghdl.exe -a -fexplicit -frelaxed-rules --mb-comments --warn-binding --ieee=synopsys --no-vital-checks --std=08 -PC:\git\GitHub\Paebbels\ghdl\build\. --work=osvvm C:\git\GitHub\Paebbels\PoC\lib\osvvm\NamePkg.vhd 2>&1
Analyzing package file 'C:\git\GitHub\Paebbels\PoC\lib\osvvm\OsvvmGlobalPkg.vhd'
  C:\Tools\GHDL\0.34-dev-gnatgpl32-mcode\bin\ghdl.exe -a -fexplicit -frelaxed-rules --mb-comments --warn-binding --ieee=synopsys --no-vital-checks --std=08 -PC:\git\GitHub\Paebbels\ghdl\build\. --work=osvvm C:\git\GitHub\Paebbels\PoC\lib\osvvm\OsvvmGlobalPkg.vhd 2>&1
Analyzing package file 'C:\git\GitHub\Paebbels\PoC\lib\osvvm\VendorCovApiPkg.vhd'
  C:\Tools\GHDL\0.34-dev-gnatgpl32-mcode\bin\ghdl.exe -a -fexplicit -frelaxed-rules --mb-comments --warn-binding --ieee=synopsys --no-vital-checks --std=08 -PC:\git\GitHub\Paebbels\ghdl\build\. --work=osvvm C:\git\GitHub\Paebbels\PoC\lib\osvvm\VendorCovApiPkg.vhd 2>&1
....
....
Analyzing package file 'C:\git\GitHub\Paebbels\PoC\lib\osvvm\OsvvmContext.vhd'
  C:\Tools\GHDL\0.34-dev-gnatgpl32-mcode\bin\ghdl.exe -a -fexplicit -frelaxed-rules --mb-comments --warn-binding --ieee=synopsys --no-vital-checks --std=08 -PC:\git\GitHub\Paebbels\ghdl\build\. --work=osvvm C:\git\GitHub\Paebbels\PoC\lib\osvvm\OsvvmContext.vhd 2>&1
--------------------------------------------------------------------------------
Compiling OSVVM packages [SUCCESSFUL]
C:\git\GitHub\Paebbels\ghdl\build [paebbels/master ↑7 +0 ~10 -0 !]>
```

Paebbels, 2017-09-05:
@grilialex Can you check the latest release for this issue? There have been some big changes to the pre-compile scripts.

tgingold, 2017-10-30:
Closing as there was no follow-up.

closed 2017-10-30 by tgingold