ghdl/ghdl #331: GHDL with musl: libbacktrace, libunwind, pthread...

1138-4EB, 2017-04-07:
At now, GHDL cannot be built on [Alpine Linux](https://alpinelinux.org/), because [musl](https://www.musl-libc.org/) is used instead of [glibc](https://www.gnu.org/software/libc/). Therefore errors about missing `backtrace()` are thrown: https://bugs.alpinelinux.org/issues/5079

@tgingold provided the following workaround:

> [@tgingold] You need to slightly patch src/grt/jumps.c, since it assumes that backtrace() is available on linux. Adding something like && defined (GLIBC) or like && !defined (MUSL) should fix the issue.

`__MUSL__` is not defined, and is likely not to be in the future:

- https://blogs.gentoo.org/blueness/2016/03/31/why-macros-like-__glibc__-and-__uclibc__-are-bad/
- http://www.openwall.com/lists/musl/2013/02/08/9
- http://wiki.musl-libc.org/wiki/FAQ

Nevertheless, I changed [L32 in src/grt/config/jumps.c](https://github.com/tgingold/ghdl/blob/master/src/grt/config/jumps.c#L32) to `#if (defined (__linux__) || defined (__APPLE__)) && defined (__GLIBC__) `. No backtrace errors are not reported now.

---

However, on the one hand, the solution above disables that feature on Alpine Linux. On the other hand, according to:

> [@tgingold]  If LLVM is used without the libbacktrace, backtrace() is useless. libbacktrace is not needed for mcode mcode need backtrace() but it doesn't use libbacktrace().

and 

> - http://www.openwall.com/lists/musl/2015/04/09/3
> - http://www.openwall.com/lists/musl/2015/04/09/4
> Please note: libbacktrace may be overkill if you don't need DWARF parsing at runtime. For example, if you can perform symbol resolution separately as post-processing, and capturing only stack frame addresses at runtime suffices, libunwind may be the tool for the job.  Also, libunwind provides backtrace() as a weak alias to its own unw_backtrace(), so in simple cases it's a drop-in
replacement.

GHDL might be able to use [libunwind](http://www.nongnu.org/libunwind/): http://www.nongnu.org/libunwind/man/libunwind(3).html

@tgingold is it doable? I already installed `libunwind` and `libunwind-dev` in the corresponding docker container. You can checkout this branch to make any quick try: https://github.com/1138-4EB/ghdl/tree/alpinebuild It is a single commit ahead of your master branch. That commit includes the change mentioned above, and disables any other container on travis, so that only alpine is checked in this branch.

---

Apart from backtrace, which is just disabled now, I am getting this error: https://travis-ci.org/1138-4EB/ghdl/builds/219529962

```
-ldl -lm -Wl,--version-script=../src/grt/grt.ver -Wl,--export-dynamic
/usr/lib/gcc/x86_64-alpine-linux-musl/6.2.1/adalib//libgnarl-6.so: undefined reference to `pthread_rwlockattr_setkind_np'
collect2: error: ld returned 1 exit status
gnatlink: error when calling /usr/bin/gcc
gnatmake: *** link failed.
make: *** [Makefile:134: ghdl_mcode] Error 4
```

I found that it is only used in `src/grt/fst/fstapi`:

```
$ find . -type f | xargs grep "pthread"
./src/grt/fst/fstapi.c:#include <pthread.h>
./src/grt/fst/fstapi.c:pthread_mutex_t mutex;
./src/grt/fst/fstapi.c:pthread_t thread;
./src/grt/fst/fstapi.c:pthread_attr_t thread_attr;
./src/grt/fst/fstapi.c: pthread_mutex_init(&xc->mutex, NULL);
./src/grt/fst/fstapi.c: pthread_attr_init(&xc->thread_attr);
./src/grt/fst/fstapi.c: pthread_attr_setdetachstate(&xc->thread_attr, PTHREAD_CREATE_DETACHED);
./src/grt/fst/fstapi.c: pthread_mutex_unlock(&(xc->xc_parent->mutex));
./src/grt/fst/fstapi.c: pthread_mutex_lock(&xc->mutex);
./src/grt/fst/fstapi.c: pthread_mutex_unlock(&xc->mutex);
./src/grt/fst/fstapi.c: pthread_mutex_lock(&xc->mutex);
./src/grt/fst/fstapi.c: pthread_create(&xc->thread, &xc->thread_attr, fstWriterFlushContextPrivate1, xc2);
./src/grt/fst/fstapi.c: pthread_mutex_lock(&xc->mutex);
./src/grt/fst/fstapi.c: pthread_mutex_unlock(&xc->mutex);
./src/grt/fst/fstapi.c: pthread_mutex_lock(&xc->mutex);
./src/grt/fst/fstapi.c: pthread_mutex_unlock(&xc->mutex);
./src/grt/fst/fstapi.c: pthread_mutex_lock(&xc->mutex);
./src/grt/fst/fstapi.c: pthread_mutex_unlock(&xc->mutex);
./src/grt/fst/fstapi.c: pthread_mutex_destroy(&xc->mutex);
./src/grt/fst/fstapi.c: pthread_attr_destroy(&xc->thread_attr);
./src/grt/Makefile.inc: GRT_EXTRA_LIB+=-lpthread
```

I added package `libpthread-stubs`, but it makes no difference.

tgingold, 2017-04-07:
Note that Apple doesn't use glibc, so you must write
` #if (defined (__linux__) && defined (__GLIBC__)) || defined (__APPLE__)`

If you need/want to use libunwind, why not.  Maybe simply provide a --with-libunwind option to configure.

There is something very wrong with the link error: the error comes from libgnarl which is provided by the compiler.  Furthermore, libgnarl is needed only when tasking features are used, but this is not the case in ghdl.  So if we understand why libgnarl is incorrectly added to the link and get rid of that, the error should disappear.

During the build, gnatmake creates b~xxx.ads and b~xxx.adb files.  If you can post them, that would be useful (they contain the list of files and libraries to be linked).

1138-4EB, 2017-04-07:
I changed the definition as you told.

This log shows the files created during the build: https://s3.amazonaws.com/archive.travis-ci.org/jobs/219697970/log.txt Those with extension `.ads` or `.adb` are:

```
-rw-r--r--    1 root     root         32923 Apr  7 13:58 b~ghdl_jit.adb
-rw-r--r--    1 root     root         71800 Apr  7 13:58 b~ghdl_jit.ads
-rw-r--r--    1 root     root          1447 Apr  7 13:58 default_pathes.ads
-rw-r--r--    1 root     root            54 Apr  7 13:58 elf_arch.ads
-rw-r--r--    1 root     root            81 Apr  7 13:58 grt-backtraces-impl.ads
-rw-r--r--    1 root     root           102 Apr  7 13:58 ortho_code-x86-flags.ads
-rw-r--r--    1 root     root           157 Apr  7 13:58 version.ads
```

The content of the first two files is:

```
# cat b~ghdl_jit.adb
pragma Ada_95;
pragma Warnings (Off);
pragma Source_File_Name (ada_main, Spec_File_Name => "b~ghdl_jit.ads");
pragma Source_File_Name (ada_main, Body_File_Name => "b~ghdl_jit.adb");
pragma Suppress (Overflow_Check);
with Ada.Exceptions;

package body ada_main is

   E081 : Short_Integer; pragma Import (Ada, E081, "system__os_lib_E");
   E013 : Short_Integer; pragma Import (Ada, E013, "system__soft_links_E");
   E023 : Short_Integer; pragma Import (Ada, E023, "system__exception_table_E");
   E067 : Short_Integer; pragma Import (Ada, E067, "ada__io_exceptions_E");
   E095 : Short_Integer; pragma Import (Ada, E095, "ada__strings_E");
   E101 : Short_Integer; pragma Import (Ada, E101, "ada__strings__maps_E");
   E141 : Short_Integer; pragma Import (Ada, E141, "ada__strings__maps__constants_E");
   E005 : Short_Integer; pragma Import (Ada, E005, "ada__tags_E");
   E066 : Short_Integer; pragma Import (Ada, E066, "ada__streams_E");
   E079 : Short_Integer; pragma Import (Ada, E079, "interfaces__c_E");
   E209 : Short_Integer; pragma Import (Ada, E209, "system__aux_dec_E");
   E025 : Short_Integer; pragma Import (Ada, E025, "system__exceptions_E");
   E084 : Short_Integer; pragma Import (Ada, E084, "system__file_control_block_E");
   E074 : Short_Integer; pragma Import (Ada, E074, "system__file_io_E");
   E077 : Short_Integer; pragma Import (Ada, E077, "system__finalization_root_E");
   E075 : Short_Integer; pragma Import (Ada, E075, "ada__finalization_E");
   E119 : Short_Integer; pragma Import (Ada, E119, "system__storage_pools_E");
   E113 : Short_Integer; pragma Import (Ada, E113, "system__finalization_masters_E");
   E111 : Short_Integer; pragma Import (Ada, E111, "system__storage_pools__subpools_E");
   E416 : Short_Integer; pragma Import (Ada, E416, "system__task_info_E");
   E129 : Short_Integer; pragma Import (Ada, E129, "ada__calendar_E");
   E133 : Short_Integer; pragma Import (Ada, E133, "ada__calendar__time_zones_E");
   E318 : Short_Integer; pragma Import (Ada, E318, "ada__command_line__response_file_E");
   E138 : Short_Integer; pragma Import (Ada, E138, "gnat__directory_operations_E");
   E148 : Short_Integer; pragma Import (Ada, E148, "gnat__secure_hashes_E");
   E150 : Short_Integer; pragma Import (Ada, E150, "gnat__secure_hashes__sha1_E");
   E146 : Short_Integer; pragma Import (Ada, E146, "gnat__sha1_E");
   E157 : Short_Integer; pragma Import (Ada, E157, "system__assertions_E");
   E323 : Short_Integer; pragma Import (Ada, E323, "system__pool_global_E");
   E017 : Short_Integer; pragma Import (Ada, E017, "system__secondary_stack_E");
   E097 : Short_Integer; pragma Import (Ada, E097, "ada__strings__unbounded_E");
   E069 : Short_Integer; pragma Import (Ada, E069, "ada__text_io_E");
   E353 : Short_Integer; pragma Import (Ada, E353, "grt__hooks_E");
   E172 : Short_Integer; pragma Import (Ada, E172, "types_E");
   E242 : Short_Integer; pragma Import (Ada, E242, "algos_E");
   E320 : Short_Integer; pragma Import (Ada, E320, "bug_E");
   E479 : Short_Integer; pragma Import (Ada, E479, "disa_x86_E");
   E135 : Short_Integer; pragma Import (Ada, E135, "dyn_tables_E");
   E494 : Short_Integer; pragma Import (Ada, E494, "elf_common_E");
   E492 : Short_Integer; pragma Import (Ada, E492, "elf32_E");
   E496 : Short_Integer; pragma Import (Ada, E496, "elf64_E");
   E182 : Short_Integer; pragma Import (Ada, E182, "flags_E");
   E343 : Short_Integer; pragma Import (Ada, E343, "foreigns_E");
[... 2646 lines cut ...]
   --  trans.chap3%b
   --  trans.chap2%b
   --  trans.chap12%b
   --  trans.chap1%b
   --  ghdlrun%b
   --  version%s
   --  ghdlmain%b
   --  ghdlcomp%b
   --  bug%b
   --  xrefs%s
   --  xrefs%b
   --  sem_types%b
   --  sem_stmts%b
   --  sem_specs%b
   --  sem_psl%b
   --  sem_names%b
   --  sem_assocs%b
   --  sem_expr%b
   --  sem_decls%b
   --  sem%b
   --  parse%b
   --  ghdlprint%b
   --  END ELABORATION ORDER
end ada_main;
```

tgingold, 2017-04-08:
I fear that the gnat compiler on alpine linux is not fully usable: it requires libgnarl even when not needed, and on top of that the libgnarl is not linkable.

Is gcc 5.x available on alpine linux ?  Otherwise you may have to wait for gcc 7.x.

I close this issue as I cannot make it progress, but feel free to reopen it.

closed 2017-04-08 by tgingold