ghdl/ghdl #332: Generation of .GHW Fails with UVVM on module IRQC

grilialex, 2017-04-09:
Using GHDL (01/03/2017) version, and running the UVVM/IRQC module failed to deliver a .ghw file.
I tried and isolated the problem creating a 4 file testbench.

See: [TestGHDL_SBI.zip](https://github.com/tgingold/ghdl/files/907995/TestGHDL_SBI.zip)

Compile the files:
```
ghdl -a -fexplicit -frelaxed-rules --mb-comments --warn-binding --ieee=synopsys --no-vital-checks --std=08 ilos_sim_pkg.vhd
ghdl -a -fexplicit -frelaxed-rules --mb-comments --warn-binding --ieee=synopsys --no-vital-checks --std=08 irqc_pif_pkg.vhd
ghdl -a -fexplicit -frelaxed-rules --mb-comments --warn-binding --ieee=synopsys --no-vital-checks --std=08 irqc_pif.vhd
ghdl -a -fexplicit -frelaxed-rules --mb-comments --warn-binding --ieee=synopsys --no-vital-checks --std=08 irqc_tb.vhd
ghdl -r -fexplicit -frelaxed-rules --mb-comments --warn-binding --ieee=synopsys --no-vital-checks --std=08 tb_irqc --wave=sim.ghw
rem ghdl -r -fexplicit -frelaxed-rules --mb-comments --warn-binding --ieee=synopsys --no-vital-checks --std=08 tb_irqc --vcd=sim.vcd
```

Provides the following output on my PowerShell:
```
PS F:\Users\Work\Tools\TestGHDL> ./comp.bat

F:\Users\Work\Tools\TestGHDL>ghdl -a -fexplicit -frelaxed-rules --mb-comments --warn-binding --ieee=synopsys --no-vital-checks --std=08 ilos_sim_pkg.vhd
F:\Users\Work\Tools\TestGHDL>ghdl -a -fexplicit -frelaxed-rules --mb-comments --warn-binding --ieee=synopsys --no-vital-checks --std=08 irqc_pif_pkg.vhd
F:\Users\Work\Tools\TestGHDL>ghdl -a -fexplicit -frelaxed-rules --mb-comments --warn-binding --ieee=synopsys --no-vital-checks --std=08 irqc_pif.vhd
F:\Users\Work\Tools\TestGHDL>ghdl -a -fexplicit -frelaxed-rules --mb-comments --warn-binding --ieee=synopsys --no-vital-checks --std=08 irqc_tb.vhd
F:\Users\Work\Tools\TestGHDL>ghdl -r -fexplicit -frelaxed-rules --mb-comments --warn-binding --ieee=synopsys --no-vital-checks --std=08 tb_irqc --wave=sim.ghw
E:\Prog\CAD\Ghdl\Bin\ghdl.exe:internal error: wave.create_type
PS F:\Users\Work\Tools\TestGHDL>
```

The problem occurs from the following Testbench line in file irqc_tb.vhd:
```
signal sbi_if : t_sbi_if(addr(2 downto 0), wdata(7 downto 0), rdata(7 downto 0)) := init_sbi_if_signals(3, 8);
```

t_sbi_if is defined in this test in irqc_pif_pkg.vhd (originally in UVVM is in bitvis_vip_sbi/src/sbi_bfm_pkg.vhd):

```
  type t_sbi_if is record
    cs      : std_logic;          -- to dut
    addr    : unsigned;           -- to dut
    rd      : std_logic;          -- to dut
    wr      : std_logic;          -- to dut
    wdata   : std_logic_vector;   -- to dut
    ready   : std_logic;          -- from dut
    rdata   : std_logic_vector;   -- from dut
  end record;
```

Removing from the testbench this line, .ghw is generated normally.

grilialex, 2017-04-09:
This seems to be an issue with unconstrained unsigned and std_logic_vector types inside the record.
Constraining them on the type .ghw works.

tgingold, 2017-06-09:
Now fixed.  ghw reader updated in gtkwave SVN revision 1185.

closed 2017-06-09 by tgingold