ghdl/ghdl #333: Silent overflow between physical types (64 bit integer) and Natural (32 bit integer)

Brian-Drummond, 2017-04-12:
I came across the following surprise (to me) :  a product of a natural (32 bit integer) and a ratio of two physical quantities (time_in_ms / 1 ms, a 64-bit quantity), exceeding the 32 bit range, is silently truncated to 32 bits before conversion to a real.

Something of the form 
constant CLOCK_FREQ : natural := 30000000;
real(30000000 * 200ms / 1 ms);  -- works
real(CLOCK_FREQ * 200ms / 1 ms); -- fails

Both Modelsim and Synplify have the same issue, which results in silent incorrect behaviour. 

So this may be  a grey area in the LRM where the "natural" constant imposes a narrowing of the type where a "universal integer" literal doesn't, but it would seem possible and very worthwhile to at least warn of the overflow. 

I would tend to prefer a warning. I'm not sure if silently implementing correct behaviour would be a good thing since synthesis can just silently re-introduce the issue. (In fact, given a literal not a constant, Synplify does reintroduce the issue but it warns about the truncation, so I would regard correct behaviour as acceptable, if less convenient than catching the error in simulation)

Simple testcase with permission from customer...

```
entity Time_test is
end Time_test;

architecture Behavioral of Time_test is

constant c_CLOCK_FREQUENCY : natural := 32000000;
constant c_SWITCH_ON_TIME_ms : time := 200 ms; 
constant c_SWITCH_COUNT_CYCLES : integer := integer(real(c_CLOCK_FREQUENCY  * (c_SWITCH_ON_TIME_ms / 1 ms)) / 1000.0);

begin
process is
begin
        report " Switch ON time " & time'image(c_SWITCH_ON_TIME_ms) severity NOTE;        
        report " Switch count " & natural'image(c_SWITCH_COUNT_CYCLES) severity NOTE;
        -- sanity checks on time constraints
        report "Clock frequency =  " & natural'image(c_CLOCK_FREQUENCY) & " Hz" severity NOTE;
        report "Clock period =  " & time'image( 1 sec / c_CLOCK_FREQUENCY) severity NOTE;
        report "Switch period = " & time'image ( 1 sec / c_CLOCK_FREQUENCY * c_SWITCH_COUNT_CYCLES) severity NOTE;
        Assert c_SWITCH_ON_TIME_ms =  1 sec / c_CLOCK_FREQUENCY * c_SWITCH_COUNT_CYCLES 
            report "Wrong Switch ON time = " & time'image ( 1 sec / c_CLOCK_FREQUENCY * c_SWITCH_COUNT_CYCLES) severity ERROR;
        wait;
end process;
end Behavioral;
```

tgingold, 2017-04-15:
On 12/04/17 12:55, Brian-Drummond wrote:
> I came across the following surprise (to me) : a product of a natural
> (32 bit integer) and a ratio of two physical quantities (time_in_ms / 1
> ms, a 64-bit quantity), exceeding the 32 bit range, is [...]

I haven't tests but the mcode backend implements overflow checks.

Brian-Drummond, 2017-04-26:
Thanks for the tip. I haven't tested it yet but I'll update when I can.

Brian-Drummond, 2017-05-15:
Having finally had time to build the mcode ghdl and try it, I can report it doesn't catch the overflow either.

tgingold, 2017-05-17:
Indeed, there is a missing overflow check for multiplication.

commit b8655214 (2017-05-19): Testcase for #333
Close #333
closed 2017-05-19 by tgingold