ghdl/ghdl #335: ghdl:error: overflow detected without explanation

gardners, 2017-04-16:
Using https://github.com/MEGA65/mega65-core/commit/740919fc26011077884c5730b90dd41e8a379106, and from the src/ sub-directory, typing make simulate yields:

ghdl -r cpu_test
...
ghdl:error: overflow detected
ghdl:error: simulation failed

There is no further explanation as to the cause of the error.

tgingold, 2017-04-16:
On 16/04/17 08:04, gardners wrote:
> Using MEGA65/mega65-core@740919f
> <https://github.com/MEGA65/mega65-core/commit/740919fc26011077884c5730b90dd41e8a379106>,
> and from the src/ sub-directory, typing make simulate yields:
>
> ghdl -r cpu_test
> ...
> ghdl:error: overflow detected
> ghdl:error: simulation failed
>
> There is no further explanation as to the cause of the error.

If the backtrace feature is enabled (either with i386 mcode or with 
--with-libbacktrace), you have better indications:

elaborate and simulate (failure expected) univ1
/Users/gingold/devel/ghdl/ghdl_mcode:error: overflow detected
   from: work.univ1(foo).DECL_ELAB at univ1.vhdl:5
   from: Elaboration of design

Tristan.

closed 2017-04-16 by tgingold

gardners, 2017-04-16:
I'm using mcode, but I guess it is 64-bit.  I'll see if I can force it to build 32-bit.

tgingold, 2017-04-16:
On 16/04/17 10:01, gardners wrote:
> I'm using mcode, but I guess it is 64-bit. I'll see if I can force it to
> build 32-bit.

Yes, that should work in 32-bit.
There is still an issue for 64 bits.