ghdl/ghdl #339: Using arrays of unresolved types cause error during simulation

gbd360, 2017-04-30:
[code.zip](https://github.com/tgingold/ghdl/files/966504/code.zip)

translate_array_subtype_element_subtype: cannot handle IIR_KIND_ARRAY_SUBTYPE_DEFINITION (test_pkg.vhd:13:64)

******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: GHDL 0.34-dev (2017-03-01-58-g3031cd3) [Dunoon edition]
Compiled with GNAT Version: 4.9.3
Target: x86_64-linux-gnu
In directory: /home/gdavey/Projects/hfmm/trunk/fpga/simulation/
Command line:
ghdl -e -g -v -fexplicit -frelaxed-rules --warn-no-vital-generic --std=08 --work=work --workdir=./test_work test_bench
Exception TYPES.INTERNAL_ERROR raised
Exception information:
Exception name: TYPES.INTERNAL_ERROR
Message: errorout.adb:246
Call stack traceback locations:
0x4fd8eb 0x63dfb3 0x644b4c 0x64539d 0x62c304 0x632bb8 0x6349ea 0x64c417 0x63562c 0x64e102 0x5fee08 0x6554da 0x60975f 0x659c19 0x6633c8 0x66049d 0x4e2a12 0x409608 0x7fa28465d82e 0x408427 0xfffffffffffffffe
******************************************************************

VHDL built and run using:

BUILD_FLAGS="-g -v -fexplicit -frelaxed-rules --warn-no-vital-generic --std=08"
RUN_FLAGS="--ieee-asserts=disable --wave=out.ghw --stop-time=700ns"
ghdl -a $BUILD_FLAGS --work=work --workdir=$WORK_LIB test_pkg.vhd
ghdl -a $BUILD_FLAGS --work=work --workdir=$WORK_LIB test_bench.vhd
ghdl -e $BUILD_FLAGS --work=work --workdir=$WORK_LIB test_bench
ghdl -r $BUILD_FLAGS --work=work --workdir=$WORK_LIB test_bench $RUN_FLAGS

tgingold, 2017-05-19:
I can reproduce the issue; now investigating.

tgingold, 2017-05-24:
I am marking this issue as feature request as arrays of unbounded arrays are not yet supported.

Paebbels, 2018-12-31:
@tgingold Here is another issue related to unbound arrays. Is this issue solved by the latest updates?

diogratia, 2019-01-01:
```vhdl
        for i in 0 to (samples - 1) loop
            report "i = " & integer'image(i);
            init_array(i)   := to_sfixed(1.0/real(i), ret_type);
        end loop;
```

That's division by 0 for i = 0. Limiting the range to 1 to samples - 1 will eliminate the bound check failure and the function call will complete.

tgingold, 2019-01-02:
Yes indeed.  Not sure why it is reported as a bound check failure however.

closed 2019-01-03 by tgingold

gbd360, 2019-01-03:
I don't want to reopen this issue as I agree it seems to be working.  However, could one of you please check with the "--wave=out.ghw" added to the run flags.  I seem to get a stack dump when doing this.

Also, I think my test code should be update so that line 25 sets s_rst to '0' for completeness.

tgingold, 2019-01-04:
Indeed, there is currently an issue with --wave.  Being fixed.

commit 46813f4e (2019-01-03): Add testcase #339.

Close #339