ghdl/ghdl #340: Aliases and homographs

rqou, 2017-05-05:
I believe the following code should be legal (at least under VHDL-2008):
```vhdl
package test is
    type test is (foo);
    alias test2 is test;
    alias test is test2;
end;
```
However, I instead get an error:
```text
$ ghdl -a --std=08 test.vhd 
test.vhd:4:11:error: identifier "test" already used for a declaration
test.vhd:2:10:error: previous declaration: type "test"
ghdl:error: compilation error
$ ghdl --version
GHDL 0.34-dev (2017-03-01-29-g18149662) [Dunoon edition]
 Compiled with GNAT Version: 5.4.1
 GCC back-end code generator
```

I believe that Principle 1 of IR2099 should apply here ("An alias of a declaration is not a homograph of that declaration"), and that this code should be legal because `test` (the alias) is referring to the same named entity as `test` (the type declaration). Is my understanding correct?

<hr>

In addition, I believe the following code is also legal:
```vhdl
package test is
    type test is (foo);
    alias test is test;
end;
```
The VHDL spec has a note stating that an identifier cannot be referenced in its own declaration and gives as an example the following illegal declaration:
```vhdl
constant K: INTEGER := K*K;
```
However, I believe that this isn't actually true for aliases! From what I understand, the reason that constant declaration is illegal is because
* there must be some K visible at the declaration
* that old K cannot be in the current declarative region or else this new K will be a homograph and thus illegal
* this new K begins hiding the old K at the beginning of the declaration, but
* this new K isn't visible until the end of the declaration

However, I believe what happens with my alias example is:
* there must be a `test` visible at the declaration, but the old `test` will stay visible
* the old `test` and the new `test` will not become homographs because they will end up referring to the same named entity
* the new `test` cannot hide the old `test` because it will not be a homograph
* the new `test` indeed isn't visible until the end of the declaration, but because it cannot hide the old `test`, this does not matter

<hr>

Is my understanding of these two cases correct, or have I missed some critical points?

tgingold, 2017-05-06:
For the first case, I think it is indeed legal for vhdl2008.  The new rule isn't yet implemented.

For the second case, I am less sure.  I have to investigate.

I also have a new case:
```vhdl
entity tc3 is
end tc3;

architecture behav of tc3 is
  constant k : natural := 5;
begin
  b : block
    constant k : natural := k * k;
  begin
    assert k = 25 severity failure;
  end block;
end;
```
Do you think it is valid ?

diogratia, 2017-05-06:
```vhdl
entity tc3 is
end tc3;

architecture behav of tc3 is
  constant k : natural := 5;
begin
  b : block
    constant k : natural := k * k;
  begin
    assert k = 25 severity failure;
  end block;
end;
```
12.2 Scope of declarations, para 2 in part:

>The scope of a declaration, except for an architecture body, extends from the beginning of the declaration to the end of the immediately closing declarative region; the scope of an architecture body extends from the beginning to the end of the architecture body. In either case, this part of the scope of a declaration is called the *immediate scope*.

12.3 Visibility, para 12 in part:

>Whenever a declaration with a certain identifier is visible from a given point, the identifier and the named entity (if any) are also said to be visible from that point.

12.3 Visibility, para 7:

>A declaration is said to be *hidden* within (part of) an inner declarative region if the inner region contains a homograph of this declaration; the outer declaration is then hidden within the immediate scope of the inner homograph. Each of two declarations is said to be a *homograph* of the other if and only if both declarations have the same designator, and they denote different named entities, and either overloading is allowed for at most one of the two, or overloading is allowed for both declarations and they have the same parameter and result type profile (see 4.5.1).

The architecture declarative item constant k is hidden within the scope of the block declarative item declaration of constant k, and

12.3  Visibility, para 3:

>A declaration is visible only within a certain part of its scope; this part starts at the end of the declaration except in the declaration of a design unit other than a PSL verification unit, a package declaration, or a protected type declaration, in which case it starts immediately after the reserved word is occurring after the identifier of the design unit, a package declaration, or protected type declaration. This rule applies to both explicit and implicit declarations.

the inner declaration of k is not visible until the end of it's declaration, which is after the semicolon following the expression supplying the constant's value

6.4.2.2 Constant declarations, notably the BNF:

>constant_declaration ::=  
&nbsp;&nbsp;&nbsp;&nbsp;**constant** identifier_list : subtype_indication [ := expression ] ;

So, no the new case is not valid:

ghdl -a tc3.vhdl  
tc3.vhdl:8:29:error: constant "k" is not visible here  
ghdl:error: compilation error  

and ghdl does the right thing here.

The above references support the note (excerpted in part):

>NOTE 2—The rules defining immediate scope, hiding, and visibility imply that a reference to an identifier, character literal, or operator symbol within its own declaration is illegal (except for design units). The identifier, character literal, or operator symbol hides outer homographs within its immediate scope—that is, from the start of the declaration.

 which is not part of the language definition, See 1.3.4 Front matter, examples, notes, references, and annexes) and both scope and visibility are involved.

The idea here is to not rely on notes as the sole support for a supposition in interpreting semantics. Both uses of the identifier k are embraced in NOTE 2 albeit for different reasons.

rqou, 2017-05-09:
I agree that the new example is not valid. However, I believe that the argument for why that is not valid does not apply to my second example:
```vhdl
package test is
    type test is (foo);
    alias test is test;
end;
```
Because the `test` alias isn't a homograph of the `test` type, it doesn't hide the `test` type. So even though the `test` alias isn't visible yet, the `test` type remains visible and can be used to create test `test` alias. Is there a reason why this interpretation is incorrect?

tgingold, 2017-05-09:
I think the interpretation is correct, but illustrates an issue in the LRM.

The problem is during the analysis of the alias name: an identifier may be hidden or not depending on the result of the analysis.  This self-referencing definition is wrong.

Maybe ghdl should simply ignore:
```vhdl
alias X is X;
```
when X is in the immediate scope.

(This is also a corner case).

What about:
```vhdl
  alias p is my_proc [ integer, p];
```
Do you agree that this is not valid, as the `p` in the signature is hidden by the `p` of the alias designator ?

diogratia, 2017-05-09:
```vhdl
package test is
    type test is (foo);
    alias test is test;
end;
```
6. Declarations  

6.1 General
para 1:
>An alias declaration declares an alternate name for an existing named entity.
para 3:
>For each form of declaration, the language rules define a certain region of text called the *scope* of the declaration (see 12.2). Each form of declaration associates an identifier, operator symbol, or character literal with a named entity. The identifier, operator symbol, or character literal is called the *designator* of the declaration. Only within its scope, there are places where it is possible to use the designator to refer to the associated declared entity; these places are defined by the visibility rules (see 12.3). At such places the designator is said to be a name of the entity; the name is said to denote the associated entity.

12.3 Visibility, para 2:
>For each identifier and at each place in the text, the visibility rules determine a set of declarations (with this identifier) that define the possible meanings of an occurrence of the identifier. A declaration is said to be visible at a given place in the text when, according to the visibility rules, the declaration defines a possible meaning of this occurrence. The following two cases arise in determining the meaning of such a declaration:  
>>— The visibility rules determine at most one possible meaning. In such a case, the visibility rules are sufficient to determine the declaration defining the meaning of the occurrence of the identifier, or in the absence of such a declaration, to determine that the occurrence is not legal at the given point.  
>>— The visibility rules determine more than one possible meaning. In such a case, the occurrence of the identifier is legal at this point if and only if *exactly one* visible declaration is acceptable for the overloading rules in the given context **or all visible declarations denote the same named entity**.

6.6 Alias declarations, paras 3 and 4:
>An *object alias* is an alias whose alias designator denotes an object (i.e., a constant, a variable, a signal, or a file). A *nonobject alias* is an alias whose alias designator denotes some named entity other than an object. An alias can be declared for all named entities except for labels, loop parameters, and generate parameters.  
>  
>The alias designator in an alias declaration denotes the named entity specified by the name and, if present, the signature in the alias declaration. An alias of a signal denotes a signal; an alias of a variable denotes a variable; an alias of a constant denotes a constant; and an alias of a file denotes a file. Similarly, an alias of a subprogram (including an operator) denotes a subprogram, an alias of an enumeration literal denotes an enumeration literal, and so forth.

Where we see both the alias test and the type test are designators that denote the same named entity.

And this tells us Tristan can ignore the alias X is X case, there  is no method of distinguishing the two identical designators denoting the same named entity in a design description.

```vhdl
entity myproc is
end entity;
architecture foo of myproc is
    type p is range 0 to 0;
    procedure my_proc (a: integer; b: p) is
    begin
    end procedure;
    alias p is my_proc [integer, p];
begin 
end architecture;
```

In this case the two designators p do not denote the same named entity.

12.3 Visibility, para 7:

> A declaration is said to be hidden within (part of) an inner declarative region if the inner region contains a homograph of this declaration; the outer declaration is then hidden within the [...]

Paragraph 7 is included for the definition of homograph.

para 10:

>Two declarations that occur immediately within the same declarative region, other than the declarative region of a block implied by a component instantiation or the declarative region of a generic-mapped package or subprogram equivalent to a package instance or a subprogram instance, shall not be homographs, unless exactly one of them is the implicit declaration of a predefined operation or is an implicit alias of such an implicit declaration. In such cases, a predefined operation or alias thereof is always hidden by the other homograph. Where hidden in this manner, an implicit declaration is hidden within the entire scope of the other declaration (regardless of which declaration occurs first); the implicit declaration is visible neither by selection nor directly. For a declarative region of a block implied by a component instantiation or the declarative region of a generic-mapped package or subprogram equivalent to a package instance or a subprogram instance, the rules of this paragraph are applied to the corresponding entity declaration, component declaration, uninstantiated package declaration, or uninstantiated subprogram declaration, as appropriate.

"shall not be" is mandatory, and shall implies an error.