ghdl/ghdl #341: raised CONSTRAINT_ERROR : trans.adb:1357 access check failed

maximouth, 2017-05-07:
I tried to compile this file : 

-- type declaration in a package
library IEEE;
use IEEE.STD_LOGIC_1164.ALL;

package my_type is

  type sensor is
    record
      addr : STD_LOGIC_VECTOR (7 downto 0) ;
      dir  : STD_LOGIC_VECTOR (1 downto 0) ;
    end record ;

  -- sensors state type
  type SE_state is array (31 downto 0) of sensor ;

  --track circuit type
  type TC_St is array (19 downto 0) of STD_LOGIC_VECTOR (7 downto 0) ;
  
end package my_type;

library IEEE;
use IEEE.STD_LOGIC_1164.ALL;

-- Uncomment the following library declaration if using
-- arithmetic functions with Signed or Unsigned values
use IEEE.NUMERIC_STD.ALL;

use work.my_type.all;

entity IXL_tb is
end IXL_tb;

architecture simu of IXL_tb is

  -- synchro   
  signal CLK        :   STD_LOGIC;
  signal reset      :   STD_LOGIC;

  -- input
  signal valid_in   :   STD_LOGIC; 
  signal Sw_Cmd_Req :   STD_LOGIC_VECTOR (7 downto 0);
  signal Sw_State   :   STD_LOGIC_VECTOR (7 downto 0);
  signal Sensor     :   SE_state;

  -- output
  signal valid_out  :   STD_LOGIC;
  signal Sw_Cmd_Aut :   STD_LOGIC_VECTOR (7 downto 0);

  --debug output
  signal TC_out     :   TC_St;

  
begin
  LO : entity work.IXL
    port map (CLK, reset, valid_in, Sw_Cmd_Req, Sw_State, Sensor, valid_out, Sw_Cmd_Aut, TC_out);

  process
  begin

    for i in 0 to X loop
      clk <= not clk;
      wait for 1 us;      
    end loop;

    wait;
  end process;
end simu;

with this command line : 
ghdl -a -v Simul/IXL_tb.vhdl

the non test bench file compile with no error

tgingold, 2017-05-07:
Hello,

can you also post the source of entity (or a complete reproducer) ?  I cannot reproduce the crash.

Paebbels, 2017-05-07:
Hello,

please post the complete error message and your GHDL version (`ghdl -v` => version number, back-end, operating system).

maximouth, 2017-05-07:
IXL_tb.vhdl : 

-- type declaration in a package
library IEEE;
use IEEE.STD_LOGIC_1164.ALL;

package my_type is

  type sensor is
    record
      addr : STD_LOGIC_VECTOR (7 downto 0) ;
      dir  : STD_LOGIC_VECTOR (1 downto 0) ;
    end record ;

  -- sensors state type
  type SE_state is array (31 downto 0) of sensor ;

  --track circuit type
  type TC_St is array (19 downto 0) of STD_LOGIC_VECTOR (7 downto 0) ;
  
end package my_type;

library IEEE;
use IEEE.STD_LOGIC_1164.ALL;

-- Uncomment the following library declaration if using
-- arithmetic functions with Signed or Unsigned values
use IEEE.NUMERIC_STD.ALL;

use work.my_type.all;

entity IXL_tb is
end IXL_tb;

architecture simu of IXL_tb is

  -- synchro   
  signal CLK        :   STD_LOGIC;
  signal reset      :   STD_LOGIC;

  -- input
  signal valid_in   :   STD_LOGIC; 
  signal Sw_Cmd_Req :   STD_LOGIC_VECTOR (7 downto 0);
  signal Sw_State   :   STD_LOGIC_VECTOR (7 downto 0);
  signal Sensor     :   SE_state;

  -- output
  signal valid_out  :   STD_LOGIC;
  signal Sw_Cmd_Aut :   STD_LOGIC_VECTOR (7 downto 0);

  --debug output
  signal TC_out     :   TC_St;

  
begin
  LO : entity work.IXL
    port map (CLK, reset, valid_in, Sw_Cmd_Req, Sw_State, Sensor, valid_out, Sw_Cmd_Aut, TC_out);

  process
  begin

    -- initialiser signaux
    reset <= '1';

    -- bouger de 1 cycle d'horloge 
    for i in 0 to 2 loop
      CLK <= not CLK;
      wait for 1 us;      
    end loop;

[... 186 lines cut ...]
/usr/libexec/gcc/x86_64-fedora_ghdl-linux/4.9.2/ghdl1 -P/usr/lib/gcc/x86_64-fedora_ghdl-linux/4.9.2/vhdl//v93/std/ -P/usr/lib/gcc/x86_64-fedora_ghdl-linux/4.9.2/vhdl//v93/ieee/ -quiet -o IXL_tb.s Simul/IXL_tb.vhdl

******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: GHDL 0.34dev (20151126) [Dunoon edition]
Compiled with GNAT Version: 6.1.1 20160621 (Red Hat 6.1.1-3)
In directory: /home/maximouth/fac/psesi/CBI/Devel/IXL/
Command line:
/usr/libexec/gcc/x86_64-fedora_ghdl-linux/4.9.2/ghdl1 -P/usr/lib/gcc/x86_64-fedora_ghdl-linux/4.9.2/vhdl//v93/std/ -P/usr/lib/gcc/x86_64-fedora_ghdl-linux/4.9.2/vhdl//v93/ieee/ -quiet -o IXL_tb.s Simul/IXL_tb.vhdl
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : trans.adb:1357 access check failed
Call stack traceback locations:
0x5f9d3a 0x61796e 0x617abc 0x61cb35 0x630318 0x5fd0dd 0x600bd7 0x5b3488 0x5b35cb 0x645f11 0x64a58f 0x63f33c 0x639f03 0x63bd22 0x63c62e 0x664a7b 0x600a33 0x632189 0x57fddd 0x55c561 0x9577ef 0x95957e 0x57fe39 0x5599a7 0x7f6c1673672f 0x559fe7 0xfffffffffffffffe
******************************************************************

Execution terminated by unhandled exception
raised CONSTRAINT_ERROR : trans.adb:1357 access check failed
Call stack traceback locations:
0x5f9d3a 0x61796e 0x617abc 0x61cb35 0x630318 0x5fd0dd 0x600bd7 0x5b3488 0x5b35cb 0x645f11 0x64a58f 0x63f33c 0x639f03 0x63bd22 0x63c62e 0x664a7b 0x600a33 0x632189 0x57fddd 0x55c561 0x9577ef 0x95957e 0x57fe39 0x5599a7 0x7f6c1673672f 0x559fe7 0xfffffffffffffffe
ghdl: compilation error
Makefile_vhdl:9 : la recette pour la cible « IXL_tb.o » a échouée
make: *** [IXL_tb.o] Erreur 1

I guess it is my fault, but in case

maximouth, 2017-05-07:
nnaaa forget it it's me...
Sorry....

closed 2017-05-07 by maximouth