ghdl/ghdl #342: work and the current primary unit

rqou, 2017-05-09:
I believe this code should be legal:
```vhdl
entity test is
    type test1 is (foo);
    subtype test2 is work.test.test1;
begin end;
```
However, ghdl will reject it with different errors depending on whether there is a stale work-obj08.cf lying around:
```
$ ghdl -a --std=08 test.vhd 
test.vhd:3:27:error: file "test.vhd" has changed and must be reanalysed
ghdl:error: compilation error
$ rm work-obj08.cf 
$ ghdl -a --std=08 test.vhd 
test.vhd:3:27:error: unit "test" not found in library "work"
ghdl:error: compilation error
$ ghdl --version
GHDL 0.34-dev (2017-03-01-29-g18149662) [Dunoon edition]
 Compiled with GNAT Version: 5.4.1
 GCC back-end code generator
Written by Tristan Gingold.
```

The spec states that "A declaration is visible only within a certain part of its scope; this part starts at the end of the declaration except [doesn't apply here], in which case it starts immediately after the reserved word **is** occurring after the identifier of the design unit, a package declaration, or protected type declaration." So the entity `test` should be visible in the subtype declaration.

This code is accepted by ghdl if the `work.` part is removed.

tgingold, 2017-05-09:
But LRM 13.2 says:
A working library is the library into which the library unit resulting from the analysis of a design unit is placed.

So before and during analysis, the unit is not in the working library, so I would say that `work.test` is not defined.

(This is somewhat a corner case that nobody writes in real code).

rqou, 2017-05-09:
Hmm, I suppse you are right. Closing.

closed 2017-05-09 by rqou