ghdl/ghdl #343: Compiling UVVM Failed

andresdemski, 2017-05-09:
Hi,
I'm trying to compile UVVM using the script provided with ghdl and i get the following output:

[failed.txt](https://github.com/tgingold/ghdl/files/986973/failed.txt)

The command line was: ` /usr/lib/ghdl/vendors/compile-uvvm.sh -c -a --src $HOME/Documents/VHDL_Curse/UVVM_All --out $HOME/Documents/VHDL_Curse/out`

My ghdl version is:
```
ghdl -v
GHDL 0.34-dev (2017-03-01-58-g3031cd34) [Dunoon edition]
 Compiled with GNAT Version: 6.3.1 20170306
 llvm code generator
```
My pc is running Arch Linux:
```
uname -a
Linux demskinator.xps 4.10.6-1-ARCH #1 SMP PREEMPT Mon Mar 27 08:28:22 CEST 2017 x86_64 GNU/Linux
```

Am i doing something wrong or is it a ghdl bug?

Regards

Andres

Paebbels, 2017-05-09:
Hello @andresdemski,

the provided log file starts with these lines:

```
PS C:\Users\Patrick Lehmann> Open-Colored .\Downloads\failed.txt
Vendor directory '/home/ademski/Documents/VHDL_Curse/out' already exists.
Cleaning up vendor directory ...
Compiling library 'uvvm_util'...
Analyzing package '/home/ademski/Documents/VHDL_Curse/UVVM_All/uvvm_util/src/types_pkg.vhd'
/home/ademski/Documents/VHDL_Curse/UVVM_All/uvvm_util/src/types_pkg.vhd:134:14:warning: package "types_pkg" does not require a body [-Wbody]
error: cannot create library file for uvvm_util
/usr/sbin/ghdl: compilation error
Analyzing package '/home/ademski/Documents/VHDL_Curse/UVVM_All/uvvm_util/src/adaptations_pkg.vhd'
/home/ademski/Documents/VHDL_Curse/UVVM_All/uvvm_util/src/adaptations_pkg.vhd:63:19:warning: type of a shared variable must be a protected type [-Wshared]
/home/ademski/Documents/VHDL_Curse/UVVM_All/uvvm_util/src/adaptations_pkg.vhd:200:19:warning: type of a shared variable must be a protected type [-Wshared]
/home/ademski/Documents/VHDL_Curse/UVVM_All/uvvm_util/src/adaptations_pkg.vhd:257:14:warning: package "adaptations_pkg" does not require a body [-Wbody]
error: cannot create library file for uvvm_util
/usr/sbin/ghdl: compilation error
Analyzing package '/home/ademski/Documents/VHDL_Curse/UVVM_All/uvvm_util/src/string_methods_pkg.vhd'
error: cannot create library file for uvvm_util
/usr/sbin/ghdl: compilation error
Analyzing package '/home/ademski/Documents/VHDL_Curse/UVVM_All/uvvm_util/src/protected_types_pkg.vhd'
error: cannot create library file for uvvm_util
/usr/sbin/ghdl: compilation error
Analyzing package '/home/ademski/Documents/VHDL_Curse/UVVM_All/uvvm_util/src/hierarchy_linked_list_pkg.vhd'
error: cannot create library file for uvvm_util
/usr/sbin/ghdl: compilation error
Analyzing package '/home/ademski/Documents/VHDL_Curse/UVVM_All/uvvm_util/src/alert_hierarchy_pkg.vhd'
error: cannot create library file for uvvm_util
/usr/sbin/ghdl: compilation error
Analyzing package '/home/ademski/Documents/VHDL_Curse/UVVM_All/uvvm_util/src/license_pkg.vhd'
error: cannot create library file for uvvm_util
/usr/sbin/ghdl: compilation error
...
```

GHDL reports:

```
error: cannot create library file for uvvm_util
/usr/sbin/ghdl: compilation error
```

Does GHDL have the appropriate rights to write into the destination folder and does it have the rights to create new subfolders?

You can run the pre-compile script in verbose (`-v`) or debug (`-d`) mode to see more details. You can also enable that ist stops after the first error (`-H`). See the integrated help (`-h`) for more details.

The [new documentation](http://ghdl.readthedocs.io/en/new-documentation-structure/building/PrecompileVendorPrimitives.html) is still under construction, but it should list all options.

andresdemski, 2017-05-09:
Hello @Paebbels 
I set a ghdl path, i remove the out folder and i'm still having the same output:
[failed.txt](https://github.com/tgingold/ghdl/files/987557/failed.txt)

Command line: `/usr/lib/ghdl/vendors/compile-uvvm.sh -c -a --ghdl /usr/bin --src $HOME/Documents/VHDL_Curse/UVVM_All --out $HOME/Documents/VHDL_Curse/out > failed.txt `

I suppose that is not a rights problem beacause it generates some output:
```
out
├── bitvis_vip_axilite
│   └── v08
├── bitvis_vip_axistream
│   └── v08
├── bitvis_vip_i2c
│   └── v08
├── bitvis_vip_sbi
│   └── v08
├── bitvis_vip_uart
│   └── v08
├── uvvm_util
│   └── v08
│       ├── adaptations_pkg.o
│       ├── alert_hierarchy_pkg.o
│       ├── hierarchy_linked_list_pkg.o
│       ├── license_pkg.o
│       ├── protected_types_pkg.o
│       ├── string_methods_pkg.o
│       ├── types_pkg.o
│       └── uvvm_util-obj08.cf
└── uvvm_vvc_framework
    └── v08
```

Paebbels, 2017-05-09:
No, the output and error messages have changed:

```
...
Analyzing package '/home/ademski/Documents/VHDL_Curse/UVVM_All/uvvm_util/src/methods_pkg.vhd'
/home/ademski/Documents/VHDL_Curse/UVVM_All/uvvm_util/src/methods_pkg.vhd:33:19:warning: type of a shared variable must be a protected type [-Wshared]
/home/ademski/Documents/VHDL_Curse/UVVM_All/uvvm_util/src/methods_pkg.vhd:34:19:warning: type of a shared variable must be a protected type [-Wshared]
/home/ademski/Documents/VHDL_Curse/UVVM_All/uvvm_util/src/methods_pkg.vhd:35:19:warning: type of a shared variable must be a protected type [-Wshared]
/home/ademski/Documents/VHDL_Curse/UVVM_All/uvvm_util/src/methods_pkg.vhd:36:19:warning: type of a shared variable must be a protected type [-Wshared]
/home/ademski/Documents/VHDL_Curse/UVVM_All/uvvm_util/src/methods_pkg.vhd:37:19:warning: type of a shared variable must be a protected type [-Wshared]
/home/ademski/Documents/VHDL_Curse/UVVM_All/uvvm_util/src/methods_pkg.vhd:38:19:warning: type of a shared variable must be a protected type [-Wshared]
/home/ademski/Documents/VHDL_Curse/UVVM_All/uvvm_util/src/methods_pkg.vhd:39:19:warning: type of a shared variable must be a protected type [-Wshared]
/home/ademski/Documents/VHDL_Curse/UVVM_All/uvvm_util/src/methods_pkg.vhd:40:19:warning: type of a shared variable must be a protected type [-Wshared]
/home/ademski/Documents/VHDL_Curse/UVVM_All/uvvm_util/src/methods_pkg.vhd:41:19:warning: type of a shared variable must be a protected type [-Wshared]
/home/ademski/Documents/VHDL_Curse/UVVM_All/uvvm_util/src/methods_pkg.vhd:42:19:warning: type of a shared variable must be a protected type [-Wshared]
/home/ademski/Documents/VHDL_Curse/UVVM_All/uvvm_util/src/methods_pkg.vhd:43:19:warning: type of a shared variable must be a protected type [-Wshared]
/home/ademski/Documents/VHDL_Curse/UVVM_All/uvvm_util/src/methods_pkg.vhd:44:19:warning: type of a shared variable must be a protected type [-Wshared]
/home/ademski/Documents/VHDL_Curse/UVVM_All/uvvm_util/src/methods_pkg.vhd:5151:10: signal interface "clock_count" cannot be read
/home/ademski/Documents/VHDL_Curse/UVVM_All/uvvm_util/src/methods_pkg.vhd:5152:24: signal interface "clock_count" cannot be read
/home/ademski/Documents/VHDL_Curse/UVVM_All/uvvm_util/src/methods_pkg.vhd:5182:10: signal interface "clock_count" cannot be read
/home/ademski/Documents/VHDL_Curse/UVVM_All/uvvm_util/src/methods_pkg.vhd:5183:24: signal interface "clock_count" cannot be read
/usr/bin/ghdl: compilation error
...
```

There are source code errors in line 5151, 5152, 5182 and 5183.

What's the used UVVM version (tag name or commit hash) for your compilation?

-------------
There are still some GHDL exceptions, that should not occur, but let's start to fix first things first...

andresdemski, 2017-05-10:
The uvvm version is the current one ( commit hash: 2bf09c56c578e20486b8ecc53e5d50010610e7c7). The code i suppose it's right because it works with ModelSim. It could be an unsupported syntax.
In line 5151, they are trying to read a procedure output. Is this ok? Is this supported?

Paebbels, 2017-05-10:
Hello @andresdemski,

I created an issue (#345) for your case. I think the UVVM code is correct at that place.

tgingold, 2017-05-22:
Please reopen (or create a new ticket) if a new problem appears.

closed 2017-05-22 by tgingold