ghdl/ghdl #344: "make libghdlsynth.so" fails to build

m-thu, 2017-05-09:
`make libghdlsynth.so` fails on both Ubuntu 14.04 (GNAT 4.6, gcc version 4.8.4 (Ubuntu 4.8.4-2ubuntu1~14.04.3)) and Arch Linux (GNAT 6.3.1 20170306, gcc version 6.3.1 20170306 (GCC)). `./configure && make` works fine on both distributions. (Tested with commit f250fb5c371ad8e286f4fe7c442ea7e7866f12a0)

Error message on Ubuntu 14.04:
```sh
make libghdlsynth.so
if [ -d ./.git ]; then \
	  desc=`cd .; git describe --dirty`; \
	  sed -e "s/[(].*[)]/($desc)/" < src/version.in > version.tmp; \
	else \
	  cp src/version.in version.tmp; \
	fi
if [ ! -r version.ads ] || ! cmp version.tmp version.ads > /dev/null; then cp version.tmp version.ads; fi
gnatmake -I- -aI. -D pic -z libghdlsynth -o libghdlsynth.so -gnaty3befhkmr -gnatwae -gnatf -gnat05 -g -gnata -fPIC -gnat12 -aI./src -aI./src/vhdl -aI./src/grt -aI./src/psl -aI./src/vhdl/translate -aI./src/synth -aI./src/ghdldrv -aI./src/vhdl/simulate -bargs -shared -Llibghdlsynth_ -largs -shared  pic/jumps.o  pic/times.o  pic/grt-cbinding.o  pic/grt-cvpi.o  pic/fstapi.o  pic/lz4.o  pic/fastlz.o -Wl,-soname,libghdlsynth.so
gcc-4.6 -c -I. -gnaty3befhkmr -gnatwae -gnatf -gnat05 -g -gnata -fPIC -gnat12 -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/synth -I./src/ghdldrv -I./src/vhdl/simulate -I- -o /home/matthias/ghdl/pic/netlists.o /home/matthias/ghdl/src/synth/netlists.adb
netlists.ads:248:04: warning: no warning suppressed by this pragma
gnatmake: "/home/matthias/ghdl/src/synth/netlists.adb" compilation error
make: *** [libghdlsynth.so] Error 4
```

Error message on Arch Linux:
```sh
make libghdlsynth.so 
if [ -d ./.git ]; then \
  desc=`cd .; git describe --dirty`; \
  sed -e "s/[(].*[)]/($desc)/" < src/version.in > version.tmp; \
else \
  cp src/version.in version.tmp; \
fi
if [ ! -r version.ads ] || ! cmp version.tmp version.ads > /dev/null; then cp version.tmp version.ads; fi
gnatmake -I- -aI. -D pic -z libghdlsynth -o libghdlsynth.so -gnaty3befhkmr -gnatwae -gnatf -gnat05 -g -gnata -fPIC -gnat12 -aI./src -aI./src/vhdl -aI./src/grt -aI./src/psl -aI./src/vhdl/translate -aI./src/synth -aI./src/ghdldrv -aI./src/vhdl/simulate -bargs -shared -Llibghdlsynth_ -largs -shared  pic/jumps.o  pic/times.o  pic/grt-cbinding.o  pic/grt-cvpi.o  pic/fstapi.o  pic/lz4.o  pic/fastlz.o -Wl,-soname,libghdlsynth.so
gcc -c -I. -gnaty3befhkmr -gnatwae -gnatf -gnat05 -g -gnata -fPIC -gnat12 -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/synth -I./src/ghdldrv -I./src/vhdl/simulate -I- -o /home/matthias/ghdl/pic/execution.o /home/matthias/ghdl/src/vhdl/simulate/execution.adb
execution.adb:1352:34: "String_Real_Digits" not declared in "Vstrings"
gnatmake: "/home/matthias/ghdl/src/vhdl/simulate/execution.adb" compilation error
make: *** [Makefile:327: libghdlsynth.so] Error 4
```

tgingold, 2017-05-09:
On 09/05/17 20:59, Matthias Thumann wrote:
> |make libghdlsynth.so| fails on both Ubuntu 14.04 (GNAT 4.6, gcc version
> 4.8.4 (Ubuntu 4.8.4-2ubuntu1~14.04.3)) and Arch Linux (GNAT 6.3.1
> 20170306, gcc version 6.3.1 20170306 (GCC)). |./configure && make| works
> fine on both distributions. (Tested with commit f250fb5
> <https://github.com/tgingold/ghdl/commit/f250fb5c371ad8e286f4fe7c442ea7e7866f12a0>)
>
> Error message on Ubuntu 14.04:
>
> make libghdlsynth.so
> if [ -d ./.git ]; then \
> 	  desc=`cd .; git describe --dirty`; \
> 	  sed -e "s/[(].*[)]/($desc)/" < src/version.in > version.tmp; \
> 	else \
> 	  cp src/version.in version.tmp; \
> 	fi
> if [ ! -r version.ads ] || ! cmp version.tmp version.ads > /dev/null; then cp version.tmp version.ads; fi
> gnatmake -I- -aI. -D pic -z libghdlsynth -o libghdlsynth.so -gnaty3befhkmr -gnatwae -gnatf -gnat05 -g -gnata -fPIC -gnat12 -aI./src -aI./src/vhdl -aI./src/grt -aI./src/psl -aI./src/vhdl/translate -aI./src/synth -aI./src/ghdldrv -aI./src/vhdl/simulate -bargs -shared -Llibghdlsynth_ -largs -shared  pic/jumps.o  pic/times.o  pic/grt-cbinding.o  pic/grt-cvpi.o  pic/fstapi.o  pic/lz4.o  pic/fastlz.o -Wl,-soname,libghdlsynth.so
> gcc-4.6 -c -I. -gnaty3befhkmr -gnatwae -gnatf -gnat05 -g -gnata -fPIC -gnat12 -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/synth -I./src/ghdldrv -I./src/vhdl/simulate -I- -o /home/matthias/ghdl/pic/netlists.o /home/matthias/ghdl/src/synth/netlists.adb
> netlists.ads:248:04: warning: no warning suppressed by this pragma
> gnatmake: "/home/matthias/ghdl/src/synth/netlists.adb" compilation error
> make: *** [libghdlsynth.so] Error 4

This one is due to an older compiler.  Replace -gnatwae with -gnatwa in 
Makefile.

> Error message on Arch Linux:
>
> make libghdlsynth.so
> if [ -d ./.git ]; then \
>   desc=`cd .; git describe --dirty`; \
>   sed -e "s/[(].*[)]/($desc)/" < src/version.in > version.tmp; \
> else \
>   cp src/version.in version.tmp; \
> fi
> if [ ! -r version.ads ] || ! cmp version.tmp version.ads > /dev/null; then cp version.tmp version.ads; fi
> gnatmake -I- -aI. -D pic -z libghdlsynth -o libghdlsynth.so -gnaty3befhkmr -gnatwae -gnatf -gnat05 -g -gnata -fPIC -gnat12 -aI./src -aI./src/vhdl -aI./src/grt -aI./src/psl -aI./src/vhdl/translate -aI./src/synth -aI./src/ghdldrv -aI./src/vhdl/simulate -bargs -shared -Llibghdlsynth_ -largs -shared  pic/jumps.o  pic/times.o  pic/grt-cbinding.o  pic/grt-cvpi.o  pic/fstapi.o  pic/lz4.o  pic/fastlz.o -Wl,-soname,libghdlsynth.so
> gcc -c -I. -gnaty3befhkmr -gnatwae -gnatf -gnat05 -g -gnata -fPIC -gnat12 -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/synth -I./src/ghdldrv -I./src/vhdl/simulate -I- -o /home/matthias/ghdl/pic/execution.o /home/matthias/ghdl/src/vhdl/simulate/execution.adb
> execution.adb:1352:34: "String_Real_Digits" not declared in "Vstrings"
> gnatmake: "/home/matthias/ghdl/src/vhdl/simulate/execution.adb" compilation error
> make: *** [Makefile:327: libghdlsynth.so] Error 4

I will push the patch soon (Replace _Digits by _Format).

Thanks!
Tristan.

m-thu, 2017-05-10:
If I replace `Str : Grt.Vstrings.String_Real_Digits;` with `Str : Grt.Vstrings.String_Real_Format;` in Line 1352 in `src/vhdl/simulate/execution.adb`, `make libghdlsynth.so` works fine on Arch Linux.

On Ubuntu compilation still fails with this error:
```sh
make libghdlsynth.so
if [ -d ./.git ]; then \
	  desc=`cd .; git describe --dirty`; \
	  sed -e "s/[(].*[)]/($desc)/" < src/version.in > version.tmp; \
	else \
	  cp src/version.in version.tmp; \
	fi
if [ ! -r version.ads ] || ! cmp version.tmp version.ads > /dev/null; then cp version.tmp version.ads; fi
gnatmake -I- -aI. -D pic -z libghdlsynth -o libghdlsynth.so -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata -fPIC -gnat12 -aI./src -aI./src/vhdl -aI./src/grt -aI./src/psl -aI./src/vhdl/translate -aI./src/synth -aI./src/ghdldrv -aI./src/vhdl/simulate -bargs -shared -Llibghdlsynth_ -largs -shared  pic/jumps.o  pic/times.o  pic/grt-cbinding.o  pic/grt-cvpi.o  pic/fstapi.o  pic/lz4.o  pic/fastlz.o -Wl,-soname,libghdlsynth.so
gcc-4.6 -c -I. -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata -fPIC -gnat12 -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/synth -I./src/ghdldrv -I./src/vhdl/simulate -I- -o /home/matthias/ghdl/pic/netlists-dump.o /home/matthias/ghdl/src/synth/netlists-dump.adb
netlists-dump.adb:147:32: missing ".."
netlists-dump.adb:394:35: missing ".."
gnatmake: "/home/matthias/ghdl/src/synth/netlists-dump.adb" compilation error
make: *** [libghdlsynth.so] Error 4

```

tgingold, 2017-05-10:
You are using a too old version of gcc on Ubuntu.

m-thu, 2017-05-11:
> You are using a too old version of gcc on Ubuntu.
Ok, thanks!

closed 2017-05-11 by m-thu

RCoeurjoly, 2017-12-28:
I get the same error as reported by m-thu.

make libghdlsynth.so 
if test -d ./.git \
	   && desc=`cd .; git describe --dirty`; then \
	  sed -e "s/[(].*[)]/($desc)/" < src/version.in > version.tmp; \
	else \
	  cp src/version.in version.tmp; \
	fi
if [ ! -r version.ads ] || ! cmp version.tmp version.ads > /dev/null; then cp version.tmp version.ads; fi
gnatmake -I- -aI. -D pic -z libghdlsynth -o libghdlsynth.so -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata -gnatwe -fPIC -gnat12 -aI./src -aI./src/vhdl -aI./src/grt -aI./src/psl -aI./src/vhdl/translate -aI./src/synth -aI./src/ghdldrv -aI./src/vhdl/simulate -bargs -shared -Llibghdlsynth_ -largs -shared  pic/jumps.o  pic/times.o  pic/grt-cbinding.o -Wl,-soname,libghdlsynth.so
gcc-4.6 -c -I. -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata -gnatwe -fPIC -gnat12 -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/synth -I./src/ghdldrv -I./src/vhdl/simulate -I- -o /home/rcl/ghdl/pic/netlists.o /home/rcl/ghdl/src/synth/netlists.adb
netlists.ads:248:04: warning: no warning suppressed by this pragma
gnatmake: "/home/rcl/ghdl/src/synth/netlists.adb" compilation error
make: *** [libghdlsynth.so] Error 4

the version of gcc instaled are the following:
g++ -v
Using built-in specs.
COLLECT_GCC=g++
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/7/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 7.2.0-1ubuntu1~14.04' --with-bugurl=file:///usr/share/doc/gcc-7/README.Bugs --enable-languages=c,ada,c++,go,brig,d,fortran,objc,obj-c++ --prefix=/usr --with-gcc-major-version-only --program-suffix=-7 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=gcc4-compatible --disable-libstdcxx-dual-abi --enable-gnu-unique-object --disable-vtable-verify --enable-libmpx --enable-plugin --with-system-zlib --with-target-system-zlib --enable-objc-gc=auto --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 7.2.0 (Ubuntu 7.2.0-1ubuntu1~14.04) 

gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/7/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 7.2.0-1ubuntu1~14.04' --with-bugurl=file:///usr/share/doc/gcc-7/README.Bugs --enable-languages=c,ada,c++,go,brig,d,fortran,objc,obj-c++ --prefix=/usr --with-gcc-major-version-only --program-suffix=-7 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=gcc4-compatible --disable-libstdcxx-dual-abi --enable-gnu-unique-object --disable-vtable-verify --enable-libmpx --enable-plugin --with-system-zlib --with-target-system-zlib --enable-objc-gc=auto --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 7.2.0 (Ubuntu 7.2.0-1ubuntu1~14.04) 

What gcc version should I install?

Thank you for all the good work! I am really looking forward to use GHDL for synthesis.

1138-4EB, 2017-12-28:
@RCoeurjoly , can you please edit the comment to put code in code blocks?
[Do not reply to this message, I will remove it after some time]