ghdl/ghdl #345: Reading out parameters in procedures is not supported

Paebbels, 2017-05-10:
Hello Tristan,

this is a breakdown of [issue 343 - Compiling UVVM with compile-uvvm.sh failed](https://github.com/tgingold/ghdl/issues/343).

**Description:**
`out` parameters in procedures cannot be read in VHDL-2008 mode.
The parameter kind in the provided test case is `signal`.

As far as I remember, the read-back of values from interface objects of mode `out` is not restricted to (signal) ports in an entity. Right?

**Example code:**
```VHDL
procedure test(
  constant value : in  natural;
  signal   count : out natural
) is
begin
  for i in 1 to value loop
    if count < 10 then
      count = count + 1;
    else
      count = 0;
    end if;
  end loop;
end procedure;
```

Please see the linked issue #343 for a complete example provided in UVVM.

tgingold, 2017-05-11:
On 11/05/17 00:52, Patrick Lehmann wrote:
> Hello Tristan,
>
> this is a breakdown of issue 343 - Compiling UVVM with compile-uvvm.sh
> failed <https://github.com/tgingold/ghdl/issues/343>.
>
> *Description:*
> |out| parameters in procedures cannot be [...]

I don't think so.  LRM 6.5.2:

out. The value of the interface object is allowed to be updated and, 
provided it is not a signal parameter, read.

Tristan.

Paebbels, 2017-05-11:
(I tired to update your comment - originally received by email - but the HTML/CSS is broken. So I write a new comment.)

Yes, you and the LRM are right :). I'll file an issue at UVVM.

> **LRM 6.5.2 Interface object declarations**
> ...
> An interface object has one of the following modes:
> * **in** ...
> * **out**. The value of the interface object is allowed to be updated and,  provided it is not a signal parameter, read. Reading the attributes of the interface object is allowed, unless the interface object
is a signal parameter and the attribute is one of 'STABLE, 'QUIET, 'DELAYED, 'TRANSACTION, 'EVENT, 'ACTIVE, 'LAST_EVENT, 'LAST_ACTIVE, or 'LAST_VALUE.
> * **inout** or **buffer** ...
> * **linkage** ...

closed 2017-05-11 by Paebbels