ghdl/ghdl #346: Strange gcov result

markNZed, 2017-05-12:
I'm experimenting with gcov and get results like:
```
        -:  303:        procedure START_WR_XRM (a : UNSIGNED (15 downto 0)) is
        -:  304:        begin
    #####:  305:            xrm_addr <= a;
    #####:  306:            xrm_rd <= '0';
    17401:  307:            xrm_wr <= '1';
        -:  308:        end START_WR_XRM;
```
Line 307 can't be executed 17401 times while line 306 is never executed. I believe the test is not using that procedure so the report for line 307 is probably wrong.

The next procedure in the same file:
```
        -:  312:        procedure STOP_RD_WR_XRM is
        -:  313:        begin
        -:  314:
    17401:  315:            xrm_addr <= CD_16; -- 
    17401:  316:            xrm_out_data <= CD_8; -- 
        -:  317:
    17401:  318:            xrm_rd <= '0';
    19449:  319:            xrm_wr <= '0';
        -:  320:        end STOP_RD_WR_XRM;
        -:  321:
```
Line 319 should execute the same number of times as line 318. Maybe just coincidence but it is again the xrm_wr signal and the count in the previous procedure of 17401 matches the counts of line 318. 

```
ghdl -v
GHDL 0.34-dev (2016-09-14-303-g603e2fe) [Dunoon edition]
 Compiled with GNAT Version: 4.9.3
```

Is there a known issue with gcov accuracy when used with VHDL ?

tgingold, 2017-05-14:
It looks like code executed for the next procedure is accounted to the previous one.
Should be fixable.

Paebbels, 2017-05-15:
@markNZed Did you compile the VHDL code with/without optimizations?

markNZed, 2017-05-15:
I used:

```
...
ghdl -a -fprofile-arcs -ftest-coverage --ieee=synopsys --std=93 --workdir=work --work=work ../vhdl/i8051_ctr.vhd
...
ghdl -e -Wl,-L/opt/local/lib/gcc/x86_64-unknown-linux-gnu/4.9.4 -Wl,-lgcov --ieee=synopsys --std=93 --workdir=work  --work=work CFG_I8051_TSB
```

Brian-Drummond, 2017-05-18:
I have seen similar, sometimes activity is attributed to the last active line in a function, entity or package when it really belongs somewhere else (presumably "location" was last set to this line and not updated in code generation when "trivial" events like "end procedure" which generate no active code happen. It would be nice to have this fixed!

Brian-Drummond, 2018-04-15:
Just re-checked with ghdl --version
GHDL 0.36-dev (v0.35-153-gba14e9da) [Dunoon edition]
 Compiled with GNAT Version: 6.3.0
 GCC back-end code generator

Possibly not the exact same problem as in the original question, but the similar behaviour I commented on is still there. 

There's a testcase at https://stackoverflow.com/questions/42079222/why-does-this-fsm-not-reach-100-code-coverage/42079819#42079819 which may be useful.  Adding a "null" statement after "end case" moves the 20 executions from the "others" clause to the "null" statement; clearly indicating they were spurious (not a result of nonexistent case choices).

Perhaps there would be some way of labelling "end process" as an active line for the purposes of code coverage?

Paebbels, 2018-04-15:
@Brian-Drummond  
Do you recommend to reopen this issue or are you going to open a new issue for this behavior and link it to this one?