ghdl/ghdl #348: Checkpoint during simulation

Bamban, 2017-05-17:
Hi,
Is there a way to checkpoint during a simulation in order to restart later ?
Ideally, I have a circuit initialization that lasts 30 minutes. If I could save that and skip it for other iterations that would be great. Indeed, I don't want to change the source files between iteration but only a portion of a waveopt file. The idea is to select portions of the circuit to visualize with gtkwave and adjust the waveopt file after viewing. I could restart simulation at the checkpoint with a different waveopt file (so I would skip a new compilation and the circuit initialization phase).
Thanks for your tip !

Paebbels, 2017-05-17:
Currently, I don't know any simulator with such a feature. It could be implemented with simulator interfaces like FLI, VPI or VHPI:
* stop at time x
* recursively iterate the design and collect all data
* write data to file
* ...
* initialize design
* load data from file
* overwrite values recursively from file
* set time to x+1 and continue

That might be a good use case for a Python API.

tgingold, 2017-05-19:
I think everyone agree that this is a nice feature, albeit not yet implemented!

closed 2017-05-19 by tgingold