ghdl/ghdl #349: "error: NULL access dereferenced" at t=0 ps on recent OSVVM package...

Brian-Drummond, 2017-05-18:
ghdl-gcc compiles and elaborates this testcase without error, but 'ghdl -r' raises the above error at the start of the run - apparently from the ELAB_SPEC procedure within the new (as of Nov/Dec 2016) ResolutionPkg package.

ghdl mcode version seems to be happy with it.

Testcase...

```
library osvvm;
use osvvm.ResolutionPkg.all;

entity Resolution_TB is
end Resolution_TB;

architecture none of Resolution_TB is
begin
end none;
```
Build script ... assumes OSVVM (post Nov 2016) has been installed as part of Vunit. Adjust paths to OSVVM package files otherwise...

```
ghdl -a --std=08 -g --work=OSVVM /usr/local/lib/python2.7/dist-packages/vunit/vhdl/osvvm/TranscriptPkg.vhd
ghdl -a --std=08 -g --work=OSVVM /usr/local/lib/python2.7/dist-packages/vunit/vhdl/osvvm/TextUtilPkg.vhd
ghdl -a --std=08 -g --work=OSVVM /usr/local/lib/python2.7/dist-packages/vunit/vhdl/osvvm/NamePkg.vhd
ghdl -a --std=08 -g --work=OSVVM /usr/local/lib/python2.7/dist-packages/vunit/vhdl/osvvm/OsvvmGlobalPkg.vhd
ghdl -a --std=08 -g --work=OSVVM /usr/local/lib/python2.7/dist-packages/vunit/vhdl/osvvm/AlertLogPkg.vhd
ghdl -a --std=08 -g --work=OSVVM /usr/local/lib/python2.7/dist-packages/vunit/vhdl/osvvm/ResolutionPkg.vhd
ghdl -a --std=08 -g Resolution_TB.vhd
ghdl -e --std=08 Resolution_TB
ghdl -r Resolution_TB
```
Error:

> ./resolution_tb:error: NULL access dereferenced
>   from: osvvm.resolutionpkg.ELAB_SPEC at ResolutionPkg.vhd:59
>   from: osvvm.resolutionpkg.ELAB_BDY at ResolutionPkg.vhd:389
>   from: work.resolution_tb.PKG_ELAB at Resolution_TB.vhd:4
> ./resolution_tb:error: error during elaboration

GHDL version (built this morning)

> ghdl --version
> GHDL 0.34-dev (2017-03-01-74-g669d25e) [Dunoon edition]
>  Compiled with GNAT Version: 4.9.2
>  GCC back-end code generator
> Written by Tristan Gingold.
> 
> Copyright (C) 2003 - 2015 Tristan Gingold.
> GHDL is free software, covered by the GNU General Public License.  There is NO
> warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Workaround: For my purpose, commenting this package ( ResolutionPkg) out of OsvvmContext.vhd seems to be good enough.

If there is a problem with the OSVVM package, I would still expect better diagnostics than this!

Paebbels, 2017-05-18:
Hello @Brian-Drummond,

to be clear: Just the reference to library OSVVM and the usage of the package causes the error, right?

@tgingold: The message `NULL access dereferenced` is a pure runtime error caused by the user's VHDL code and not related to GHDL's Ada code, right?

As far as I can see in the code, there are no (VHDL) access types involved near that lines. So is at a message from Ada?

* osvvm.resolutionpkg.ELAB_SPEC at [ResolutionPkg.vhd:59](https://github.com/OSVVM/OSVVM/blob/master/ResolutionPkg.vhd?ts=2#L59)
* osvvm.resolutionpkg.ELAB_BDY at [ResolutionPkg.vhd:389](https://github.com/OSVVM/OSVVM/blob/master/ResolutionPkg.vhd?ts=2#L389)
  I suspect it should be line 387, because this line requires line 59 for `MULTIPLE_DRIVER_SEVERITY`.

Does GHDL support assert/report-statements in resolution functions? I know from @JimLewis that most of his tested simulators don't support or have big problems if a resolution functions has some kind of external interaction like asserting/reporting or accessing a PT instance.

-----------------------
Notification list: @JimLewis

Brian-Drummond, 2017-05-18:
Just the usage of that package, which causes its ELAB hooks to be called.

I'm not clear that there are no (VHDL) access types; for example  I'm not sure how Integer_Vector is implemented internally. 
There's only one occurrence of "NULL access dereferenced" in the source ... Grt_Null_Access_Error ... which has "pragma Export (C, Grt_Null_Access_Error, "grt_null_access_error");" so I think this is the generated runtime, not Ada (which makes sense, since the mcode backend doesn't do the same).

JimLewis, 2017-05-18:
Hi,
I suggest two modifications to the package (done separately) to
try to determine what it in the package is causing the issue.
1)  Remove the constant MULTIPLE_DRIVER_SEVERITY and replace its value with ERROR directly in the code.
2)  If 1 still fails, then remove assert from the resolution functions.

If #1 fixes the issue, I would be happy to remove the constant from the OSVVM release.

The resolution functions with the simple name "resolved" are in the package for
backward compatibility.   I recommend the use of resolved_max instead as these
provide a simpler methodology.

In other tools assert within a resolution functions does not cause tool
execution errors.   Instead, where they stop when an assert failure is
encountered in regular code, they simply do not stop when an assert
failure is contained within a resolution function.

Since failure does not operate properly, there is no reason to use it.
Hence, the current versions of resolved_max do not use any error checking.
Within the OSVVM methodology this is ok as first, there should not be multiple
drivers on the interface and if there are, the wrong value will be transferred -
with only a single fault, a failure will occur at a higher level in the test due
to the checkers receiving the wrong value.   It would take two identical failures
on two different interfaces to cause a false pass on a test.

Hence, from an OSVVM methodology perspective, the error detection in the
resolution functions are what I call "Valley Girl Checks" - they are trying to be
"for sure, for sure".

Going further, to facilitate "Valley Girl Checks", I may eventually create debug
versions of ResolutionPkg.   One will use alerts in the checkers as commercial
tools erroneously allow them and they work well.    The other will use assert.
Once the VHDL-2017 assert API is implemented, if the counting works within
resolution functions, then it will become the preferred solution - I am not as
concerned about stopping as I am about the error being reported to the user
in a way that they have to actively choose to ignore it.

Cheers,
Jim

> Just the usage of that package, which causes its ELAB hooks to be called.
>
> I'm not clear that there are no (VHDL) access types; for example I'm not sure how Integer_Vector is implemented internally. [...]

-- 
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Jim Lewis
VHDL Training Expert, SynthWorks
IEEE 1076 VHDL Working Group Chair
Open Source VHDL Verification Methodology (OSVVM), Chief Architect and Co-founder

1-503-590-4787
Jim@SynthWorks.com
http://www.SynthWorks.com

VHDL Training on leading-edge, best coding practices for hardware design and verification.
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

Brian-Drummond, 2017-05-19:
Thanks Jim ... both of those are fine.
The problem turns out to be two subtype declarations, defining subtypes of time_vector ... a bit of a corner case!
```
  subtype  time_vector_max is (resolved_max) time_vector ;   -- line 94
  subtype  time_vector_sum is (resolved_sum) time_vector ;   -- line 122
```
I see nothing wrong with these declarations, but comment out just these two lines and ghdl is happy with the lot.
I think this puts the ball in ghdl's court.

JimLewis, 2017-05-19:
Hi Brian,
But the subtype for integer_vector did work?
   subtype  integer_vector_max is (resolved_max) integer_vector ;

If yes, then wierd.

Cheers,
Jim

> Thanks Jim ... both of those are fine.
> The problem turns out to be two subtype declarations, defining subtypes of time_vector ... a bit of a corner case!
>
> |subtype time_vector_max is (resolved_max) [...]

-- 
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Jim Lewis
VHDL Training Expert, SynthWorks
IEEE 1076 VHDL Working Group Chair
Open Source VHDL Verification Methodology (OSVVM), Chief Architect and Co-founder

1-503-590-4787
Jim@SynthWorks.com
http://www.SynthWorks.com

VHDL Training on leading-edge, best coding practices for hardware design and verification.
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

tgingold, 2017-05-20:
I can reproduce the issue.

commit 4d4e8d28 (2017-05-22): translate: avoid to duplicate range_var.

Fix #349
closed 2017-05-22 by tgingold