ghdl/ghdl #350: Compilation of ghdl using GNAT GPL 2016 fails on OS X 10.11.6 El Capitan

maehne, 2017-05-19:
I just tried to compile GHDL from the master branch (commit 897f45) on OS X El Capitan using GNAT GPL 2016. After an successful ```./configure``` step, compilation using ```make``` fails due to warnings in ```trans-chap9.adb```:
```
$ ./configure
[...]
$ make
[...]
gcc -c -I./ -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/mcode -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata -gnatwe -gnatw.A -I- /Users/maehne/Build/ghdl/src/vhdl/translate/trans-chap9.adb
trans-chap9.adb:1589:62: warning: others choice is redundant
trans-chap9.adb:1589:62: warning: previous choices cover all components
gnatmake: "/Users/maehne/Build/ghdl/src/vhdl/translate/trans-chap9.adb" compilation error
make: *** [ghdl_mcode] Error 4
```
Simply removing the ```others``` choice in the line 1589 of file ```trans-chap9.adb``` does not fix the problem. Does anyone have an idea 

The following are the used gcc/gnat versions:
```
$ gcc --version
gcc (GCC) 4.9.4 20160426 (for GNAT GPL 2016 20160515)
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.
See your AdaCore support agreement for details of warranty and support.
If you do not have a current support agreement, then there is absolutely
no warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR
PURPOSE.

$ gnat --version
GNAT GPL 2016 (20160515-49)
Copyright (C) 1996-2016, Free Software Foundation, Inc.
This is free software; see the source for copying conditions.
There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

tgingold, 2017-05-20:
Delete the following line from Makefile:
GNATFLAGS+=-gnatwe

Looks like an incorrect warning.

tgingold, 2017-05-20:
I also have added a new configure option: --disable-werror
Will push it soon.

closed 2017-05-20 by tgingold

maehne, 2017-05-21:
Thanks a lot for the quick reply! Your workaround works well. I was able to compile and install ghdl with mcode backend on OS X El Capitan without further difficulties.

maehne, 2017-05-21:
For information, the same error occurs with the latest version of GNAT GCC available from:
https://sourceforge.net/projects/gnuada/files/GNAT_GCC%20Mac%20OS%20X/7.1.0/native-2016/