ghdl/ghdl #352: GHDL with LLVM backend: std_standard.o does not get installed by make install

maehne, 2017-05-21:
I have built ghdl with LLVM backend on OS X El Capitan using GNAT GPL 2016 and LLVM-4.0 from MacPorts:
```
$ ./configure --with-llvm-config=/opt/local/bin/llvm-config-mp-4.0
```
* Disable ```-gnatwe``` in Makefile (see issue #350)
```
$ make
$ sudo make install
```
Compilation and installation gives no obvious error.
However, once I elaborate even the simplest test bench (in my case the test bench for a 7 segment decoder), I get the following GHDL compilation error:
```
$ ghdl -e BCDTo7SegTb
gcc: error: /usr/local/lib/ghdl/v93/std/std_standard.o: No such file or directory
ghdl:error: compilation error
```
Sure enough ```std_standard.o``` does not exist in the indicated directory (nor for the other VHDL standards):
```
$ ls /usr/local/lib/ghdl/v93/std/
std-obj93.cf  textio.o      textio_body.o
```
So it doesn't seem to get installed by ```make install```. Though, the GHDL build directory contains ```std_standard.o```.

The ```Makefile.in``` of GHDL contains a related TODO item "# TODO?: move std_standard in libraries/Makefile". However, I could not find any obvious error in the surrounding.

By the way, before building GHDL only with the LLVM backend, I built and install GHDL with LLVM backend and libbacktrace. When I did the same test (elaborating BCDTo7SegTb), it showed the same error message plus an error that it could not find ``` /usr/local/lib/ghdl/libbacktrace.a```. Should this maybe get a separate issue?

tgingold, 2017-05-21:
The install.vhdllib copy the content of local dirs to installation dir.  So it looks weird that std_standard.o is locally present but not installed.

Can you show the output of `make install` ?

For libbacktrace.a, how did you configure ghdl ?

maehne, 2017-05-21:
```
$ ./configure --with-llvm-config=/opt/local/bin/llvm-config-mp-4.0
Build machine is: x86_64-apple-darwin14.5.0
Debugging is not enabled with llvm 4.0.0
Creating ghdl.gpr
Creating Makefile
Generate default_paths.ads
$ vi Makefile
[Disable -gnatwe]
$ make -j4
[...]
$ sudo make install
$ sudo make install
[ -d /usr/local ] || mkdir -p /usr/local
[ -d /usr/local/bin ] || mkdir -p /usr/local/bin
[ -d /usr/local/lib ] || mkdir -p /usr/local/lib
[ -d /usr/local/lib/ghdl ] || \
	  mkdir -p /usr/local/lib/ghdl
[ -d /usr/local/include ] || mkdir -p /usr/local/include
if [ -d ./.git ]; then \
	  desc=`cd .; git describe --dirty`; \
	  sed -e "s/[(].*[)]/($desc)/" < src/version.in > version.tmp; \
	else \
	  cp src/version.in version.tmp; \
	fi
if [ ! -r version.ads ] || ! cmp version.tmp version.ads > /dev/null; then cp version.tmp version.ads; fi
/Applications/Xcode.app/Contents/Developer/usr/bin/make -f ./src/ortho/llvm4-nodebug/Makefile \
	 ortho_srcdir=./src/ortho ortho_exec=ghdl1-llvm \
	 GNAT_FLAGS="-aI./src -aI./src/vhdl -aI./src/grt -aI./src/psl -aI./src/vhdl/translate -aI./src/ghdldrv -aI./src/ortho -aI./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata" LDFLAGS="-Wl,-no_compact_unwind" \
	 LLVM_CONFIG="/opt/local/bin/llvm-config-mp-4.0" CXX="clang++" \
	 all
gnatmake -o ghdl1-llvm -aI./src/ortho/llvm4-nodebug -aI./src/ortho \
	-aI./src -aI./src/vhdl -aI./src/grt -aI./src/psl -aI./src/vhdl/translate -aI./src/ghdldrv -aI./src/ortho -aI./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata ortho_code_main -bargs -E \
	-largs llvm-cbindings.o --LINK=clang++ \
	-Wl,-no_compact_unwind `/opt/local/bin/llvm-config-mp-4.0 --ldflags --libs --system-libs`
gnatmake: "ghdl1-llvm" up to date.
gnatmake -aI./src -aI./src/vhdl -aI./src/grt -aI./src/psl -aI./src/vhdl/translate -aI./src/ghdldrv -aI./src/ortho -aI./src/ortho/llvm4-nodebug -aI./src/ghdldrv -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata\
	 ghdl_llvm -bargs -E -largs -Wl,-no_compact_unwind
gnatmake: "ghdl_llvm" up to date.
install -m 755 ghdl_llvm /usr/local/bin/ghdl
install -m 755 ghdl1-llvm /usr/local/bin/ghdl1-llvm
for d in src/std src/ieee src/vital95 src/vital2000 src/synopsys src/mentor src/ieee2008 v87/std v87/ieee v87/synopsys v93/std v93/ieee v93/mentor v93/synopsys v08/std v08/ieee v08/synopsys ; do \
	  mkdir -p /usr/local/lib/ghdl/$d; \
	  install -m 644 -p \
	    lib/ghdl/$d/* /usr/local/lib/ghdl/$d; \
	done
# ANSI color
install -m 644 -p \
	    ./dist/ansi_color.sh /usr/local/lib/ghdl/;
mkdir -p /usr/local/lib/ghdl/vendors
install -m 644 -p \
	    ./libraries/vendors/* /usr/local/lib/ghdl/vendors/
install -m 755 -p \
	    ./libraries/vendors/*.sh /usr/local/lib/ghdl/vendors/
cd grt; gnatmake -c -aI/Users/maehne/Build/ghdl/./src/grt -aI.. -gnatec/Users/maehne/Build/ghdl/./src/grt/grt.adc -gnat05 -gnatdY \
	  ghdl_main  -cargs -g
gcc -c -I./ -I/Users/maehne/Build/ghdl/./src/grt -I.. -gnatec/Users/maehne/Build/ghdl/./src/grt/grt.adc -gnat05 -gnatdY -g -I- /Users/maehne/Build/ghdl/src/grt/ghdl_main.adb
gcc -c -I./ -I/Users/maehne/Build/ghdl/./src/grt -I.. -gnatec/Users/maehne/Build/ghdl/./src/grt/grt.adc -gnat05 -gnatdY -g -I- /Users/maehne/Build/ghdl/src/grt/grt.ads
gcc -c -I./ -I/Users/maehne/Build/ghdl/./src/grt -I.. -gnatec/Users/maehne/Build/ghdl/./src/grt/grt.adc -gnat05 -gnatdY -g -I- /Users/maehne/Build/ghdl/src/grt/grt-errors.adb
gcc -c -I./ -I/Users/maehne/Build/ghdl/./src/grt -I.. -gnatec/Users/maehne/Build/ghdl/./src/grt/grt.adc -gnat05 -gnatdY -g -I- /Users/maehne/Build/ghdl/src/grt/grt-main.adb
gcc -c -I./ -I/Users/maehne/Build/ghdl/./src/grt -I.. -gnatec/Users/maehne/Build/ghdl/./src/grt/grt.adc -gnat05 -gnatdY -g -I- /Users/maehne/Build/ghdl/src/grt/grt-options.adb
gcc -c -I./ -I/Users/maehne/Build/ghdl/./src/grt -I.. -gnatec/Users/maehne/Build/ghdl/./src/grt/grt.adc -gnat05 -gnatdY -g -I- /Users/maehne/Build/ghdl/src/grt/grt-rtis_binding.ads
gcc -c -I./ -I/Users/maehne/Build/ghdl/./src/grt -I.. -gnatec/Users/maehne/Build/ghdl/./src/grt/grt.adc -gnat05 -gnatdY -g -I- /Users/maehne/Build/ghdl/src/grt/grt-std_logic_1164.adb
gcc -c -I./ -I/Users/maehne/Build/ghdl/./src/grt -I.. -gnatec/Users/maehne/Build/ghdl/./src/grt/grt.adc -gnat05 -gnatdY -g -I- /Users/maehne/Build/ghdl/src/grt/grt-types.ads
gcc -c -I./ -I/Users/maehne/Build/ghdl/./src/grt -I.. -gnatec/Users/maehne/Build/ghdl/./src/grt/grt.adc -gnat05 -gnatdY -g -I- /Users/maehne/Build/ghdl/src/grt/grt-astdio.adb
gcc -c -I./ -I/Users/maehne/Build/ghdl/./src/grt -I.. -gnatec/Users/maehne/Build/ghdl/./src/grt/grt.adc -gnat05 -gnatdY -g -I- /Users/maehne/Build/ghdl/src/grt/grt-backtraces.adb
gcc -c -I./ -I/Users/maehne/Build/ghdl/./src/grt -I.. -gnatec/Users/maehne/Build/ghdl/./src/grt/grt.adc -gnat05 -gnatdY -g -I- /Users/maehne/Build/ghdl/src/grt/grt-hooks.adb
gcc -c -I./ -I/Users/maehne/Build/ghdl/./src/grt -I.. -gnatec/Users/maehne/Build/ghdl/./src/grt/grt.adc -gnat05 -gnatdY -g -I- /Users/maehne/Build/ghdl/src/grt/grt-stdio.ads
gcc -c -I./ -I/Users/maehne/Build/ghdl/./src/grt -I.. -gnatec/Users/maehne/Build/ghdl/./src/grt/grt.adc -gnat05 -gnatdY -g -I- /Users/maehne/Build/ghdl/src/grt/grt-change_generics.adb
gcc -c -I./ -I/Users/maehne/Build/ghdl/./src/grt -I.. -gnatec/Users/maehne/Build/ghdl/./src/grt/grt.adc -gnat05 -gnatdY -g -I- /Users/maehne/Build/ghdl/src/grt/grt-disp.adb
[... 113 lines cut ...]
a - grt/./grt-vcdz.o
a - jumps.o
a - times.o
a - grt-cbinding.o
a - grt-cvpi.o
a - fstapi.o
a - lz4.o
a - fastlz.o
[... cut ...]
sed -e "\!^.[/\\]!d" -e "/-shared/d" -e "/-static/d" -e "/-lgnat/d" \
	  -e "\X-L/Xd" < grt/grt-files > grt/grt-files.in
echo "@/libgrt.a" > grt.lst
cat grt/grt-files.in >> grt.lst
install -m 644 libgrt.a /usr/local/lib/ghdl/libgrt.a
install -m 644 grt.lst /usr/local/lib/ghdl/grt.lst
install -m 644 /Users/maehne/Build/ghdl/./src/grt/grt.ver /usr/local/lib/ghdl/grt.ver
install -m 644 -p libghdlvpi.dylib /usr/local/lib/
install -m 644 -p /Users/maehne/Build/ghdl/./src/grt/vpi_user.h /usr/local/include/
```

maehne, 2017-05-21:
Regarding libbacktrace. This is the call to GHDL's configure script:
```
$ ./configure --with-llvm-config=/opt/local/bin/llvm-config-mp-4.0 --with-backtrace-lib=/Users/maehne/Build/gcc-7.1.0/libbacktrace/.libs/libbacktrace.a
Build machine is: x86_64-apple-darwin14.5.0
Debugging is not enabled with llvm 4.0.0
Creating ghdl.gpr
Creating Makefile
Generate default_paths.ads
```
Before calling configure, I went into ```~/Build/gcc-7.1.0/libbacktrace/``` and did ```./configure && make``` there.

maehne, 2017-05-21:
This is the GHDL error message that I'm getting with LLVM backend and libbacktrace configured:
```
$ ghdl -a ../vhdl/BCDTo7Seg.vhdl ../vhdl/BCDTo7SegTb.vhdl
$ ghdl -e BCDTo7SegTb
gcc: error: /usr/local/lib/ghdl/v93/std/std_standard.o: No such file or directory
gcc: error: /usr/local/lib/ghdl/libbacktrace.a: No such file or directory
ghdl:error: compilation error
```

maehne, 2017-05-22:
I looked into the issue a bit further and discovered that ```std_standard.o``` is correctly generated in ```lib/ghdl/v??/std``` for each standard version only when I do a single-thread build with ```make```! Then, ```make install``` will also copy correctly all files so ghdl with LLVM backend works.

Once I do a parallel build with ```make -j4```, then there's a single ```std_standard.o``` generated in the root directory of the GHDL sources. So it may be some kind of race condition? This may be caused by not correctly listing all dependencies for the different targets in the Makefile.

I also noticed that ```make```when called repeatedly will compile the same targets over and over again instead of not touching them because they are already up-to-date.

Regarding ```libbacktrace.a```, even when doing a single-thead build and install of GHDL using just ```make && make install```, I get the following error on my test case:
```
$ ghdl -e BCDTo7SegTb
gcc: error: /usr/local/lib/ghdl/libbacktrace.a: No such file or directory
ghdl:error: compilation error
```
Sure enough libbacktrace.a did not get copied to that location.

maehne, 2017-05-22:
One minor thing, which I also noted: ```make clean``` does not remove ```libghdlvpi.dylib```, because the ```*.dylib``` is missing as an argument to ``rm`` in the ```clean``` target.

tgingold, 2017-05-24:
Should be fixed now.

closed 2017-05-24 by tgingold

maehne, 2017-05-25:
Thanks Tristan, I tested the GHDL HEAD (commit b572c73) on my side and can confirm that ```std_standard.o``` is now reliably built even when using ```make -j4```.

However, I still observe that many targets get rebuild ```make``` or ```make -j4``` multiple times. So some dependencies in the Makefile are still not quite right.

Also, ```libbacktrace.a``` doesn't seem to get correctly picked up by the Makefile. It doesn't get installed and even after a manual copy to the destination, GHDL simulations don't show the promised backtrace when assertion errors occur.

I will open separate issues.