ghdl/ghdl #353: Vunit compile script compile-vunit.sh error

Elvegard, 2017-05-23:
There are some errors when using the compile script for Vunit (compile-vunit.sh), that occurs when the files are being compiled for vunit_lib.
 
**Wrong path in script:**
- run/src/stop_api.vhd -> core/src/stop_api.vhd
- run/src/stop_body_2008.vhd -> core/src/stop_body_2008.vhd

**Listed in script but missing in Vunit:** 
- vhdl/src/lib/std/textio.vhd

**Missing in compile script:**
- array/src/integer_vector_ptr_pkg.vhd
- array/src/integer_array_pkg.vhd
- core/src/core_pkg.vhd

( GHDL 0.34-dev (2017-03-01-98-ge824379) [Dunoon edition]  Compiled with GNAT Version: 4.6)

tgingold, 2017-05-24:
Should be fixed.

Paebbels, 2017-05-31:
The PowerShell version of the script needs changes too.

kraigher, 2017-07-17:
Why do you guys have a script to compile VUnit libraries? VUnit will compile itself using its own Python infrastructure. I do not understand the use case. Seems hard to maintain a hardcoded list of files within VUnit it would be more robust to ask VUnit Python library what files is has.

Paebbels, 2017-09-05:
@tgingold Do we want to remove the pre-compilation scripts for VUnit? Most people seam to build these files via VUnit itself, so no pre-compiled version is required.

tgingold, 2017-09-06:
Make sense.

----- Mail original -----

commit b572c73c (2017-05-24): Update compile-vunit.sh script.

Fix #353
commit 46c50154 (2018-03-14): Removed VUnit pre-compile scripts from GHDL.

Fixes #353, #534, #528.
closed 2018-03-14 by tgingold