ghdl/ghdl #354: Repeated calls to make cause many targets to recompile unnecessarily

maehne, 2017-05-25:
I configure GHDL (at commit  b572c73) on OS X El Capitan with GNAT GPL 2016 in the following way:
```sh
$ ./configure --with-llvm-config=/opt/local/bin/llvm-config-mp-4.0 --disable-werror
Build machine is: x86_64-apple-darwin14.5.0
Debugging is not enabled with llvm 4.0.0
Creating ghdl.gpr
Creating Makefile
mkdir -p /usr/local/lib/ghdl
for d in src/std src/ieee src/vital95 src/vital2000 src/synopsys src/mentor src/ieee2008 v87/std v87/ieee v87/synopsys v93/std v93/ieee v93/mentor v93/synopsys v08/std v08/ieee v08/synopsys ; do \
	  mkdir -p lib/ghdl/$d; \
	done
Generate default_paths.ads
```
After that I build GHDL by calling ```make``` giving the following output:
```sh
$ make
if [ -d ./.git ]; then \
	  desc=`cd .; git describe --dirty`; \
	  sed -e "s/[(].*[)]/($desc)/" < src/version.in > version.tmp; \
	else \
	  cp src/version.in version.tmp; \
	fi
if [ ! -r version.ads ] || ! cmp version.tmp version.ads > /dev/null; then cp version.tmp version.ads; fi
/Applications/Xcode.app/Contents/Developer/usr/bin/make -f ./src/ortho/llvm4-nodebug/Makefile \
	 ortho_srcdir=./src/ortho ortho_exec=ghdl1-llvm \
	 GNAT_FLAGS="-aI./src -aI./src/vhdl -aI./src/grt -aI./src/psl -aI./src/vhdl/translate -aI./src/ghdldrv -aI./src/ortho -aI./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata" LDFLAGS="-Wl,-no_compact_unwind" \
	 LLVM_CONFIG="/opt/local/bin/llvm-config-mp-4.0" CXX="clang++" \
	 all
clang++ -c `/opt/local/bin/llvm-config-mp-4.0 --cxxflags` -o llvm-cbindings.o src/ortho/llvm4-nodebug/llvm-cbindings.cpp
gnatmake -o ghdl1-llvm -aI./src/ortho/llvm4-nodebug -aI./src/ortho \
	-aI./src -aI./src/vhdl -aI./src/grt -aI./src/psl -aI./src/vhdl/translate -aI./src/ghdldrv -aI./src/ortho -aI./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata ortho_code_main -bargs -E \
	-largs llvm-cbindings.o --LINK=clang++ \
	-Wl,-no_compact_unwind `/opt/local/bin/llvm-config-mp-4.0 --ldflags --libs --system-libs`
gcc -c -I./ -I./src/ortho/llvm4-nodebug -I./src/ortho -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata -I- /Users/maehne/Build/ghdl/src/ortho/llvm4-nodebug/ortho_code_main.adb
gcc -c -I./ -I./src/ortho/llvm4-nodebug -I./src/ortho -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata -I- /Users/maehne/Build/ghdl/src/ortho/llvm4-nodebug/llvm.ads
gcc -c -I./ -I./src/ortho/llvm4-nodebug -I./src/ortho -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata -I- /Users/maehne/Build/ghdl/src/ortho/llvm4-nodebug/llvm-analysis.ads
gcc -c -I./ -I./src/ortho/llvm4-nodebug -I./src/ortho -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata -I- /Users/maehne/Build/ghdl/src/ortho/llvm4-nodebug/llvm-bitwriter.ads
gcc -c -I./ -I./src/ortho/llvm4-nodebug -I./src/ortho -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata -I- /Users/maehne/Build/ghdl/src/ortho/llvm4-nodebug/llvm-core.ads
gcc -c -I./ -I./src/ortho/llvm4-nodebug -I./src/ortho -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata -I- /Users/maehne/Build/ghdl/src/ortho/llvm4-nodebug/llvm-target.ads
gcc -c -I./ -I./src/ortho/llvm4-nodebug -I./src/ortho -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata -I- /Users/maehne/Build/ghdl/src/ortho/llvm4-nodebug/llvm-targetmachine.ads
gcc -c -I./ -I./src/ortho/llvm4-nodebug -I./src/ortho -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata -I- /Users/maehne/Build/ghdl/src/ortho/llvm4-nodebug/llvm-transforms.ads
gcc -c -I./ -I./src/ortho/llvm4-nodebug -I./src/ortho -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata -I- /Users/maehne/Build/ghdl/src/ortho/llvm4-nodebug/llvm-transforms-scalar.ads
gcc -c -I./ -I./src/ortho/llvm4-nodebug -I./src/ortho -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata -I- /Users/maehne/Build/ghdl/src/vhdl/translate/ortho_front.adb
gcc -c -I./ -I./src/ortho/llvm4-nodebug -I./src/ortho -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata -I- /Users/maehne/Build/ghdl/src/ortho/llvm4-nodebug/ortho_llvm.adb
gcc -c -I./ -I./src/ortho/llvm4-nodebug -I./src/ortho -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata -I- /Users/maehne/Build/ghdl/src/bug.adb
gcc -c -I./ -I./src/ortho/llvm4-nodebug -I./src/ortho -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata -I- /Users/maehne/Build/ghdl/src/vhdl/configuration.adb
gcc -c -I./ -I./src/ortho/llvm4-nodebug -I./src/ortho -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata -I- /Users/maehne/Build/ghdl/src/vhdl/errorout.adb
gcc -c -I./ -I./src/ortho/llvm4-nodebug -I./src/ortho -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata -I- /Users/maehne/Build/ghdl/src/flags.adb
gcc -c -I./ -I./src/ortho/llvm4-nodebug -I./src/ortho -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata -I- /Users/maehne/Build/ghdl/src/vhdl/iirs.adb
gcc -c -I./ -I./src/ortho/llvm4-nodebug -I./src/ortho -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata -I- /Users/maehne/Build/ghdl/src/vhdl/iirs_utils.adb
gcc -c -I./ -I./src/ortho/llvm4-nodebug -I./src/ortho -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata -I- /Users/maehne/Build/ghdl/src/libraries.adb
gcc -c -I./ -I./src/ortho/llvm4-nodebug -I./src/ortho -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata -I- /Users/maehne/Build/ghdl/src/name_table.adb
gcc -c -I./ -I./src/ortho/llvm4-nodebug -I./src/ortho -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata -I- /Users/maehne/Build/ghdl/src/options.adb
gcc -c -I./ -I./src/ortho/llvm4-nodebug -I./src/ortho -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata -I- /Users/maehne/Build/ghdl/src/vhdl/sem.adb
gcc -c -I./ -I./src/ortho/llvm4-nodebug -I./src/ortho -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata -I- /Users/maehne/Build/ghdl/src/vhdl/std_package.adb
gcc -c -I./ -I./src/ortho/llvm4-nodebug -I./src/ortho -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g  [... cut ...]
echo "@/libgrt.a" > lib/ghdl/grt.lst
cat grt/grt-files.in >> lib/ghdl/grt.lst
install -m 644 lib/ghdl/libgrt.a /usr/local/lib/ghdl/libgrt.a
install -m 644 lib/ghdl/grt.lst /usr/local/lib/ghdl/grt.lst
install -m 644 /Users/maehne/Build/ghdl/./src/grt/grt.ver /usr/local/lib/ghdl/grt.ver
test "x" = x || test "xllvm" = xgcc || \
	  install -m 644  /usr/local/lib/ghdl/libbacktrace.a
install -m 644 -p libghdlvpi.dylib /usr/local/lib/
install -m 644 -p /Users/maehne/Build/ghdl/./src/grt/vpi_user.h /usr/local/include/
```

tgingold, 2017-05-29:
It looks like the issue affects only grt.  I will try to reproduce it.

tgingold, 2017-06-18:
I cannot reproduce it, so I close this issue.
If this is a real annoyance, I can give you some ideas to investigate it.

closed 2017-06-18 by tgingold