ghdl/ghdl #355: Building GHDL with LLVM backend and libbacktrace gives build error

maehne, 2017-05-25:
I configure the GHDL HEAD (at commit b572c73) using the following command on Mac OS X El Capitan with GNAT GPL 2016 for using the LLVM backend and libbacktrace (specified through an absolute path):
```sh
$ ./configure --with-llvm-config=/opt/local/bin/llvm-config-mp-4.0 --disable-werror --with-backtrace-lib=/Users/maehne/Build/gcc-7.1.0/libbacktrace/.libs/libbacktrace.a
Build machine is: x86_64-apple-darwin14.5.0
Debugging is not enabled with llvm 4.0.0
Creating ghdl.gpr
Creating Makefile
mkdir -p /usr/local/lib/ghdl
for d in src/std src/ieee src/vital95 src/vital2000 src/synopsys src/mentor src/ieee2008 v87/std v87/ieee v87/synopsys v93/std v93/ieee v93/mentor v93/synopsys v08/std v08/ieee v08/synopsys ; do \
	  mkdir -p lib/ghdl/$d; \
	done
Generate default_paths.ads
```
Building GHDL gives an error about ```make``` lacking a rule to make target ```lib/ghdl/libbacktrace.a```:
```sh
$ make
if [ -d ./.git ]; then \
	  desc=`cd .; git describe --dirty`; \
	  sed -e "s/[(].*[)]/($desc)/" < src/version.in > version.tmp; \
	else \
	  cp src/version.in version.tmp; \
	fi
if [ ! -r version.ads ] || ! cmp version.tmp version.ads > /dev/null; then cp version.tmp version.ads; fi
/Applications/Xcode.app/Contents/Developer/usr/bin/make -f ./src/ortho/llvm4-nodebug/Makefile \
	 ortho_srcdir=./src/ortho ortho_exec=ghdl1-llvm \
	 GNAT_FLAGS="-aI./src -aI./src/vhdl -aI./src/grt -aI./src/psl -aI./src/vhdl/translate -aI./src/ghdldrv -aI./src/ortho -aI./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata" LDFLAGS="-Wl,-no_compact_unwind" \
	 LLVM_CONFIG="/opt/local/bin/llvm-config-mp-4.0" CXX="clang++" \
	 all
clang++ -c `/opt/local/bin/llvm-config-mp-4.0 --cxxflags` -o llvm-cbindings.o src/ortho/llvm4-nodebug/llvm-cbindings.cpp
gnatmake -o ghdl1-llvm -aI./src/ortho/llvm4-nodebug -aI./src/ortho \
	-aI./src -aI./src/vhdl -aI./src/grt -aI./src/psl -aI./src/vhdl/translate -aI./src/ghdldrv -aI./src/ortho -aI./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata ortho_code_main -bargs -E \
	-largs llvm-cbindings.o --LINK=clang++ \
	-Wl,-no_compact_unwind `/opt/local/bin/llvm-config-mp-4.0 --ldflags --libs --system-libs`
gcc -c -I./ -I./src/ortho/llvm4-nodebug -I./src/ortho -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata -I- /Users/maehne/Build/ghdl/src/ortho/llvm4-nodebug/ortho_code_main.adb
gcc -c -I./ -I./src/ortho/llvm4-nodebug -I./src/ortho -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata -I- /Users/maehne/Build/ghdl/src/ortho/llvm4-nodebug/llvm.ads
gcc -c -I./ -I./src/ortho/llvm4-nodebug -I./src/ortho -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata -I- /Users/maehne/Build/ghdl/src/ortho/llvm4-nodebug/llvm-analysis.ads
gcc -c -I./ -I./src/ortho/llvm4-nodebug -I./src/ortho -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata -I- /Users/maehne/Build/ghdl/src/ortho/llvm4-nodebug/llvm-bitwriter.ads
gcc -c -I./ -I./src/ortho/llvm4-nodebug -I./src/ortho -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata -I- /Users/maehne/Build/ghdl/src/ortho/llvm4-nodebug/llvm-core.ads
gcc -c -I./ -I./src/ortho/llvm4-nodebug -I./src/ortho -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata -I- /Users/maehne/Build/ghdl/src/ortho/llvm4-nodebug/llvm-target.ads
gcc -c -I./ -I./src/ortho/llvm4-nodebug -I./src/ortho -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata -I- /Users/maehne/Build/ghdl/src/ortho/llvm4-nodebug/llvm-targetmachine.ads
gcc -c -I./ -I./src/ortho/llvm4-nodebug -I./src/ortho -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata -I- /Users/maehne/Build/ghdl/src/ortho/llvm4-nodebug/llvm-transforms.ads
gcc -c -I./ -I./src/ortho/llvm4-nodebug -I./src/ortho -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata -I- /Users/maehne/Build/ghdl/src/ortho/llvm4-nodebug/llvm-transforms-scalar.ads
gcc -c -I./ -I./src/ortho/llvm4-nodebug -I./src/ortho -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata -I- /Users/maehne/Build/ghdl/src/vhdl/translate/ortho_front.adb
gcc -c -I./ -I./src/ortho/llvm4-nodebug -I./src/ortho -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata -I- /Users/maehne/Build/ghdl/src/ortho/llvm4-nodebug/ortho_llvm.adb
gcc -c -I./ -I./src/ortho/llvm4-nodebug -I./src/ortho -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata -I- /Users/maehne/Build/ghdl/src/bug.adb
gcc -c -I./ -I./src/ortho/llvm4-nodebug -I./src/ortho -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata -I- /Users/maehne/Build/ghdl/src/vhdl/configuration.adb
gcc -c -I./ -I./src/ortho/llvm4-nodebug -I./src/ortho -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata -I- /Users/maehne/Build/ghdl/src/vhdl/errorout.adb
gcc -c -I./ -I./src/ortho/llvm4-nodebug -I./src/ortho -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata -I- /Users/maehne/Build/ghdl/src/flags.adb
gcc -c -I./ -I./src/ortho/llvm4-nodebug -I./src/ortho -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata -I- /Users/maehne/Build/ghdl/src/vhdl/iirs.adb
gcc -c -I./ -I./src/ortho/llvm4-nodebug -I./src/ortho -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata -I- /Users/maehne/Build/ghdl/src/vhdl/iirs_utils.adb
gcc -c -I./ -I./src/ortho/llvm4-nodebug -I./src/ortho -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata -I- /Users/maehne/Build/ghdl/src/libraries.adb
gcc -c -I./ -I./src/ortho/llvm4-nodebug -I./src/ortho -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata -I- /Users/maehne/Build/ghdl/src/name_table.adb
gcc -c -I./ -I./src/ortho/llvm4-nodebug -I./src/ortho -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata -I- /Users/maehne/Build/ghdl/src/options.adb
gcc -c -I./ -I./src/ortho/llvm4-nodebug -I./src/ortho -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata -I- /Users/maehne/Build/ghdl/src/vhdl/sem.adb
gcc -c -I./ -I./src/ortho/llvm4-nodebug -I./src/ortho -I./src -I./src/vhdl -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm4-nodebug -gnaty3befhkmr -gnatwa -gnatf -gnat05 -g -gnata -I- /Users/maehne/Build/ghdl/src/vhdl/std_package.adb
[... cut ...]
a - grt-cbinding.o
a - grt-cvpi.o
a - fstapi.o
a - lz4.o
a - fastlz.o
a - grt/run-bind.o
a - grt/main.o
/opt/local/bin/ranlib: file: lib/ghdl/libgrt.a(grt-backtraces-impl.o) has no symbols
/opt/local/bin/ranlib: file: lib/ghdl/libgrt.a(grt-threads.o) has no symbols
ranlib lib/ghdl/libgrt.a
ranlib: file: lib/ghdl/libgrt.a(grt-backtraces-impl.o) has no symbols
ranlib: file: lib/ghdl/libgrt.a(grt-threads.o) has no symbols
make: *** No rule to make target `lib/ghdl/libbacktrace.a', needed by `grt-all'.  Stop.
```

Brian-Drummond, 2017-10-02:
I have just hit this error trying to build GHDL with the LLVM backend on Debian (Stretch) with gcc6.3 and LLVM3.8, both the Stretch defaults.

As far as I can understand the Makefiles, `libbacktrace.a` is simply copied from the location originally supplied by `--with-backtrace-lib=`,  to `lib/ghdl/libbacktrace.a`
For whatever reason, this copy doesn't seem to happen (though the file exists, and permissions seem OK). Then something dependent on `lib/ghdl/libbacktrace.a` can't find it, and the error is reported.

Now at this point I pig-headedly copied the file by hand, and re-ran make.
Remarkably, it proceeded to completion, and make install worked, and the resulting ghdl seems to work correctly.
(Unlike ghdl-llvm built without the `--with-backtrace-lib` option, which fails at elaboration with the error below)
```
/opt/ghdl/llvm/lib/ghdl/libgrt.a(grt-backtraces-gcc.o): In function `grt__backtraces__gcc__symbolizer':
/home/brian/Projects/ghdl/2017/ghdl/src/grt/grt-backtraces-gcc.adb:117: undefined reference to `backtrace_create_state'
/home/brian/Projects/ghdl/2017/ghdl/src/grt/grt-backtraces-gcc.adb:126: undefined reference to `backtrace_pcinfo'
collect2: error: ld returned 1 exit status
```
So I think there are two problems : (a) without `--with-backtrace-lib` the runtime is still calling the non-existent backtrace functions and (b) either Makefile or src/grt/Makefile.inc has some dependency information missing.

Brian-Drummond, 2017-10-05:
Fix now merged ... https://github.com/tgingold/ghdl/pull/428
should we close this?

eine, 2021-02-02:
@maehne, can you please confirm whether this is fixed?