ghdl/ghdl #356: Concurrent selected signal assignment fails with "type mark is not a locally static subtype"

m-thu, 2017-05-27:
Analyzing the following code snippet with `ghdl -a --std=08 ex_5_19.vhd` results in this error message (tested with commit e82437906ec350554835a7c8ba288a41e167fa14):
```
ex_5_19.vhd:11:13:error: type mark is not a locally static subtype
```
To my understanding, this should be valid in VHDL-2008.

```vhdl
-- ghdl -a --std=08 ex_5_19.vhd

entity mux4 is
        port ( a0, a1, a2, a3 : in bit;
               sel0, sel1 : in bit;
               y : out bit );
end entity mux4;

architecture behav of mux4 is
begin   
        with bit_vector'(sel0, sel1) select
                y <= a0 when "00",
                     a1 when "01",
                     a2 when "10",
                     a3 when "11";
end architecture behav;
```

tgingold, 2017-05-29:
This vhdl-08 feature is not yet implemented.

tgingold, 2017-10-22:
Now implemented.

closed 2017-10-22 by tgingold

ZX-SPECTRUM128, 2017-12-18:
tgingold you SAID "NOW Implemented" but i have just installed ghdl 0.33 and i am having the same problem.
could you tell me how i can catch up with this newly implemented feature?

Paebbels, 2017-12-18:
As you can see in the sidebar of this issue, it was fixed in [version 0.35](https://github.com/tgingold/ghdl/milestone/3?closed=1), which was releases a few days ago. So you need to install [GHDL 0.35](https://github.com/tgingold/ghdl/releases/tag/v0.35) on your machine.

ZX-SPECTRUM128, 2017-12-18:
thank you for the quick answer 
actually i am extremely new to using ghdl  (i learned VHDL with aldec active-hdl) and i have just installed ghdl the day before yesterday after 4 hours of battling with its installation. i tried the normal way simple way of installing ghdl  `sudo apt-get install ghdl` but it didn't work 
so i tried  these instructions: [https://theintobooks.wordpress.com/2016/08/24/installing-ghdl-from-source-on-ubuntu-16-04/](url)
it worked after 4 hours of trail ad error 

so could you please tell em hoe to update my ghdl with latest version 0.35?

thanks a lot

svenn71, 2017-12-18:
Clone the repository here on github, and not the one on sourceforge.

Paebbels, 2017-12-18:
Just download a suitable zip package from [GHDL's GitHub releases](https://github.com/tgingold/ghdl/releases/tag/v0.35) page and extract the content.

Or follow the instructions to compile GHDL with [LLVM backend](http://ghdl.readthedocs.io/en/latest/building/llvm/GNULinux-GNAT.html) (very simple).

ZX-SPECTRUM128, 2017-12-18:
if i clone it with gcc what would i do with the previous ghdl 0.33 that i have should i leave as is ?

ZX-SPECTRUM128, 2017-12-18:
ok so that is what i did:

`git [clone](https://github.com/tgingold/ghdl.git) 
cd ~/ghdl
mkdir build
cd build
 ../configure --prefix=/opt/ghdl
make
make install`

and yet i am having the same problem 
  ghdl --version
GHDL 0.33 (20150921) [Dunoon edition]
 Compiled with GNAT Version: 4.9.3
 GCC back-end code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2015 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

1138-4EB, 2017-12-18:
If you repeated exactly those steps, either:

- Something failed, such as `make install`, which you didn't realize.
- You need to execute `/opt/ghdl/bin/ghdl --version`, to make user that you are not executing the older version from `/usr/bin` or `/usr/local/bin`.

ZX-SPECTRUM128, 2017-12-18:
ok i got it fixed finally and ghdl -version id showing me :  
GHDL 0.36-dev (v0.35-20-g1d61cbb) [Dunoon edition]

and now if i type : 
`ghdl -s --std=08 my_file.vhd`
i don't get : :error: type mark is not a locally static subtype
 BUT
within vim iam still having the same error message each time i save my source code 
is possible to configure vim to compile using 2008 vhdl

1138-4EB, 2017-12-18:
> is possible to configure vim to compile using 2008 vhd

I believe, this is out of scope of this issue and, probably, of GHDL itself, as it is related to how you configured your editor. Being vim I'd say that there is some option to add `--std=08`, but it is difficult to guess without knowing which plugins/shortcuts are you actually using.

ZX-SPECTRUM128, 2017-12-18:
actually i am using syntactic plugin

1138-4EB, 2017-12-18:
I think that 'synctactic' does not exists. Not as vim plugin, at least.

Please, have a look at [How To Ask Questions The Smart Way](http://www.catb.org/~esr/faqs/smart-questions.html) and take your time before posting. We falling towards [lmgtfy](http://lmgtfy.com/?q=vim+syntactic+ghdl).

https://github.com/vim-syntastic/syntastic/blob/master/syntax_checkers/vhdl/ghdl.vim#L22

tgingold, 2017-12-21:
Maybe you can use 'ghdl -i' for syntactic check.