ghdl/ghdl #357: Parameter --stop-time does not have any effect

darkdragon-001, 2017-05-27:
The following command misses the last signal change of `b` at _26 ns_ from _7_ to _8_

    ghdl -r main --vcd=main.vcd --stop-time=40ns

The source code is as follows

```
library IEEE;
use IEEE.STD_LOGIC_1164.ALL;

ENTITY main IS
END main;

ARCHITECTURE behavioral OF main IS
  signal a : std_logic;
  signal b : integer := 0;
BEGIN
  a <= TRANSPORT '0' AFTER  0 ns, '1' AFTER  3 ns,
                 '0' AFTER  7 ns, '1' AFTER 10 ns,
                 '0' AFTER 18 ns, '1' AFTER 25 ns;
  p1 : PROCESS (a,b)
  BEGIN
    IF a='0' THEN
      b <= (b+1) AFTER 2 ns;
    END IF;
  END PROCESS;
END behavioral;
```

diogratia, 2017-05-28:
Add monitor processes:
```vhdl
monitor_a:
    process (a)
    begin
        report "a = " & std_logic'image(a);
    end process;
monitor_b:
    process (b)
    begin
        report "b = " & integer'image(b);
    end process;
```

And:

> main.vhdl:24:9:@0ms:(report note): a = 'U'  
> main.vhdl:29:9:@0ms:(report note): b = 0  
> main.vhdl:24:9:@0ms:(report note): a = '0'  
> main.vhdl:29:9:@2ns:(report note): b = 1  
> main.vhdl:24:9:@3ns:(report note): a = '1'  
> main.vhdl:29:9:@4ns:(report note): b = 2  
> main.vhdl:24:9:@7ns:(report note): a = '0'  
> main.vhdl:29:9:@9ns:(report note): b = 3  
> main.vhdl:24:9:@10ns:(report note): a = '1'  
> main.vhdl:29:9:@11ns:(report note): b = 4  
> main.vhdl:24:9:@18ns:(report note): a = '0'  
> main.vhdl:29:9:@20ns:(report note): b = 5  
> main.vhdl:29:9:@22ns:(report note): b = 6  
> main.vhdl:29:9:@24ns:(report note): b = 7  
> main.vhdl:24:9:@25ns:(report note): a = '1'  
> main.vhdl:29:9:@26ns:(report note): b = 8  

your transition at 26 ns shows up. It would also show up with gtkwave, GHW zoomed to fit (but zoomed in the last transaction doesn't appear to show up - gtkwave v3.3.71).

Please describe the version of ghdl you are using and demonstrate how you are determining the transition on b is missing.

This was done with a recent version of ghdl:

> ghdl -v  
> GHDL 0.34-dev (2017-03-02-27-g1814966) [Dunoon edition]  
>  Compiled with GNAT Version: GPL 2015 (20150428-49)  
>  llvm code generator

darkdragon-001, 2017-05-28:
Thanks for your hint with the monitor processes!

I had a look in the bare `.vcd` file noticing that it contains the transition at 26 ns. 

Therefore the problems seems to lie in GTKWave since it stops displaying anything after the last transition in contrast to VHDL where signals are assumed to stay constant after the last transition.

diogratia, 2017-05-28:
A waveform dump conveys transactions at specific times not duration.  Simulation time does not advance at a linear rate, VHDL is event driven, and events are dumped to waveform viewers.

When there are no more events at the current simulation time, simulation time is advanced to the next time a signal is scheduled to be updated. When there are no more events scheduled, simulation time is advanced to TIME'HIGH and simulation ceases.

Your last projected output waveform change occurred at 26 ns. Simulation ended with no more scheduled output waveform transactions pending and not because of the stop-time argument.

This is a usage issue rather than a tool issue. Provide a signal event at the time you want your simulation to end for the waveform to show that time.

The only 'bug' I see in gtkwave is as mentioned, that the last transition at 26 ns was only visible in a displayed waveform zoomed to fit (the full waveform). Whether that is specific to GHW hasn't been demonstrated, the transition is in both a GHW dump and as you related a VCD dump.

closed 2017-05-28 by darkdragon-001