ghdl/ghdl #362: configure script aborts in MinGW, while comparing `srcdir`

Paebbels, 2017-05-31:
The `configure` script aborts execution with this error message:

```Bash
$ ../../configure
Cannot compute absolute source dir (or incorrect srcdir)
srcdir=../..
abs_srcdir=/c/git/GitHub/Paebbels/ghdl/build/mingw64-llvm40/../..
```

Paebbels, 2017-05-31:
Argh. MinGW's base installation doesn't contain tools like:
- **cmd** => difftools
  installation command: `pacman -S msys/diffutils`
- **make** => make
  installation command: `pacman -S msys/make`

Is it possible to check if these tools exist before using them?
(And display the needed package name if known?)

tgingold, 2017-06-01:
There are two different issues:

* configure on mingw:  I prefer to handle that by documentation
* fstapi.c on win64 -> this file comes from gtkwave, so maybe this was already fixed, or that's the place to fix it.

Paebbels, 2017-06-01:
I have seen some configure scripts checking all preconditions. Are these configure scripts auto generated by one of the "auto" tools?

Paebbels, 2017-06-02:
I wrote an email to the GTKwave mailing list. This was the answer:

> I looked at the source for `fstapi.c` found in gtkwave SVN and it uses `intptr_t` where the compiler is complaining. The GHDL codebase might need to be updated.

tgingold, 2017-06-07:
I have updated fstapi.

commit 1ffa7bee (2017-06-12): Document requirements for MSYS.

Close #362
closed 2017-06-12 by tgingold