ghdl/ghdl #365: MinGW64 + LLVM build failed on AppVeyor

Paebbels, 2017-06-01:
My local MinGW setup doesn't compile GHDL if the LLVM backend is selected. It complains about a missing build rule for `ghdl_main`. I looked into the AppVeyor logs and found the [same issue](https://ci.appveyor.com/project/tgingold/ghdl/build/0.34-dev-267/job/ewc81o0h14aq2kul#L483).

I don't know why the AppVeyor flow:
* doesn't recognize the error from make and stops
* why there is a difference for `ghdl_main` depending the chosen backend (mcode vs. llvm)

tgingold, 2017-06-02:
Looks like a path issue: /c/projects/... is not a windows path.

Paebbels, 2017-06-02:
This path is looks valid to me, because MinGW translates paths like `C:\projects` to `/c/projects`. CygWin would have used `/cygdrive/c/projects`.

tgingold, 2017-06-02:
On 02/06/17 07:50, Patrick Lehmann wrote:
> This path is looks valid to me, because MinGW translates paths like
> |C:\projects| to |/c/projects|. CygWin would have used
> |/cygdrive/c/projects|.

I am pretty sure that /c/projects is not a valid windows path, therefore 
ignored by gnatmake.

Paebbels, 2017-06-02:
I manually called `gnatmake` on my local computer from within the `grt` directory in the build directory with all paths converted to `C:\...`, but there is no difference.

tgingold, 2017-06-03:
Log please!  I you are calling from bash, don't forget to escape '\', ie write c:\\aaa\\bbb
Use of '/' should be ok (ie: c:/aaa/bbb/...)

Paebbels, 2017-06-07:
I compared the current compilation output logs with older logs from AppVeyor. The call to `gnatmake` uses absolute paths, whereas the older version used relative paths.

Old:
```
gnatmake -c -aI../../src/grt -gnatec../../src/grt/grt.adc -gnat05 -gnatdY ghdl_main  -cargs -g
```

New:
```
gnatmake -v -c -aI/c/git/GitHub/Paebbels/ghdl/build/mingw64-llvm40/../../src/grt -aI.. -gnatec/c/git/GitHub/Paebbels/ghdl/build/mingw64-llvm40/../../src/grt/grt.adc -gnat05 -gnatdY ghdl_main  -cargs -g
```

You are right, changing the path format to Windows paths and putting everything into double quotes solved the problem while compiling manually in the `<build>/grt` directory.

I echoed the make variable `GRTSRCDIR` in `grt\Makefile.inc:107`:
```make
grt/run-bind.adb: grt $(GRT_SRC_DEPS) grt-force
	cd grt; echo "-$(GRTSRCDIR)-"; $(GNATMAKE) -v -c "-aI$(GRTSRCDIR)" -aI.. $(GRT_PRAGMA_FLAG) \
	  ghdl_main $(GRT_ADAFLAGS) -cargs $(GRT_FLAGS)
```

It prints: `-/c/git/GitHub/Paebbels/ghdl/build/mingw64-llvm40/../../src/grt-`

As far as I can see, `grt/Makefile.inc` is included by `grt/Makefile`. The variable `GRTSRCDIR` is defined as `.` in line 25. I don't understand how dot can expand to a full absolute path.

commit f023bbb2 (2017-06-18): Avoid MSYS2 path translation.

Fix #365
closed 2017-06-18 by tgingold