ghdl/ghdl #370: Loading AXILite BFM package from UVVM causes an exception

Paebbels, 2017-06-18:
This is a followup issue on [#361 - Loading AXIStream BFM package from UVVM causes an exception](https://github.com/tgingold/ghdl/issues/361)

Yes, the issue for `bitvis_vip_axistream` is resolved, but the same happens for `bitvis_vip_axilite` again. I changed the title of the old question, because the presented example was actually about `axistream` and not `axilite`.

```
  ******************** GHDL Bug occured ****************************
  Please report this bug on https://github.com/tgingold/ghdl/issues
  GHDL release: GHDL 0.34-dev (commit: 2017-06-18;  git branch: paebbels/master';  hash: fb339de2) [Dunoon edition]
  Compiled with GNAT Version: GPL 2016 (20160515-49)
  Target: i686-pc-mingw32
  In directory: C:\git\GitHub\Paebbels\PoC\temp\ghdl\
  Command line:
  C:\Tools\GHDL\0.34-dev-gnatgpl32-mcode\bin\ghdl.exe -r -fexplicit -frelaxed-rules --warn-binding --no-vital-checks --mb-comments --syn-binding -fpsl --ieee=synopsys --std=08 --work=test -PC:\git\GitHub\Paebbels\PoC\temp\precompiled\ghdl\uvvm iic_Controller_tb --ieee-asserts=disable-at-0
  Exception TYPES.INTERNAL_ERROR raised
  Exception information:
  raised TYPES.INTERNAL_ERROR : trans.adb:2049
  ******************************************************************
```

tgingold, 2017-06-19:
A reproducer would be nice!

Paebbels, 2017-06-19:
**This is the reproducer:**

```VHDL
library bitvis_vip_axilite;
use     bitvis_vip_axilite.axilite_bfm_pkg.all;

entity uvvm_axilite is
end entity;

architecture tb of uvvm_axilite is
begin
end architecture;
```

**Executing GHDL:**
```PowerShell
> ghdl.exe -a --std=08 "-P..\precompiled\ghdl\uvvm" -frelaxed-rules .\uvvm_axilite.vhdl
> ghdl.exe -r --std=08 "-P..\precompiled\ghdl\uvvm" -frelaxed-rules uvvm_axilite

******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: GHDL 0.34-dev (commit: 2017-06-18;  git branch: paebbels/master';  hash: fb339de2) [Dunoon edition]
Compiled with GNAT Version: GPL 2016 (20160515-49)
Target: i686-pc-mingw32
In directory: C:\git\GitHub\Paebbels\PoC\temp\bugreport\
Command line:
C:\Tools\GHDL\0.34-dev-gnatgpl32-mcode\bin\ghdl.exe -r --std=08 -P..\precompiled\ghdl\uvvm -frelaxed-rules uvvm_axilite
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : trans.adb:2049
******************************************************************
```

commit 620ddf92 (2017-06-21): Translate: use create_temp for create_record_type_builder.

Fix #370
closed 2017-06-21 by tgingold