ghdl/ghdl #371: Handling of string in ghw and gtkwave

svenn71, 2017-06-19:
Wondering if it is ghdl or gtkwave (or me) 

who gets the string wrong in my small test case.?
[test_string.txt](https://github.com/tgingold/ghdl/files/1085548/test_string.txt)
```
-- VHDL source file
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test_string is
end entity;

architecture rtl of test_string is
    signal clk  : std_logic := '1';
    signal test : string (1 to 11) := "Hello World";
begin
    clk <= not clk after 10 ns;
    p_test : process
    begin
        wait for 1 us;
        test <= "hello world";
        wait;
    end process;
end rtl;
```
ghdl -i test_string.vhd
./test_string --stop-time=2us --wave=test_string.ghw
gtkwave test_string.ghw

tgingold, 2017-06-20:
The signals are correctly displayed when you expand it.
Still investigating.

tgingold, 2017-06-23:
Now fixed in Gtkwave.

closed 2017-06-23 by tgingold