ghdl/ghdl #372: Error when creating array of zero-width std_logic_vector

benreynwar, 2017-06-20:
```vhdl
library ieee;
use ieee.std_logic_1164.all;

entity Dummy is
end entity;

architecture arch of Dummy is
  subtype t_null is std_logic_vector(-1 downto 0);
  type array_of_nulls is array(1 downto 0) of t_null;
begin
end architecture;
```

ben@lorimer:~/Code/ghdl_testcase$ ghdl -a dummy.vhd 
ben@lorimer:~/Code/ghdl_testcase$ ghdl -e Dummy

******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: GHDL 0.34-dev (2017-03-01-142-g7b2da98) [Dunoon edition]
Compiled with GNAT Version: 4.6
Target: x86_64-pc-linux-gnu
In directory: /home/ben/Code/ghdl_testcase/
Command line:
ghdl -e Dummy
Exception PROGRAM_ERROR raised
Exception information:
Exception name: PROGRAM_ERROR
Message: ortho_code-exprs.adb:884 explicit raise
Call stack traceback locations:
0x546f9c 0x547050 0x547165 0x4c5cfa 0x618dcf 0x5b1492 0x5b529d 0x607f7d 0x5e268c 0x628a5d 0x5d25f2 0x62710c 0x5db56c 0x62b681 0x634019 0x6316c8 0x4c6708 0x408753 0x7f3de2855f43 0x407d17 0xfffffffffffffffe
******************************************************************

benreynwar, 2017-06-20:
I got something resembling a fix by changing line 912 of ortho_code-exprs.adb to
```
      if Is_Pow2 (El_Size) and (El_Size /= 0) then
```
, basically just adding in the /= 0 check since is_Pow2 thinks that 0 is a power of two.

tgingold, 2017-06-20:
Yes!  Thank you for the report and the fix.

commit 9a297666 (2017-06-21): Avoid a crash while slicing array of length 0.

Fix #372
closed 2017-06-21 by tgingold