ghdl/ghdl #373: is_valid_conversion(2): cannot handle IIR_KIND_INTERFACE_SIGNAL_DECLARATION

vicencb, 2017-06-22:
Hello,

if this code
```vhdl
entity x is
  port(
    t :out bit_vector(0 to 0);
    z :out bit_vector(0 to 0)
  );
end entity;
architecture a of x is begin end architecture;

entity e is end entity;
architecture a of e is
  constant z :integer := 0;
  subtype t is bit_vector(0 to 0);
  signal actual_for_t :bit;
  signal actual_for_z :t;
begin
  inst: entity work.x port map(
    t(z) => actual_for_t,
    t(z) => actual_for_z
  );
end architecture;
```
is compiled with `ghdl -a --std=08 e.vhd` then GHDL reports an error and then crashes.
Tested with 7b2da98caf3c389c4bd29df887050414597aeb7f.

Please, do not get me wrong, I am not reporting bugs just for the sake of it.
This idea has just ocurred to me and asked myself: What would happen if...?
If GHDL just reported an error, that would have been a valid answer to my question, but it crashed instead and asked me to report it.

In fact I am not even sure that the code is valid VHDL.
Strictly speaking we have:
1. The indexed name `t(z)` is ambiguous, but that is not an error because it is not a complete context.
1. The formal `t(z)` in the form of a type conversion *(which is not a type conversion)* is again ambiguous, but that is neither an error because it is not a complete context.
1. The component instantiation statement is a complete context and at this level there is enough information to resolve all ambiguities.

So, I think that this might be a valid VHDL description.
But I also recognize this as just an academical exercise with no practical use cases, so, would be happy if it is classified as erroneous.

Regards,
&nbsp;&nbsp;Vicenç.

Paebbels, 2017-06-22:
Shouldn't it be: `z => actual_for_z` ?
Otherwise the types don't match, right?

vicencb, 2017-06-22:
If lines 3 and 17 are commented out, it works fine:
```vhdl
entity x is
  port(
--  t :out bit_vector(0 to 0);
    z :out bit_vector(0 to 0)
  );
end entity;
architecture a of x is begin end architecture;

entity e is end entity;
architecture a of e is
  constant z :integer := 0;
  subtype t is bit_vector(0 to 0);
  signal actual_for_t :bit;
  signal actual_for_z :t;
begin
  inst: entity work.x port map(
--  t(z) => actual_for_t,
    t(z) => actual_for_z
  );
end architecture;
```
So, `t(z) => actual_for_z` is accepted as valid and types are correctly matched.

vicencb, 2017-06-22:
@Paebbels, do not assign a milestone to it, I do not know about GHDL internals, but this may not be easy to fix.

Paebbels, 2017-06-22:
`t(z)` is equivalent to `t(0)`. It's an index name, thus the LHS type is `bit`.

LHS: `actual_for_z` is a `bit_vector` of one element.

A scalar and an array can't match.

Paebbels, 2017-06-22:
@tgingold can rearrange milestones or classify it as some day maybe :). It's just a first classification idea.

vicencb, 2017-06-22:
There are two `t(z)`, the first one is, as you say, an indexed name where `t` is the formal and `z` is an integer constant.
The second one is like a type conversion where `t` is a type mark and `z` is the formal, so, the array `z` is type converted to the subtype `t` which are closely related and also match the type of `actual_for_z`.

tgingold, 2017-06-23:
Interesting case.
Artificial but still interesting.

tgingold, 2017-06-27:
I am testing a fix.
I am also quite sure that you could find other issues if you dig this direction...

closed 2017-06-29 by tgingold

vicencb, 2017-06-29:
Thank you!
With your last comment, are you encouraging me to dig?

Paebbels, 2017-06-29:
I would like to see some digging :).

<https://github.com/tgingold/ghdl/issues/373#issuecomment-312095362>, or mute
the thread

tgingold, 2017-06-30:
> Thank you!
> With your last comment, are you encouraging me to dig?

It's up to you!

vicencb, 2017-07-02:
Hello Tristan,
indeed you were right, there are more issues:
```vhdl
entity e is end entity;
architecture a of e is
begin
  process
    constant z :integer := 0;
    type t is array(0 to 0) of bit;
    procedure x(
      z :out bit_vector(0 to 0);
      f :out bit_vector(0 to 0)
    ) is
    begin
    end procedure;
    procedure x(
      z :out t;
      f :out bit_vector(0 to 0)
    ) is
    begin
    end procedure;
    function f(arg:t) return bit is
    begin
    end function;
    variable actual_for_f :bit;
    variable actual_for_z :t; -- bit
  begin
    x(
      f(z) => actual_for_f,
      f(z) => actual_for_z
    );
  end process;
end architecture;
```
This may or may not be related to #373, but it was found while trying to combine ambiguities due to overloading plus ambiguities due to namespaces.
There are two versions of this code:
 1. `variable actual_for_z :t;`
 1. `variable actual_for_z :bit;`

Both versions are incorrect:
 1. Type mismatch
 2. An unsolvable ambiguity arises

But, in both cases, it triggers a crash report:
`raised TYPES.INTERNAL_ERROR : sem_assocs.adb:2476`
If either `procedure x` is commented out, the issue remains, so, ambiguities due to overloading seem irrelevant here.
Tested with aeae722a.
Please, don't waste time on it, I would prefer full 08 support before a fix for this abomination.
In any case, GHDL is now more robust than before.

Thank you!
&nbsp;&nbsp;Vicenç.

Paebbels, 2017-07-02:
Hello @vicencb,

please open a new issue in GitHub for each possible bug. It helps us to keep track of what is solved and what not. Additionally, @tgingold can close issues directly from a commit by writing a special commit message. So the solving commit and the issue are then connected.

tgingold, 2017-09-25:
In fact the testcase is invalid.  I have fixed ghdl so that it behaves correctly now.

vicencb, 2017-09-27:
Hi Tristan,
thanks for the update.
Please, could you explain why the testcase is invalid?

Regards,
&nbsp;&nbsp;Vicenç.

tgingold, 2017-09-28:
The testcase was:

entity x is
  port(
    t :out bit_vector(0 to 0);
    z :out bit_vector(0 to 0)
  );
end entity;
architecture a of x is begin end architecture;

entity e is end entity;
architecture a of e is
  constant z :integer := 0;
  subtype t is bit_vector(0 to 0);
  signal actual_for_t :bit;
  signal actual_for_z :t;
begin
  inst: entity work.x port map(
    t(z) => actual_for_t,
    t(z) => actual_for_z
  );
end architecture;

Within the port map, both t and z refers to the ports.
So t(z) is invalid as port z is not a correct index for bit_vector.

Regards,
Tristan.

Paebbels, 2017-09-28:
Why does in

    inst: entity work.x port map(
      t(z) => actual_for_t,
      t(z) => actual_for_z
    );

the `z` in `t(z)` refer to an other actual instead to constant `z`? Only the left most prefix of an actual's name should refer to the matching port name, right?

`t(z)` is a selected name of prefix `t` and index `z`. So `t` should be used as a lookup for the port name.

It would be ambiguous if you use `t(z'subtype'range)` as a slice name, where it might be possible to use the range from either `z` the port or `z` the constant. But given that example, both `z` would be visible, so `z` (the port) could act as an index ... Now, I'm confused.

vicencb, 2017-09-28:
Hi Tristan,
thank you for the explanation, now I see why (although it took a while...).

Hi Paebbels,
to get an understanding of why, read the following in the VHDL08 LRM:
page 187: A declaration is visible by selection at
page 188: n) For a formal port declaration: at the place of the formal part (before the compound delimiter =>) of a port association
page 189: At a place in which a given declaration is visible by selection, every declaration with the same designator is hidden

Regards,
&nbsp;&nbsp;Vicenç.

commit f0d19c32 (2017-06-29): sem_assocs: correctly check both possibilities for parenthesis formal.

Fix #373