ghdl/ghdl #378: Exception CONSTRAINT_ERROR raised

lyd1ng, 2017-07-02:
******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/tgingold/ghdl/issues
GHDL release: GHDL 0.34-dev (2017-03-01-74-g669d25e0) [Dunoon edition]
Compiled with GNAT Version: 6.3.1 20170306
Target: x86_64-pc-linux-gnu
In directory: /path/uhr_repo/
Command line:
ghdl -r clock_testbench --vcd=test.vcd
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : grt-processes.adb:761 overflow check failed
Call stack traceback locations:
0x468daf 0x46957d 0x4697de 0x4699b5 0x4051a2 0x47eaa8 0x6693a6 0x66f2bb 0x66dee4 0x4e4793 0x404e
******************************************************************

Paebbels, 2017-07-02:
Hello @lyd1ng,

can you please provide your VHDL code as a minimal example to reproduce the issue?
You can still edit your initial issue and and the code or attache a file.

tgingold, 2017-07-15:
I close this issue until we get a reproducer.

closed 2017-07-15 by tgingold

mbitsnbites, 2018-03-20:
Hello! I just got the same (or very similar) error:

```
******************** GHDL Bug occured ****************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.36-dev (v0.35-123-g8c8b6285) [Dunoon edition]
Compiled with GNAT Version: 5.4.1
Target: x86_64-linux-gnu
In directory: /home/marcus/git/mrisc32/git/vhdl/
Command line:
ghdl -r --std=08 --work=work --warn-reserved --warn-default-binding --warn-binding --warn-library --warn-specs --warn-unused pipeline_tb
Exception CONSTRAINT_ERROR raised
Exception information:
Exception name: CONSTRAINT_ERROR
Message: grt-processes.adb:770 overflow check failed
Call stack traceback locations:
0x5596255c052f 0x5596255c0d6c 0x5596255c113a 0x5596255c11d2 0x55962554e076 0x5596255d6f84 0x5596255d704d 0x5596257e9e7e 0x5596257f01f9 0x5596257eecb2 0x55962564271f 0x55962554dc9b 0x7fa8727cf1bf 0x55962554ca18 0xfffffffffffffffe
******************************************************************
```

In my case I'm simulating the pipeline of a CPU, and the error occurred after about 15 minutes (probably millions of simulated clock cycles). I will see if I can reproduce the exception more than once.

I have created a branch in my VHDL project here: https://github.com/mbitsnbites/mrisc32/tree/bug/ghdl_constraint_error

If I'm not forgetting anything you should be able to reproduce the error by running `make` in the `vhdl` folder of the MRISC32 repo (having checked out the branch `bug/ghdl_constraint_error`).  ...and wait.

tgingold, 2018-03-21:
It is due to an overflow in a counter.  I have increased the size of the counter.  Will push tomorrow.

mbitsnbites, 2018-03-21:
Sweet! I will rebuild ghdl when it has been integrated. Open source rocks! Turnaround times from bug found -> working software are awsome. :smile:

mbitsnbites, 2018-03-22:
Thanks for the quick fix!

After 67316172 simulated CPU cycles (77 minutes with GHDL mcode), I finally got the first generated image from my MRISC32 CPU:
[image]

tgingold, 2018-03-23:
Nice result!