ghdl/ghdl #379: raised TYPES.INTERNAL_ERROR : sem_assocs.adb:2476

vicencb, 2017-07-02:
Hello Tristan,
ghdl crashed on the following erroneous code:
```vhdl
entity e is end entity;
architecture a of e is
begin
  process
    constant z :integer := 0;
    type t is array(0 to 0) of bit;
    procedure x(
      z :out bit_vector(0 to 0);
      f :out bit_vector(0 to 0)
    ) is
    begin
    end procedure;
    procedure x(
      z :out t;
      f :out bit_vector(0 to 0)
    ) is
    begin
    end procedure;
    function f(arg:t) return bit is
    begin
    end function;
    variable actual_for_f :bit;
    variable actual_for_z :t; -- bit
  begin
    x(
      f(z) => actual_for_f,
      f(z) => actual_for_z
    );
  end process;
end architecture;
```
There are two versions of this code:
 1. `variable actual_for_z :t;`
 1. `variable actual_for_z :bit;`

Both versions are incorrect:
 1. Type mismatch
 2. An unsolvable ambiguity arises

But, in both cases, it triggers a bug report:
`raised TYPES.INTERNAL_ERROR : sem_assocs.adb:2476`
If either `procedure x` is commented out, the issue remains, so, ambiguities due to overloading seem irrelevant here.
Tested with aeae722a.

Thank you!
&nbsp;&nbsp;Vicenç.

diogratia, 2017-07-03:
Function f has to return statement.

tgingold, 2017-09-25:
The output is now:

e.vhdl:25:6:error: cannot resolve overloading for subprogram call

which is correct (but I agree now very informative!)

commit 685526e2 (2017-09-25): Add reproducer for #379

Close #379
closed 2017-09-25 by tgingold